lvgl/lvgl #1816: Unable to turn off font anti-aliasing

tdjastrzebski, 2020-09-25:
`LV_ANTIALIAS=1` does not turn off font anti-aliasing.
As a result, fonts exported with `Horizontal subpixel hinting` option are being anti-aliased and their edges look particularly bad when 8bit indexed colors are being used.  

Secondly, fonts converted with `Horizontal subpixel hinting` are being painted using different color (0x00) than fonts converted without this option selected (second image, color 0x25) when using 8bit color. Examples below. LVGL ver 7.4 

[image]

[image]

kisvegabor, 2020-09-28:
> LV_ANTIALIAS=1 does not turn off font anti-aliasing.

`LV_ANTIALIAS=0` should turn off antialiasing but assume you just mistyped it. 
Anyway, fonts are not antialiased by LVGL but the smooth edges are stored in the fonts' bitmaps. So to get rid of the font antialiasing you need to use fonts with 1bpp.

 > As a result, fonts exported with Horizontal subpixel hinting option are being anti-aliased and their edges look particularly bad when 8-bit indexed colors are being used.

Where do you use 8 bit indexed colors? On a canvas? LVGL support index color format for only a few special things (canvas, images)

> LVGL ver 7.6.0, LV_COLOR_DEPTH=8

It's not indexed but RGB332 format. With this color mixing works in a very ugly way because of the very few colors.

embeddedt, 2020-09-28:
@kisvegabor `LV_ANTIALIAS=1` was a typo. The bug here is that hinting needs to be disabled for 1bpp fonts to look right. It seems to me that it should automatically be disabled if there aren't enough colors.

kisvegabor, 2020-09-28:
Sorry, I'm getting confused a little :sweat_smile: 

By hinting do you mean the sub-pixel rendering? I'm asking because hinting usually means to align the pixel to the "grid" by the converter.

embeddedt, 2020-09-28:
Yes; I was referring to the horizontal subpixel hinting option mentioned above.

kisvegabor, 2020-09-28:
Still not clear. So `hinting` and subpixel rendering (`lcd`) are 2 different [parameter of the font converter](https://github.com/lvgl/lv_font_conv#cli-params). Hence I don't understand the `horizontal subpixel hinting` expression. Hinting and subpixel rendering has nothing to do with each other. (unless I miss something)

embeddedt, 2020-09-28:
The web interface calls it "Horizontal subpixel hinting".

[image]

Maybe that's misleading and should be changed.

kisvegabor, 2020-09-28:
Ahhh, that's really misleading. I'll update it. 

So the problem here is if you use a palette LVGL can't use subpixel rendering because it needs access to red, green, and blue color channels separately.  However, with indexed colors, there is no logical relation between "pixel value" and it's real color. That is if LVGL wants to set a 50% red, 80% blue, and  20% green, it can't easily guess the index from the palette. 

In summary, you can't use subpixel rendering with indexed colors.

tdjastrzebski, 2020-09-28:
Sorry for the confusion. I should read `LV_ANTIALIAS=0`. Correction has been made.

tdjastrzebski, 2020-10-01:
@kisvegabor It is understandable that LVGL (or any other software) cannot do anti-aliasing without knowing what colors it actually renders - which is the case with custom indexed colors.  
However, the problem I report is 1) the lack of ability to turn anti-aliasing completely off. Instead, to properly render fonts without anti-aliasing, fonts need to be re-converted from ttf. 2) Regardless of what font color actually should be rendered, one may expect that the rendered font color (besides edges) is the same, regardless of what options were selected during font conversion from ttf to C code.

kisvegabor, 2020-10-02:
> However, the problem I report is 1) the lack of ability to turn anti-aliasing completely off. Instead, to properly render fonts without anti-aliasing, fonts need to be re-converted from ttf.

You can disable anti-aliasing but can't disable subpixel rendering once the font is generated like this. To solve this LVGL needs to recalculate "normal" pixel values from the subpixel font format if anti-aliasing is disabled. 

> Regardless of what font color actually should be rendered, one may expect that the rendered font color (besides edges) stays the same, regardless of what options were selected during font conversion from ttf to C code.

It's because indexed colors and subpixel rendering are not compatible. Here LVGL should recalculate "normal" pixel values from the subpixel font format if the indexed color format is used. (However, there is no dedicated indexed color format in LVGL, only the RGB332 which actually can be used as indexed as well)

The root of both issues is that you are using subpixel rendered font when you don't want subpixel rendering. But why do you converted the font this way, when don't want this font feature? 

Converting subpixel fonts back to normal would have some performance and memory overhead but if can find a good use case when this conversation is really required, of course, we can discuss it. I just don't understand, why have ticked that checkbox, when you don't need this feature. :slightly_smiling_face:

embeddedt, 2020-10-02:
Another option is to generate font files with an `#if LV_ANTIALIAS`/`#else` clause that includes the non-subpixel rendered version if antialiasing is disabled. The disadvantage is that the C source files will be even larger.

kisvegabor, 2020-10-05:
It seems doable but I think it's not required. IMO the user can decide in advance whether he wants to use antialiasing and/or subpixel rendering or nor.

closed 2020-11-09 by stale[bot]