lvgl/lvgl #1818: Performance issues

iDalink, 2020-09-29:
I test following code with lvgl 6.1 & 7.xx?
It spend 56ms in lvgl6 environment, but 380ms with lvgl7.
Why does new verson lvgl has such performance?

```
unsigned int os_ticks_get(void)
{
    struct timespec ts;
    clock_gettime(CLOCK_MONOTONIC, &ts);
    return (ts.tv_sec * 1000 + ts.tv_nsec / 1000000);
}

void all_screen(void)
{
	lv_obj_t *scr = lv_scr_act();
	lv_obj_t *cont = lv_page_create(scr, NULL);
	lv_obj_set_size(cont, lv_obj_get_width(scr), lv_obj_get_height(scr));

	uint32_t ticks1 = os_ticks_get();
	for (int i = 0; i < 200; i++)
	{
		lv_obj_t *obj = lv_obj_create(cont, NULL);
		lv_obj_set_size(obj, 200, 200);
		lv_obj_align(obj, NULL, LV_ALIGN_CENTER, 0, 0);
	}
	uint32_t ticks2 = os_ticks_get();
	printf("loop used:%d\n", ticks2 - ticks1);
}

//please invoke all_screen
```

kisvegabor, 2020-09-29:
Hi, 

Some questions:
- Do you use the latest version?
- Have you disabled `LV_USE_DEBUG` in `lv_cong.h`?

iDalink, 2020-09-30:
Thanks, Disable macro LV_USE_DEBUG, and it works. 

But another case remains performance issue.It is a input method panel, and it use 1000ms at my devlopment ARM A7 board. In the previos version, it works with lvgl6 used 50ms. Could you test follow code performance?

code for lvgl6
```

static lv_style_t kb_ta_style;
static lv_style_t kb_ta_placeholder_style;
static lv_style_t kb_cursor_style;
static lv_style_t kb_label_style;
static lv_style_t kb_bottom_label_style;
static lv_style_t kb_btn_style_black;
static lv_style_t kb_btn_style_rl;
static lv_style_t kb_btn_style_pr;
static lv_style_t kb_page_style;
static lv_style_t kb_btn_style_rl;

#define COLOR_TXT_BRIGHT    lv_color_hex(0xD0D0D0)
#define COLOR_TXT_DARK      lv_color_hex(0x666666)
#define COLOR_TXT_SHINY     lv_color_hex(0xEC9709)
#define COLOR_TXT_SHINY_PR  lv_color_hex(0xF1B552)

#define COLOR_BG_SCREEN     lv_color_hex(0x000000)
#define COLOR_BG_BLOCK      lv_color_hex(0x080808)
#define COLOR_BG_BLOCK_PR   lv_color_hex(0x292929)
#define COLOR_BG_BTN_DFT    lv_color_hex(0x323232)
#define COLOR_BG_BTN_PR     lv_color_hex(0x464646)
#define COLOR_BG_SHINY      lv_color_hex(0xEC9709)
#define COLOR_BG_GREEN		lv_color_hex(0x025255)

static void ui_state_wifi_kb_add_tab(lv_obj_t *parent, int w, int h,
                                     char *first_char, char *chars, lv_obj_t **first_obj,
                                     lv_event_cb_t event_cb, void *user_data)
{
	int head_count = first_char ? 1 : 0;
    int count = strlen(chars);
    for (int i = 0; i < count + head_count; i++)
    {
        int column = i % 4;
        int line = i / 4;
        int btn_width = w / 4;
        int btn_height = h / 3;
        char title[2];

        lv_obj_t *btn = lv_btn_create(parent, NULL);
        lv_obj_set_pos(btn, btn_width * column, btn_height * line);
        lv_obj_set_size(btn, btn_width, btn_height);
        lv_btn_set_style(btn, LV_BTN_STYLE_REL, &kb_page_style);
        lv_btn_set_style(btn, LV_BTN_STYLE_PR, &kb_btn_style_pr);
        lv_btn_set_style(btn, LV_BTN_STYLE_INA, &kb_btn_style_pr);
		lv_page_glue_obj(btn, true);
		lv_obj_set_event_cb(btn, event_cb);
		btn->user_data = user_data;

        lv_obj_t *label = lv_label_create(btn, NULL);
		lv_obj_set_user_data(label, (void *)0);
        if (i < head_count)
        {
            lv_label_set_text(label, first_char);
			*first_obj = btn;
        }
        else
        {
			title[0] = chars[i - head_count];
			title[1] = 0;
            lv_label_set_text(label, title);
			if (strcmp(title, " ") == 0)
			{
				lv_obj_del(label);
[... 362 lines cut ...]
		{
			int btn_y = lv_obj_get_y(tabview) + lv_obj_get_height(tabview);
        	int btn_height = lv_obj_get_height(main_obj) - btn_y;

			lv_obj_t *rec_btn = lv_obj_create(main_obj, NULL);
			lv_obj_add_style(rec_btn,  LV_OBJ_PART_MAIN, &kb_bot_btn_style);
        	lv_obj_set_pos(rec_btn, (59+1) * j, btn_y);
        	lv_obj_set_size(rec_btn, 60, btn_height);

			lv_obj_t *btn = lv_imgbtn_create(rec_btn, NULL);
            //lv_obj_add_style(&btn, LV_IMGBTN_PART_MAIN, &kb_bot_btn_style);
			lv_obj_align(btn, NULL, LV_ALIGN_CENTER, 0, 0);

			lv_obj_set_ext_click_area(btn, 10, 17, 5, 15);
		}
	}
    printf("ui_state_wifi_kb_create 3 ticks:%d\n", os_ticks_get());
	lv_tabview_set_tab_act(tabview, 0, LV_ANIM_OFF);
    lv_obj_add_style(tabview, LV_TABVIEW_PART_BG, &kb_page_style);
    lv_obj_add_style(tabview, LV_TABVIEW_PART_TAB_BG, &kb_page_style);
    printf("ui_state_wifi_kb_create  4 ticks:%d\n", os_ticks_get());
    printf("create used:%d\n", os_ticks_get() - ticks1);
	return main_obj;
}
```

kisvegabor, 2020-09-30:
I pushed an improvement. Please try with the latest master branch.

iDalink, 2020-09-30:
> I pushed an improvement. Please try with the latest master branch.

Thanks! I merge newest code. It use 560ms now. Can it has more efficiency?

kisvegabor, 2020-09-30:
Can you run a profiler to see what takes so long?

Anyway, it'd be much faster to use a button matrix widget instead of these many buttons. So at worst case, it can be a workaround.

iDalink, 2020-10-09:
A profiler means? Is there a operation manual?

kisvegabor, 2020-10-09:
I suppose you have some kind of Linux running. If so you can use [perf](https://perf.wiki.kernel.org/index.php/Main_Page).

It can be used like:
```

perf record ./my_app 
# Let the program run and exit it when the objects are created

# Display which functions was the most "active"
perf report
```

It can be installed as part of `linux-tools`.

closed 2020-11-09 by stale[bot]