lvgl/lvgl #1819: Duplicated doxygen comments in .c and .h

jozba, 2020-09-29:
**Describe the bug**
Doxygen comments are in both .c and .h files. In fact, they are duplicated. And this brings up a work with synchronization, and a chance of missing something in case "I updated this, but forgot to update also there". 

**To Reproduce**
Open random source file and header file. 

**Expected behavior**
I guess only one file could contain comments. I prefer the header file, as, in generall, source files may not be distributed with product, so user get only binary library + headers. 

**Screenshots**
n/a

**Additional context**
n/a

embeddedt, 2020-09-29:
I've noticed this as well and I agree - the header file is probably a better spot. If @kisvegabor agrees I can gradually remove the Doxygen comments from the source files.

kisvegabor, 2020-09-30:
I also agree.

But please wait until we feeze v7 and v8's dev branch becomes more stable.

closed 2020-11-01 by stale[bot]