lvgl/lvgl #1820: Issues with label selection

ryanpowers, 2020-09-29:
**Describe the bug**

Text selection in labels appears to produce incorrect results. If there is a text selection, font color changes to the selection color. Selection color (used for background of selection) is always set to default value.

**To Reproduce**

Make a label, change text selection color (text_sel_color property) to something other than default blue, apply a text selection, entire label text will appear as the chosen text selection color, background of selection will appear blue.

I am working in monochrome, so the steps to reproduce may not be 100% accurate, as for me blue and white are identical colors.

**Expected behavior**

Text selection background changes to text_sel_color, font color remains unchanged.

**Additional context**

I was looking to use text selection as a means to invert parts of a label on this monochrome display. I came across [this thread on the forum](https://forum.lvgl.io/t/how-to-change-color-of-selected-text-in-label/1946), which is the same thing I was wanting to do. It did not appear it had been implemented, so I started implementing the additional property myself (text_sel_font_color).

In doing so, I found an issue where the label descriptor never has the sel_color property set other than to the default value, and there appeared to be an additional bug in lv_obj_init_draw_label_desc (lv_obj.c@3508 for the latest commit to master as of this moment) where any text selection on a label would cause the descriptor's color property (for text color) to be overwritten with the selection color (intended as background behind selected text).

I have locally fixed this bug, and additionally implemented the property to change the color of the text in a selection. Per the documentation, I have opened this issue for the discussion before submitting a pull request. I am not certain if my naming of the new property is appropriate or if there is a better name to use.

ryanpowers, 2020-09-29:
I forgot about the simulator! Great tool to have. I duplicated the problem there.

First screenshot is the unmodified UI. The left "Button" in the background is where the changes are being made.
[image]
This screenshot simply changes the text to green with the label's local style.
`lv_obj_set_style_local_text_color(label, LV_LABEL_PART_MAIN, LV_STATE_DEFAULT, LV_COLOR_GREEN);`
[image]
This screenshot adds to the previous by selecting characters 1-3. Note the text changing to black.
```c
    lv_label_set_text_sel_start(label, 1);
    lv_label_set_text_sel_end(label, 3);
```
[image]
This screenshot adds to the previous by setting the selection color to red. Note that the selection stays blue and all the text changes to red.
`lv_obj_set_style_local_text_sel_color(label, LV_LABEL_PART_MAIN, LV_STATE_DEFAULT, LV_COLOR_RED);`
[image]

ryanpowers, 2020-09-29:
Here is the same code as the last shot above, but with the lv_obj.c fix applied. I did not include the selected text color enhancement in this case. This is now correct with text green and the selection background red. 
[image]

ryanpowers, 2020-09-29:
I have forked LVGL and added my changes to a branch. The bug fix and new property are in separate commits, so I can create a pull request for either or both.
https://github.com/ryanpowers/lvgl/tree/text_selection_enhancement

kisvegabor, 2020-09-30:
Hi,

Thank you for the detailed description and the proposed fixes. 

It seems you are right with both the missing property and the bug with using the selected color.

The only this I'd change in your commits is using this property names:
```c
LV_STYLE_TEXT_SEL_COLOR
LV_STYLE_TEXT_SEL_BG_COLOR
```

It would change the current way the properties work (`SEL_COLOR` was the background so far) but it was broken so I think it's not a problem if we change this.

ryanpowers, 2020-09-30:
I will make that change and submit a PR.

ryanpowers, 2020-09-30:
So, in incorporating the full change set into the demo app, I am realizing that there do not appear to be default styles for selected text in the material/mono themes. This is resulting in a default black-on-black selection, which is probably not desirable. I haven't really dug into the theming feature yet (though I am pretty sure it would make my life a lot easier if I learned it).

Do you want me to update the material and mono themes? If so, do I just add the lv_style_set... calls to the label init?

kisvegabor, 2020-10-01:
Ture, it's really not specified. 

I need to update some other font-related stuff in the themes too so I can do it as well when the PR is open.

kisvegabor, 2020-10-07:
I've updated the themes to support this new feature. Now the mono themes use inverted colors on selections.

Thank you for your contribution!

commit 1d50bc96 (2020-10-07): theme update to support text selection background
Fixes #1820
closed 2020-10-07 by kisvegabor