lvgl/lvgl #1821: fbdev show error

Layty, 2020-09-30:
I can use QT tu show the Fb, so I think My driver is ok.

when I show `123456789` ,It show monitor like that 
[image]

so  which config should I do  

my oled is 160*160, 1bpp

Layty, 2020-09-30:
Now I have fix It  in  /lv_drivers/display/fbdev.c  

[image]

kisvegabor, 2020-09-30:
Hi,

I don't know if it was ever tested with 1bpp frame buffers. 
It seems logical to map the MSB to the left pixel so we can update the driver like this. Can you send a Pull request?

embeddedt, 2020-09-30:
It might be worth adding an option for this in `lv_drv_conf.h`, otherwise we'll get an issue opened in a few weeks mentioning that the driver is now broken for another 1bpp display we didn't know about. :wink:

kisvegabor, 2020-10-01:
True. 
I think something like `FBDEV_1BPP_MSB_LEFT` can work.

Layty, 2020-10-09:
I am a rookie, I suggess If any api about frambuf for app user to know  MSB or LSB?

kisvegabor, 2020-10-09:
> I suggess If any api about frambuf for app user to know MSB or LSB?

Do you mean if the system tells the order?

Layty, 2020-10-09:
yes ,  I think Qt works well my be it use some api to read fb driver info

Layty, 2020-10-09:
mybe use  like
struct fb_bitfield {
	__u32 offset;			/* beginning of bitfield	*/
	__u32 length;			/* length of bitfield		*/
	__u32 msb_right;		/* != 0 : Most significant bit is */ 
					/* right */ 
};

closed 2020-10-12 by Layty

kisvegabor, 2020-10-13:
Please let us know if you find a related configuration.