lvgl/lvgl #1826: New feature in 7.6.0 (style change detect) breaks image button

jupeos, 2020-09-30:
**Describe the bug**
1. Image button fails to apply `image_recolor` style.
2. Setting the state of an image button has no effect.

Although this may seem like 2 separate issues they both relate to the same change introduced in `7.6.0`. There may be other related issues that I haven't come across yet (my testing was not thorough). It just so happens that I was using an image button in my project and it's behaviour broke on upgrading to `7.6.0`

**To Reproduce**
Create a sample project based on `lv_ex_imgbtn` found in the `lv_examples` repo. Code is included here for convenience.
```
    LV_IMG_DECLARE( imgbtn_green ); // The image is irrelevant for this example.
    LV_IMG_DECLARE( imgbtn_blue );

    /*Darken the button when pressed*/
    static lv_style_t style;
    lv_style_init( &style );
    lv_style_set_image_recolor_opa( &style, LV_STATE_PRESSED, LV_OPA_30 );
    lv_style_set_image_recolor( &style, LV_STATE_PRESSED, LV_COLOR_BLACK );
    lv_style_set_text_color( &style, LV_STATE_DEFAULT, LV_COLOR_WHITE );

    /*Create an Image button*/
    lv_obj_t* imgbtn1 = lv_imgbtn_create( lv_scr_act(), NULL );
    lv_imgbtn_set_src( imgbtn1, LV_BTN_STATE_RELEASED, &imgbtn_green );
    lv_imgbtn_set_src( imgbtn1, LV_BTN_STATE_PRESSED, &imgbtn_green );
    lv_imgbtn_set_src( imgbtn1, LV_BTN_STATE_CHECKED_RELEASED, &imgbtn_blue );
    lv_imgbtn_set_src( imgbtn1, LV_BTN_STATE_CHECKED_PRESSED, &imgbtn_blue );
    lv_imgbtn_set_checkable( imgbtn1, true );
    lv_obj_add_style( imgbtn1, LV_IMGBTN_PART_MAIN, &style );
    lv_obj_align( imgbtn1, NULL, LV_ALIGN_CENTER, 0, -40 );

    /*Create a label on the Image button*/
    lv_obj_t* label = lv_label_create( imgbtn1, NULL );
    lv_label_set_text( label, "Button" );
```
The above may be used to test item 1.
Item 2 may be tested by adding something like this to toggle the buttons state.
```
    int i = 0;
    bool state = false;
    while( 1 )
    {
        /* Periodically call the lv_task handler.
     * It could be done in a timer interrupt or an OS task too.*/
        lv_task_handler();
        Sleep( 10 ); /*Just to let the system breathe */

        if(++i % 50 == 0)
        {
            // Toggle the button state
            state = !state;
            lv_btn_set_state( imgbtn1, state );
        }
    }
```

**Expected behavior**
1. Button should be recolored when pressed but is not.
2. Button should change state (visually) when the state is set but does not.

**Additional context**
This works correctly in `7.5.0`. I have isolated it to the changes made in `lv_obj.c` for `7.6.0`

jupeos, 2020-10-01:
I can confirm that the tip appears to work with the above test but this problem is a little more nuanced.
To reproduce it I have simplified the above code slightly by making a simple image button with different up/down images.

With the style removed (commented out) the button image does not update when the state is changed/pressed.
When the style is added the button works as expected.

```
    LV_IMG_DECLARE( imgbtn_green );
    LV_IMG_DECLARE( imgbtn_blue );

    /*Darken the button when pressed*/
    static lv_style_t style;
    lv_style_init( &style );
    lv_style_set_image_recolor_opa( &style, LV_STATE_PRESSED, LV_OPA_30 );
    lv_style_set_image_recolor( &style, LV_STATE_PRESSED, LV_COLOR_BLACK );
    lv_style_set_text_color( &style, LV_STATE_DEFAULT, LV_COLOR_WHITE );

    /*Create an Image button*/
    lv_obj_t* imgbtn1 = lv_imgbtn_create( lv_scr_act(), NULL );
    lv_imgbtn_set_src( imgbtn1, LV_BTN_STATE_RELEASED, &imgbtn_green );
    lv_imgbtn_set_src( imgbtn1, LV_BTN_STATE_PRESSED, &imgbtn_blue );

    // This line makes the difference
    //lv_obj_add_style( imgbtn1, LV_IMGBTN_PART_MAIN, &style );

    lv_obj_align( imgbtn1, NULL, LV_ALIGN_CENTER, 0, -40 );

    /*Create a label on the Image button*/
    lv_obj_t* label = lv_label_create( imgbtn1, NULL );
    lv_label_set_text( label, "Button" );
```

kisvegabor, 2020-10-02:
Hmm, it seems style caching introduced some strange issues.

Anyway I fixed it in `master`.

jupeos, 2020-10-02:
Nearly there but not quite (I synced and tested your changes).
Although the button works when pressed manually, a programmatic state change does not work, i.e. `lv_btn_set_state( imgbtn1, state );` unless the style is included (see previous description).

kisvegabor, 2020-10-05:
It works for me with this code:
```c
    LV_IMG_DECLARE(imgbtn_green);
    LV_IMG_DECLARE(imgbtn_blue);

    /*Create an Image button*/
    lv_obj_t * imgbtn1 = lv_imgbtn_create(lv_scr_act(), NULL);
    lv_imgbtn_set_src(imgbtn1, LV_BTN_STATE_RELEASED, &imgbtn_green);
    lv_imgbtn_set_src(imgbtn1, LV_BTN_STATE_PRESSED, &imgbtn_green);
    lv_imgbtn_set_src(imgbtn1, LV_BTN_STATE_CHECKED_RELEASED, &imgbtn_blue);
    lv_imgbtn_set_src(imgbtn1, LV_BTN_STATE_CHECKED_PRESSED, &imgbtn_blue);

    lv_imgbtn_set_checkable(imgbtn1, true);
    lv_imgbtn_set_state(imgbtn1, LV_BTN_STATE_CHECKED_RELEASED);
    lv_obj_align(imgbtn1, NULL, LV_ALIGN_CENTER, 0, -40);

    /*Create a label on the Image button*/
    lv_obj_t * label = lv_label_create(imgbtn1, NULL);
    lv_label_set_text(label, "Button");
```

What should I modify to reproduce the issue?

jupeos, 2020-10-05:
Try this...
```
    int i = 0;
    bool state = false;
    while( 1 )
    {
        /* Periodically call the lv_task handler.
     * It could be done in a timer interrupt or an OS task too.*/
        lv_task_handler();
        Sleep( 10 ); /*Just to let the system breathe */

        if(++i % 50 == 0)
        {
            // Toggle the button state
            state = !state;
            lv_btn_set_state( imgbtn1, state );
        }
    }
```

jupeos, 2020-10-05:
I do notice that you use `lv_imgbtn_set_state` in your code sample where I use `lv_btn_set_state`. I had assumed that, since an image button derives from a regular button that it could be used in place i.e. `lv_btn` operations may be passed an `lv_imgbtn` object - no?

embeddedt, 2020-10-05:
@jupeos They are [identical](https://github.com/lvgl/lvgl/blob/e557b0fbac9a198149d2cdc5919adbacd0799789/src/lv_widgets/lv_imgbtn.h#L116-L119).

kisvegabor, 2020-10-06:
It seems the issue is `lv_(img)btn_set_state` needs `LV_BTN_STATE_...` as argument and not bool. In your snippet `false` (0) will turn to `LV_BTN_STATE_RELEASED` and `true` (1) to `LV_BTN_STATE_PRESSED`.

jupeos, 2020-10-06:
That's not the issue - there is still a bug. If you look in my second code sample I am not using a checkable button so I should be able to use `LV_BTN_STATE_RELEASED` and `LV_BTN_STATE_PRESSED` (or a boolean, which I though was by design). Note: This used to work and now does not. Here is a complete set of test code for you.

    LV_IMG_DECLARE( imgbtn_green );
    LV_IMG_DECLARE( imgbtn_blue );

    /*Create an Image button*/
    lv_obj_t* imgbtn1 = lv_imgbtn_create( lv_scr_act(), NULL );
    lv_imgbtn_set_src( imgbtn1, LV_BTN_STATE_RELEASED, &imgbtn_green );
    lv_imgbtn_set_src( imgbtn1, LV_BTN_STATE_PRESSED, &imgbtn_blue );
    lv_obj_align( imgbtn1, NULL, LV_ALIGN_CENTER, 0, -40 );

    /*Create a label on the Image button*/
    lv_obj_t* label = lv_label_create( imgbtn1, NULL );
    lv_label_set_text( label, "Button" );```

    int i = 0;
    lv_btn_state_t state = LV_BTN_STATE_RELEASED;
    while( 1 )
    {
        /* Periodically call the lv_task handler.
         * It could be done in a timer interrupt or an OS task too.*/
        lv_task_handler();
        Sleep( 10 ); /*Just to let the system breathe */

        if( ++i % 50 == 0 )
        {
            // Toggle the button state
            if( state == LV_BTN_STATE_RELEASED )
            {
                state = LV_BTN_STATE_PRESSED;
            }
            else
            {
                state = LV_BTN_STATE_RELEASED;
            }
            lv_btn_set_state( imgbtn1, state );
        }
    }
```

kisvegabor, 2020-10-07:
Thanks, now I could reproduce the issue. 

I needed to update `lv_imgbtn_set_state` so you can't use `lv_btn_set_state` instead of it.

See f1a198bd

jupeos, 2020-10-07:
Looks good - thanks.

kisvegabor, 2020-10-08:
Great, thank you!

closed 2020-10-08 by kisvegabor