lvgl/lvgl #1828: lvgl image converter - Wrong format in comment in generated file

SaadTiwana, 2020-10-01:
**Describe the bug**

When i use the lvgl imager converter (web) utility to convert an image, under LV_COLOR_DEPTH==32, the comment says the format begins with "Fix 0xFF" (see screenshot below). However, the actual output format has the Fix 0xFF at 4th position (also see screenshot below). Kindly change the comment to reflect the actual format as it causes confusion.

**To Reproduce**

To reproduce issue, upload any image to the image converter web utility, and choose color format as any one of the three "True Color *" options, and output format as "C array".

**Expected behavior**
Format in comment should match actual generated format.

**Screenshots**
[image]

**Additional context**
N/A

Best Regards,
Saad

p.s Thank you for the awesome library! I am so glad i came across this!

kisvegabor, 2020-10-02:
Hi,

I've just fixed it. Can you confirm if it's ok now?

SaadTiwana, 2020-10-02:
Hi,
Yes, it looks good now. Just last night i found out that the actual format actually is GBRA, and not RGBA, but i see you have already updated the sequence in the comments as well.

Thanks a lot ! I will not close this issue.

Best Regards,
Saad

closed 2020-10-02 by SaadTiwana