lvgl/lvgl #1831: lv_label_set_text_fmt does not support float?

KiraVerSace, 2020-10-02:
`lv_label_set_text_fmt(label, "Num:[%d], T[%.2f], H[%.2f]", logEvent.id, 23.6, 23.7);
`
the %d can display, but %.2f are in the null, display nothing in the screen.

and I dont know why?

embeddedt, 2020-10-02:
https://github.com/lvgl/lvgl/blob/34e0fefeb3e41deb470a7a39685d0e23eeee60a3/lv_conf_template.h#L521

KiraVerSace, 2020-10-02:
> https://github.com/lvgl/lvgl/blob/34e0fefeb3e41deb470a7a39685d0e23eeee60a3/lv_conf_template.h#L521
> 
> Change it to 0 in your `lv_conf.h` file to enable floating-point.

https://github.com/lvgl/lv_arduino/blob/master/lv_conf.h 
I use it in stm32duino under vscode with platformio, and there is no the configuration?

Thank you very much!

embeddedt, 2020-10-02:
You can just add the line [here](https://github.com/lvgl/lv_arduino/blob/fd88bd892f0adb781556f4d12035cb8d5e574aea/lv_conf.h#L482).

KiraVerSace, 2020-10-02:
thank you!
I doubt why it will disappear in the lv_arduino?
3Q!

embeddedt, 2020-10-02:
It's because that project is using a relatively outdated `lv_conf.h` file.

closed 2020-10-02 by KiraVerSace

jonathanfoster, 2024-01-08:
Just a heads up for anyone experiencing this issue on v8, the flag has been renamed to `LV_SPRINTF_USE_FLOAT`. https://github.com/lvgl/lvgl/blob/release/v8.0/lv_conf_template.h#L205

kisvegabor, 2024-01-08:
Thanks! 

And it will be `LV_USE_FLOAT` in v9. :slightly_smiling_face: