lvgl/lvgl #1832: Styles enhancement suggestion: Descendant Selectors

amirgon, 2020-10-02:
This continues the discussion on https://github.com/lvgl/lvgl/issues/1551#issuecomment-637799157 and https://github.com/lvgl/lvgl/issues/1811#issuecomment-701242788.

I suggest adding a way to apply a style recursively on a component sub-tree (a screen or some other container component which is an ancestor of other components)  

The CSS counterpart is ["Descendant Selectors"](https://developer.mozilla.org/en-US/docs/Web/CSS/Descendant_combinator) 
> The descendant combinator — typically represented by a single space ( ) character — combines two selectors such that elements matched by the second selector are selected if they have an ancestor (parent, parent's parent, parent's parent's parent, etc) element matching the first selector. Selectors that utilize a descendant combinator are called descendant selectors.

Usage examples:

- BIDI is currently already applied this way (recursively across component subtrees) but by a specialized mechanism.  
With Descendant Selectors we could generalize it and change BIDI to a style instead of a special property of lv_obj.
- Font is usually applied together with BIDI.  
It would make sense to create a single style that sets both BIDI and font and apply that style on a container component (or screen) recursively.  
Currently font and BIDI need to be applied separately, and there is no convenient way to apply font style recursively where BIDI was applied.

kisvegabor, 2020-10-05:
Thank you for opening this issue!

> BIDI is currently already applied this way (recursively across component subtrees) but by a specialized mechanism.
With Descendant Selectors we could generalize it and change BIDI to a style instead of a special property of lv_obj.

It's similar to how some style properties are inherited.

> Font is usually applied together with BIDI.
It would make sense to create a single style that sets both BIDI and font and apply that style on a container component (or screen) recursively.
Currently font and BIDI need to be applied separately, and there is no convenient way to apply font style recursively where BIDI was applied.

I agree to turn Bidi to style property. 

---

Now we differ from CSS at an important point: LVGL defines states on property level but CSS defines it on class level.
LVGL
```c
lv_style_set_border_color(&style1, LV_STATE_PRESSED, LV_COLOR_RED)
```

CSS
```css
.style1:pressed {
  border-color: red;
}
```

I think we should follow CSS's approach because adding selector to every property is very expensive in memory. If we take this direction we would have great freedom by adding callback(s) which can decide whether a style can be applied in the current context or not. 

It'd look something like:
```c
/*Default style*/
static lv_style_t style1;
lv_style_init(&style1);
lv_style_set_bg_color(&style1, LV_COLOR_BLUE);

/*Pressed style*/
static lv_style_t style1_pr;
lv_style_init(&style1_pr);
lv_style_set_state(&style1_pr, lv_style_select state, LV_STATE_PRESSED);
lv_style_set_selector(&style1_pr, my_selector_cb, 123);         /*A custom selector with a parameter*/
lv_style_set_bg_color(&style1_pr, LV_COLOR_RED);
```

What do you think?

amirgon, 2020-10-05:
@kisvegabor - 
- How does `my_selector_cb` look like? Could you give an example to `my_selector_cb` implementation?
- Could you explain the `lv_style_select state` argument you provide to `lv_style_set_state`?

kisvegabor, 2020-10-05:
Minor update and fix on the example:
```c
lv_style_add_selector(&style1_pr, lv_style_select_state, LV_STATE_PRESSED);
lv_style_add_selector(&style1_pr, my_selector_cb, 123);         /*A custom selector with a parameter*/

bool lv_style_select_state(lv_obj_t * obj, lv_style_t * style, uint32_t param)
{
   if(obj->state & param) return true;
   else return false;
}

bool my_selector_cb(lv_obj_t * obj, lv_style_t * style, uint32_t param)
{
   lv_obj_t * parent = obj;
   while(parent) {
      if(lv_obj_has_type(parent, "lv_btn")) return true;
     parent = lv_obj_get_parent(parent);
  }
  return false;
}
```

It's still not perfect because the state matches have "weight" (only one state matches or all states matches) not only true/false but I think shows the idea I meant.

embeddedt, 2020-10-05:
Two suggestions:

* The current approach of adding a style to an individual object is decently intuitive. I'd suggest keeping this behavior if `lv_style_add_selector` is not called (i.e. the style has no selectors).
* Is it possible to implement backwards compatibility so that existing users do not need to split their styles into individual ones for each state?

Other than that this looks good.

kisvegabor, 2020-10-05:
@embeddedt Adding only an optional selector callback is certainly possible and backward-compatible too. 

However, I see some advantages of style-based states:
- Shorter function names: 
`lv_style_set_bg_color(&style1, LV_STATE_DEFAULT, LV_COLOR_RED)` will be 
`lv_style_set_bg_color(&style1, LV_COLOR_RED)`.
- `LV_STATE_...` is repeatedin every call.
- It faster because some styles can be immediately skipped.

embeddedt, 2020-10-05:
I fully agree that style-based states are better, however, we need a system in place to make the upgrade as seamless as possible.

kisvegabor, 2020-10-05:
What if we allow both for a major release and mark the property based styles as deprecated?

embeddedt, 2020-10-05:
I think that would work. Then in 9.0 we would remove the state parameter from the individual `lv_style_set_xxx` functions?

kisvegabor, 2020-10-05:
>  Then in 9.0 we would remove the state parameter from the individual lv_style_set_xxx functions?

Yes, I meant that. Just to confirm:
- v7: 
  - add grid
  - new scroll
  - complex widgets to components
  - remove `lv_page` and `lv_cont`
  -  make `copy` parameter deprecated
- v8: 
  - add tests
  -  rework groups
   - remove `copy` parameter
   - style selectors
   -  style-based states
- v9: 
   - remove property level states

amirgon, 2020-10-06:
> ```c
> lv_style_add_selector(&style1_pr, lv_style_select_state, LV_STATE_PRESSED);
> lv_style_add_selector(&style1_pr, my_selector_cb, 123);         /*A custom selector with a parameter*/
> ```

- I assume both selectors need to return `true` for the style to be applied. Is that right?
- I think we should have a list of built-in selector callbacks for the users, and not force the users to write their own callbacks.  
Instead of "my_selector_cb" which the user defines, I think LVGL should provide common selector callbacks.  
It would make it easier for the C users, but is also very important for Micropython from performance perspective. Micropython could be too slow for callbacks that are called very frequently, so ready-made C callbacks (that are available in Micropython) can be a good solution.

> v7:
> * ...
> * remove `lv_page` and `lv_cont`

I'm using `lv_page` and `lv_cont` in v7 today.  
Are you going to remove them on **current** release (v7)?  Isn't that a breaking change?  
What should I do if I need `lv_page` and `lv_cont` functionality?

kisvegabor, 2020-10-06:
> I assume both selectors need to return true for the style to be applied. Is that right?

Yes.

> I think we should have a list of built-in selector callbacks for the users, and not force the users to write their own callbacks.
Instead of "my_selector_cb" which the user defines, I think LVGL should provide common selector callbacks.

I definitely agree.

> I'm using lv_page and lv_cont in v7 today.

Upps, the version in the list should have started from v8. I updated it. So they will be removed from v8.

kisvegabor, 2020-10-06:
>> I assume both selectors need to return true for the style to be applied. Is that right?

> Yes.

I've just realized that it's not always true that you want AND relationship between selectors. What if you want to paint all labels to red if they are on a chart OR a list?  Maybe in this case you need a custom selector.

amirgon, 2020-10-06:
> I've just realized that it's not always true that you want AND relationship between selectors. What if you want to paint all labels to red if they are on a chart OR a list? Maybe in this case you need a custom selector.

Or LVGL can provide a "combining" selector that receives a list of callbacks and performs OR between them (or some other logical operation).

kisvegabor, 2020-10-06:
> Or LVGL can provide a "combining" selector that receives a list of callbacks and performs OR between them (or some other logical operation).

I think it'd be effective with `vargs` but as you mentioned earlier Micropython doesn't support it. 

> btw need to make sure that selector callbacks prototype would comply with the callback conventions. 

Can you confirm if it's ok?
```c
bool lv_style_select_state(lv_style_t * style, lv_obj_t * obj, uint32_t param)
```

amirgon, 2020-10-06:
> I think it'd be effective with `vargs` but as you mentioned earlier Micropython doesn't support it.

I support a pointer to array of callbacks.  
In this case param would need to be `void* param[]` or `selector_cb selectors[]` would be even better.

> Can you confirm if it's ok?
> 
> ```c
> bool lv_style_select_state(lv_style_t * style, lv_obj_t * obj, uint32_t param)
> ```

Yes.
But the callback itself must also receive `lv_style_t * style` as the first argument, and `lv_style_t` must contain a `user_data` field.

kisvegabor, 2020-10-06:
> I support a pointer to array of callbacks.

It's really a possible solution. Although it seems to it's easier to make people write custom functions if they need something special.  It seems more flexible and readable. 

> lv_style_t must contain a user_data field.

Okay, noted.

kisvegabor, 2020-10-16:
I added the new selector idea to the [ROADMAP](https://github.com/lvgl/lvgl/blob/master/docs/ROADMAP.md#v9) and scheduled for v9.

So I close this issue until that. Feel free to comment here if any idea or remark came to your mind.

puzrin, 2020-10-21:
IMO descendant selectors can be easily avoided if use BEM methodology (assign unique classes + modifiers to unique elements).

At first glance, all you need is `select(class_id, state_id)`.

Note, curent class (style) merge algorythm is restriced to NOT have descendant class rules. That's intended to keep things simple and fast. In this context, introduce of descendant selectors is the road to hell.

amirgon, 2020-10-21:
@puzrin Suppose I want to apply a font to all components in a screen.
Would I need to mark each and every component with a class? 
That's another function call for every component - I would like to avoid that.

puzrin, 2020-10-21:
> Suppose I want to apply a font to all components in a screen.
> Would I need to mark each and every component with a class?
> That's another function call for every component - I would like to avoid that.

No, because fonts are inherited. In terms of HTML, you need to apply font to `body` only (root node).

BEM is for different case. Imagine, you have

```
list
  item1
  item2
  item3
  ...
```

Then each item MUST have class `my_list_item`. And you can modify all items in this lists via single style.

So, instead of define descendant selector `list > item`, you can address item directly.

At first glance, it may look stupid and boring to assign class (style) for each object. But, trust me, in long term this is much more simple to maintain. When you learn this approach once, it become very easy to use.

amirgon, 2020-10-21:
> No, because fonts are inherited. In terms of HTML, you need to apply font to `body` only (root node).

In LVGL, is there a list somewhere of inherited properties?  
The [docs say](https://docs.lvgl.io/latest/en/html/overview/style.html#inheritance) "Some properties (typically that are related to texts) can be inherited" but I couldn't find a list of inherited/non-inherited properties anywhere. 

> Then each item MUST have class `my_list_item`. And you can modify all items in this lists via single style.
>
> So, instead of define descendant selector `list > item`, you can address item directly. [...]

In this case, I would prefer providing the "classes" in the component constructor instead of calling a separate function for each component.  

@kisvegabor How about providing "classes" as a constructor argument of every component?  
If classes are modeled as bits, then the "classes" argument would be an integer built from ORed (or plus since there is no carry). That could be simple, readable and efficient (although limited to 32 classes).

puzrin, 2020-10-21:
> In LVGL, is there a list somewhere of inherited properties?

I don't know. But those are expected to behave exactly as in HTML.

> In this case, I would prefer providing the "classes" in the component constructor instead of calling a separate function for each component.

Hm... i'm not fan of pushing concrete methodology into signatures, but there is something reasonable in such proposal. `lv_ovj_creae_with_styles`...

To be honest, IMO C api is not nice for comfortable programming due language limitations. I plan to create CPP wrapper for easy chained syntax & variants of signatures. May be such wrapper will be suitable for your case? I'd prefer to not bloat base C api.

I mean something like this:

```cpp
lv_obj_t obj = lv_obj_create(...);

// assign pre-defined classes
LvObjView(obj)
  	.styles(styles_list);

// assign inline class
LvObjView(obj)
  	.style(LvStyle()
		.property(WIDTH, 10)
		.property(HEIGHT, 20)
	);
```

amirgon, 2020-10-21:
> To be honest, IMO C api is not nice for comfortable programming due language limitations. I plan to create CPP wrapper for easy chained syntax & variants of signatures. May be such wrapper will be suitable for your case? I'd prefer to not bloat base C api.

@puzrin No I'm using Micropython, which is derived from C.
And I think that in general such functionality for assigning classes should be available in C as well.

puzrin, 2020-10-21:
Then the only choice is add alternate functions. I don't know is save of one line worth this or not. For my case - not principal.

kisvegabor, 2020-10-21:
> @kisvegabor How about providing "classes" as a constructor argument of every component?
If classes are modeled as bits, then the "classes" argument would be an integer built from ORed constants (or plus since there is no carry). That could be simple, readable and efficient (although limited to 32 classes).

I think for create + styling  the best is to write a function:
```c
lv_obj_t * my_btn_add(lv_obj_t *list )
{
  lv_obj_t * btn = lv_btn_create(list, NULL);
  lv_obj_add_style(btn, LV_BTN_PART_MAIN, &style1);
}
```

It's clean and flexible. If we added classes to create a function we could add size and position and some other parameters too. And finally ended up with a lot of parameters.

> In LVGL, is there a list somewhere of inherited properties?
The docs say "Some properties (typically that are related to texts) can be inherited" but I couldn't find a list of inherited/non-inherited properties anywhere.

Hm, it isn't but it really should be. Until it's added you can see [this list](https://github.com/lvgl/lvgl/blob/master/src/lv_core/lv_style.h#L103).

amirgon, 2020-10-21:
> I think for create + styling the best is to write a function:

@kisvegabor so, if I want to apply some style to all components in a screen (for example), what would be the best way?  
Call `lv_obj_add_style` for each and every object, as you suggest?  
Or do you still see value in Descendant Selectors?

kisvegabor, 2020-10-21:
> Or do you still see value in Descendant Selectors?

I'm not especially interested in descendent selectors but selectors in general. Ideally, selectors can be very abstract. Here are some examples I can imagine:
- descendant selectors, e.g. add a `red_style` to a window to make some parts to show danger (add red border to all buttons). Normally it can be complicated to style all elements individually.
- Create custom widget types e.g. by using `obj->user_data = MY_TITLE_TYPE` and change the style of all `MY_TITLE_TYPE` widget at once.
- Allow targeting cells of tables and buttons of button matrices. E.g. if the currently drawn row/col/button can get different styles can be applied to them.
- an interesting case:  "enable" a style only if the `current_rpm > 1000`. 

But it's a very difficult topic:
- selectors are powerful but could be slow and complicated to implement and manage
- [querySelectorAll](https://developer.mozilla.org/en-US/docs/Web/API/Document/querySelectorAll) as @puzrin suggested.  The problem with this it's not dynamic. I.e. if you create a new element you need to `querySelectorAll` again. But at least it's independent of the internals of LVGL. 
- simple make people create functions for custom elements. It's robust, not require changes in LVGL,  but not that convenient and limited.

puzrin, 2020-10-21:
Some kind of selector-like machinery may be useful. From the other hand, to avoid memory reallocation, this may be tree traverse function with callback. Then user will be able to do any checks. Returned value will say if traversal should continue, skip nesting, fail or end.

```
traverse(root, deepness, callback);
traverse_all(callback);
```

kisvegabor, 2020-10-22:
Let's discuss the implementation details in #1802  and use cases here.

kisvegabor, 2020-11-16:
It seems we found a good solution for this suggestion in #1802: extend and simplify themes to can be added to any object and apply these "local" themes from the parents too when creating object.

@amirgon Do you also find this approach suitable? If so I close this issue and the idea to the ROADMAP for v9.

amirgon, 2020-11-16:
@kisvegabor Could you explain or give and example to the new approach suggested in #1802?
Suppose I want to apply a style to a specific parent and all its descendants. Do I have to define a new theme for that? What about the "generic" selectors that allow traversing component trees?

kisvegabor, 2020-11-18:
Not all details are clear, but my idea is to allow adding a callback (`theme_cb`) to any object, and when a direct or indirect child is created this callback is called to allow any styling on the new object.

amirgon, 2020-11-18:
> Not all details are clear, but my idea is to allow adding a callback (`theme_cb`) to any object, and when a direct or indirect child is created this callback is called to allow any styling on the new object.

So I would need to register a callback to every object I would ever want to style that way? Isn't it a bit expensive?
Why not just provide a way to iterate all descendants and perform some operation on all of them?  
Why themes need to be involved?

kisvegabor, 2020-11-18:
> So I would need to register a callback to every object I would ever want to style that way? Isn't it a bit expensive?

Only on the descendant.  E.g on the list container to automatically style all new buttons

> Why not just provide a way to iterate all descendants and perform some operation on all of them?

It's the other idea. The callback works in creation time, the iterator works on already existing objects. If you think the iterator is enough, I'd happy to keep only it.

amirgon, 2020-11-18:
> It's the other idea. The callback works in creation time, the iterator works on already existing objects. If you think the iterator is enough, I'd happy to keep only it.

What do you think resembles CSS more, iterating component subtrees or invoking component callbacks?
The advantage of callbacks is that styles are applied on component creation. But why only on creation? 

Suppose I want to style differently all selected cells in an existing table.
I can see how to achieve this with the iterator idea, but not with the callback idea if callbacks are only applied during component creation.

kisvegabor, 2020-11-18:
In your example checked elements use the properties with checked state. So you only need to add style(s) for the checked state and let LVGL use them according to the state of the objects.

amirgon, 2020-11-18:
> In your example checked elements use the properties with checked state. So you only need to add style(s) for the checked state and let LVGL use them according to the state of the objects.

You are right, it's a bad example.
Let's go back to the example I [opened this issue with](https://github.com/lvgl/lvgl/issues/1832#issue-713897785-permalink:~:text=Usage%20examples%3A,style%20recursively%20where%20BIDI%20was%20applied.).

Suppose I want to apply font or bidi on an entire page. Would I now need to add a callback for each and every descendant in that page?

kisvegabor, 2020-11-18:
Sorry, I've noticed that I mistyped it. See the [corrected post](https://github.com/lvgl/lvgl/issues/1832#issuecomment-729537937).

So theme callback is meant to be added only to the ancestor. However, font is not really a good example either because it's inherited from the ancestor if not set on the child. If bidi is added as a style property it should be inherited too.

A real-life example I could imagine for the theme callback is settings it on a list container to style every new button in a given way.
It can't be solved by one extra line of `lv_obj_add_style` because the list element can have children (icon, label, etc) which also should be styled.  
So now to style a whole list button you need:
```c
lv_obj_add_style(list_btn, LV_BTN_PART_MAIN, &style1);
lv_obj_add_style(lv_list_get_icon(list_btn), LV_IMG_PART_MAIN, &style2);
lv_obj_add_style(lv_list_get_label(list_btn), LV_IMG_PART_MAIN, &style3);
```

Of course you can wrap it into an function like `add_styled_list_button()`. 

Honestly, I'm not fully convinced this is a really required feature, but if you find it useful it's a sign that something like this could be useful.

amirgon, 2020-11-18:
> Honestly, I'm not fully convinced this is a really required feature, but if you find it useful it's a sign that something like this could be useful.

Ok, thanks for the explanation.
I'm not sure it's really required either, once font and bidi are inherited.
Let's close this for now and reopen if the need arises.

closed 2020-11-18 by amirgon