lvgl/lvgl #1833: RTL lv_cont: child elements order is wrong

amirgon, 2020-10-03:
**Describe the bug**

When adding child elements to an RTL `lv_cont` with `PRETTY` style, elements are added in the wrong order  

[image]

**To Reproduce**

```python
scr = lv.obj()
scr.set_style_local_text_font(scr.PART.MAIN, lv.STATE.DEFAULT, lv.font_dejavu_16_persian_hebrew)
scr.set_base_dir(lv.BIDI_DIR.RTL)

cont = lv.cont(scr)
cont.set_layout(lv.LAYOUT.PRETTY_MID)
cont.set_width(scr.get_width())
l1 = lv.label(cont)
l1.set_text("ראשון 1")
l2 = lv.label(cont)
l2.set_text("שני 2")

lv.scr_load(scr)
```
**Expected behavior**

I would expect elements to be added in RTL order, such that "ראשון 1" would be on the right side and "שני 2" would be on the left side.

amirgon, 2020-10-03:
I've noticed a similar issue with RTL lv_table - the column order should be from right to left.  
Should I open a separate issue for that?

kisvegabor, 2020-10-05:
Hi,

Both issues are valid and can be fixed at first look. I can deal with this later next week.

kisvegabor, 2020-10-07:
I've fixed `lv_table`. 
The pretty layout will be fixed this week as well.

amirgon, 2020-10-08:
Looks good. Thank you!
Closing.

closed 2020-10-08 by amirgon