lvgl/lvgl #1836: Minor bug in lv_disp_get_inactive_time

xennex22, 2020-10-06:
https://github.com/lvgl/lvgl/blob/6b24c949f1d904ccbcf730849e3f3aedafc5c94f/src/lv_core/lv_disp.c#L317-L337

Code after ```if(disp) return lv_tick_elaps(disp->last_activity_time);``` is never executed.  I'm not sure of the original intent so are not going to propose a fix, plus it looks like it is designed to cope with multiple displays.

It's a good example of how multiple return statements in one function hinder readability and hide bugs.  The classic ```if(param == NULL) return;``` is usually not a good sign.

closed 2020-10-06 by kisvegabor

kisvegabor, 2020-10-06:
Hi, 

Thanks for reporting this.

Probably it was a copy-paste issue. The first lines are not required here because `disp` can be `NULL`. I've pushed a fix. 

I think the multiple vs single return statements question is an endless argument. :) I prefer single return point to avoid huge nested branches.

embeddedt, 2020-10-06:
My approach is usually to evaluate what the function is doing. If it's the type of function that does something, then always has to call a cleanup/unlock function before returning (e.g. something with mutexes) I try to stick to a single return point. Duplicated `mutex_unlock` statements everywhere are often bug-prone and messy. If someone ever added another branch they could easily forget to unlock the mutex.

In simple functions I think it matters less.

C47D, 2020-10-06:
Not an expert, but I would rather assert on NULL parameter when expecting it.

commit fff3a00d (2020-10-06): fix lv_disp_get_inactive_time if the disp==NULL
Fixes #1836