lvgl/lvgl #1837: lv_color1_t is incompatible with C++ versions before C++20

embeddedt, 2020-10-06:
**Describe the bug**

Designated initializers were not supported by the C++ standard before C++20. The following macro in `lv_color.h` depends on a designated initializer:

https://github.com/lvgl/lvgl/blob/fff3a00de241493cb80b260bed18cd622e809644/src/lv_misc/lv_color.h#L629

**To Reproduce**

Attempt to compile LVGL in 1bpp color mode.

**Expected behavior**

Compilation should succeed without needing to use C++20 mode.

**Additional context**

Originally reported here: https://forum.lvgl.io/t/use-lvgl-with-c/3410/3?u=embeddedt

Because `lv_color1_t` is a union I'm not sure how we would go about fixing this. Versions of C++ before C++20 seem to use C89 initializer semantics, meaning that we can only initialize the first member of a union.

kisvegabor, 2020-10-08:
I can imagine only this ugly workaround
```c
#define LV_COLOR_MAKE(r8, g8, b8) (_LV_COLOR_MAKE_TYPE_HELPER{ \
                                          (uint8_t)((b8 >> 7) | (g8 >> 7) | (r8 >> 7)), \
                                          (uint8_t)((b8 >> 7) | (g8 >> 7) | (r8 >> 7)), \
                                          (uint8_t)((b8 >> 7) | (g8 >> 7) | (r8 >> 7)))
```

What do you think?

embeddedt, 2020-10-08:
Does this compile? (I haven't checked.) Usually you can only initialize the first member of a union (that might be good enough for our purposes).

kisvegabor, 2020-10-08:
You are right, my code gave a warning.
It works:
```c
#define LV_COLOR_MAKE(r8, g8, b8) (_LV_COLOR_MAKE_TYPE_HELPER {{(b8 >> 7) | (g8 >> 7) | (r8 >> 7)}})
```

embeddedt, 2020-10-08:
Should we go ahead and apply this fix to `master` (since the previous syntax was technically wrong)?

kisvegabor, 2020-10-08:
Yes!

commit 6e48cde2 (2020-10-08): Fix #1837: remove last designated initializer in `lv_color.h`
PR #1843 (2020-10-09): Improve compatibility with C++ 14 
Here two small fix:
1) MSVC 2019 compiler error "C2397 conversion from 'int' to 'uint8_t' requires a narrowing conversion" introduced by Fix #1837
2) Remove last designated initializers in lv_color_mix_with_alpha

This fix tested with Microsoft Visual Studio 2019 16.7.5(C++14, 17, 20), IAR 8.50.6, Keil v5.31(ARM Compiler V6)
closed 2020-10-08 by embeddedt