lvgl/lvgl #1838: LVGL is not drawing the whole screen

vkatrenikona, 2020-10-06:
Greetings,
I tried creating a widget but I found the screen is not drawn properly. Below is the screenshot of my output
[image]
` /*-------------------------
     * Initialize your display
     * -----------------------*/
    lv_init();

    /* -----------------------------------------------
     * Display peripheral of the Board(Beaglebone Black)
     * -------------------------------------------------*/
    BBB_Board_LCDCInit();
    BBB_Board_LCDCConfigure();

    /*-----------------------------
     * Create a buffer for drawing
     *----------------------------*/
    /** Initialize the display buffer */
    lv_disp_buf_init(&lv_disp_buf_var, SrcPixelBuffer, NULL, TOTAL_PIXELS);

    /*-----------------------------------
     * Register the display in LVGL
     *----------------------------------*/

    /** Initialize the display driver */
    lv_disp_drv_init(&disp_drv);
    /** Configure the display driver */
    disp_drv.hor_res = PIXELS_PER_LINE;
    disp_drv.ver_res = LINES_PER_PANNEL;
    disp_drv.flush_cb = disp_flush;
    disp_drv.buffer = &lv_disp_buf_var;

    /** Register the display for LVGL */
    lv_disp_handle = lv_disp_drv_register(&disp_drv);
    if(lv_disp_handle == NULL)
    {
        DEBUG_ERR("Display driver register failed", __LINE__, __FILE__);
    }
    par = lv_obj_create(lv_scr_act(), NULL); /*Create a parent object on the current screen*/
    lv_obj_set_size(par, 640, 360);                     /*Set the size of the parent*/

    btn = lv_btn_create(par, NULL);     /*Add a button the current screen*/
    lv_obj_set_pos(btn, 10, 10);                   /*Set its position*/
    lv_obj_set_size(btn, 120, 50);                          /*Set its size*/
    lv_obj_set_event_cb(btn, btn_event_cb);                 /*Assign a callback to the button*/

    label = lv_label_create(btn, NULL);          /*Add a label to the button*/
    lv_label_set_text(label, "Button");                     /*Set the labels text*/
`
As you see close you can find there is partition of screen. If I try to move the widget down it looks like after 90 lines down the widget is being cut. 
[image]
` /*-------------------------
     * Initialize your display
     * -----------------------*/
    lv_init();

    /* -----------------------------------------------
     * Display peripheral of the Board(Beaglebone Black)
     * -------------------------------------------------*/
    BBB_Board_LCDCInit();
    BBB_Board_LCDCConfigure();

    /*-----------------------------
     * Create a buffer for drawing
     *----------------------------*/
    /** Initialize the display buffer */
    lv_disp_buf_init(&lv_disp_buf_var, SrcPixelBuffer, NULL, TOTAL_PIXELS);

    /*-----------------------------------
     * Register the display in LVGL
     *----------------------------------*/

    /** Initialize the display driver */
    lv_disp_drv_init(&disp_drv);
    /** Configure the display driver */
    disp_drv.hor_res = PIXELS_PER_LINE;
    disp_drv.ver_res = LINES_PER_PANNEL;
    disp_drv.flush_cb = disp_flush;
    disp_drv.buffer = &lv_disp_buf_var;

    /** Register the display for LVGL */
    lv_disp_handle = lv_disp_drv_register(&disp_drv);
    if(lv_disp_handle == NULL)
    {
        DEBUG_ERR("Display driver register failed", __LINE__, __FILE__);
    }
    par = lv_obj_create(lv_scr_act(), NULL); /*Create a parent object on the current screen*/
    lv_obj_set_size(par, 640, 360);                     /*Set the size of the parent*/

    btn = lv_btn_create(par, NULL);     /*Add a button the current screen*/
    lv_obj_set_pos(btn, 520, 42);                   /*Set its position*/
    lv_obj_set_size(btn, 120, 50);                          /*Set its size*/
    lv_obj_set_event_cb(btn, btn_event_cb);                 /*Assign a callback to the button*/

    label = lv_label_create(btn, NULL);          /*Add a label to the button*/
    lv_label_set_text(label, "Button");                     /*Set the labels text*/
`
**My buffers** 
/** Display buffer that holds the pixel data of the screen
 *  @Note: For internal use of LVGL library
 **/
lv_color_t SrcPixelBuffer[640*360];
/** Display buffer that holds the pixel data of the screen
 *  @Note: For internal use of LCDC library
 **/
UINT32 DestPixelBuffer[640*360];
My callback function
`    int32_t x;
    int32_t y;
    int32_t in =0;
    for( y = area->y1; y<=area->y2; y++)
    {
        for(x = area->x1; x<=area->x2; x++)
        {
/** Making the alfha byte to 0 for my application purpose */
            DestPixelBuffer[in] = (UINT32)(SrcPixelBuffer[in].full & 0x00FFFFFF);
            color_p++;
            in++;
        }
    }
    lv_disp_flush_ready(disp_drv);`
I am running out of ideas. Kindly let me know what might go wrong in following implementation. Thank you in advance

kisvegabor, 2020-10-08:
Hi,

It seems the issue is here:
```
`    int32_t x;
    int32_t y;
    int32_t in =0;
    for( y = area->y1; y<=area->y2; y++)
    {
        for(x = area->x1; x<=area->x2; x++)
        {
            DestPixelBuffer[in] = (UINT32)(SrcPixelBuffer[in].full & 0x00FFFFFF);
            color_p++;
            in++;
        }
    }
```

You always start to write to the beginning of `DestPixelBuffer` however you should write to the area pointed by the `area` paramter.

Take a look at this: https://github.com/lvgl/lv_drivers/blob/fed54c8d694eb24ff1a3ac78ad42a5840b3bb904/display/monitor.c#L141-L145

vkatrenikona, 2020-10-08:
> Hi,
> 
> It seems the issue is here:
> 
> ```c
>     int32_t x;
>     int32_t y;
>     int32_t in =0;
>     for( y = area->y1; y<=area->y2; y++)
>     {
>         for(x = area->x1; x<=area->x2; x++)
>         {
>             DestPixelBuffer[in] = (UINT32)(SrcPixelBuffer[in].full & 0x00FFFFFF);
>             color_p++;
>             in++;
>         }
>     }
> ```
> 
> You always start to write to the beginning of `DestPixelBuffer` however you should write to the area pointed by the `area` paramter.
> 
> Take a look at this: https://github.com/lvgl/lv_drivers/blob/fed54c8d694eb24ff1a3ac78ad42a5840b3bb904/display/monitor.c#L141-L145

Yes It worked. Thank you Gabor

kisvegabor, 2020-10-08:
You are welcome!

closed 2020-10-08 by kisvegabor