lvgl/lvgl #1839: watchdog and lv_task_handler

ali-rostami, 2020-10-07:
**Describe the bug**
LVGL version: v7.6
I have a watchdog task that is responsible for checking the state of the other tasks every 10 seconds. If there is one with not AWAKE state, log and let the watchdog reset the processor.
The problem I have is that whenever I scroll the `page` continuously and fast the reset will be occurred. in other words, in this situation for more than 10 seconds, it will not get out of `lv_task_handler` and so the watchdog takes it as a dead task and reset the processor.
I think it's not normal and seems a bug that it stuck in `lv_task_handler`. If my assumption is wrong, let me know to continue [here](https://forum.lvgl.io/t/chart-updating-lag/3319) which seems related also.

kisvegabor, 2020-10-07:
Actually, it's the intended behavior. Let's say it takes 100 ms to refresh the screen. In this case, the next refresh will start immediately.  

I suggest creating an `lv_task` too with `LV_TASK_HIGHEST` and reset watchdog there as well.

ali-rostami, 2020-10-07:
Oh, didn't know. what about the other higher priority tasks-higher than GUI task-, which are blocked because of the token semaphore by gui_task.
They won't get run until the GUI task release it. And the GUI task will not release the semaphore until `lv_task_handler` is returned.

embeddedt, 2020-10-07:
@kisvegabor I think we could just force a return from `lv_task_handler` after the task list has been processed once. Since the allowed sleeping time is now returned from there, it would simply return 0 if another refresh needed to occur immediately. I haven't measured but I doubt there's much overhead from entering and exiting that function.

kisvegabor, 2020-10-08:
@embeddedt There is not much overhead to reenter `lv_task_handler` but the user needs to take care not sleep e.g. 5ms before calling lv_task_handler again.

So it'd look something like this? (Probably there is better pattern though)
```c
while(1) {
 uint32_t time_to_next = lv_task_handler();
 if(time_to_next == 0) continue;
 
 sleep(time_to_next); //or sleep(5)
}

embeddedt, 2020-10-08:
Yes; exactly.

kisvegabor, 2020-10-08:
Alright. We need to find the right pattern (`while`, `continue`, `if`?) for it as it will be promoted to every user in examples. 

What would be policy then? Now when a lower priority task runs the higher priority task needs to run again. 
My initial guess:
- Run all highest prio and  return
- If there are some ready highest again, run them again and return
- If no highest run the first ready high and return (to allow highest to run again)
- If no highest and high go to mid
- etc...

Now the linked list of tasks is ordered according to priority but it's not really required because we can simply read the list and know which prio we are checking now.

What do you think?

embeddedt, 2020-10-08:
I think your policy would work.

> Now the linked list of tasks is ordered according to priority but it's not really required because we can simply read the list and know which prio we are checking now.

Performance-wise, I think the ordering has an advantage. Without ordering, the tasks could potentially be laid out like this:

```
HIGHEST
LOW
MEDIUM
HIGHEST
LOW
```

With ordering, they are instead laid out this way:

```
HIGHEST
HIGHEST
MEDIUM
LOW
LOW
```

The ordering means that the first call to `lv_task_handler` can return as soon as it reaches the medium-priority task (since there would never be a higher-priority task after it).

Without ordering the whole list would have to be scanned on every single call.

ali-rostami, 2020-10-08:
Hmm, Now the problem is that the other tasks which are dependent on the return of ` lv_task_handler` in GUI task might starve. As we saw in my case when I kept touching and dragging the page I had no return from `lv_task_handler` and so other tasks which were dependent to GUI task couldn't run. 
Does @kisvegabor 's solution help? I would appreciate you explain more so I can understand 🙏

ali-rostami, 2020-10-08:
> `lv_task_handler` will now return every time a set of tasks is processed, so the display will never be refreshed more than once in a single `lv_task_handler` call.

But we have `continue` when the time is 0. So, what would be the difference then?

embeddedt, 2020-10-08:
The difference is that you now know when LVGL is willing to "take a break" from rendering, so you can unlock a mutex/notify your watchdog that the task is still running.

kisvegabor, 2020-10-09:
> The difference is that you now know when LVGL is willing to "take a break" from rendering, so you can unlock a mutex/notify your watchdog that the task is still running.

Yes, that's it. The example with `continue` is to mimic the current behavior. But now would have more control.

> Without ordering the whole list would have to be scanned on every single call.

I think it's not a big deal as there are only a few tasks (max 10-20?). Dropping it might increase readability and simplify the code. Anyway, once, the ordering already works we can keep it.

kisvegabor, 2020-11-02:
After some days have passed do you still think it's any we should work on?

ali-rostami, 2020-11-02:
Not sure about others and any other situations. But for this issue, my problem is solved. 
As you might remember, I don't update the GUI in other tasks. -especially higher priority tasks-
I just save the changes the GUI needs and then in a **lv_task** check them and update the GUI if it's necessary.
so other tasks-especially higher priority tasks- should not wait for the lvgl task.

nitz, 2020-11-02:
I personally feel that there should be a built in (even if optional) mechanism to prevent infinite blocking. Something along the lines of "max tasks handled per handler call" or "max time allowed to handle tasks". Though, I definitely can understand why dev time would be better spent elsewhere.

That said, a `LV_TASK_HIGHEST` is an acceptable workaround for my use case with the watchdog on my platform. Perhaps even just documenting that `lv_task_handler()` may block a long time (or infinitely), with adding a highest priority task as a workaround to get a chance to "breath" may be enough for now?

embeddedt, 2020-11-02:
I agree. We should ensure that `lv_task_handler` does not block forever without returning at some point, otherwise one has to abuse `LV_TASK_HIGHEST` to "escape".

I like the idea of returning from `lv_task_handler` after a certain amount of time has passed; this could be made configurable in `lv_conf.h`. I think it would be very simple to implement.

kisvegabor, 2020-11-03:
> I like the idea of returning from lv_task_handler after a certain amount of time has passed; this could be made configurable in lv_conf.h. I think it would be very simple to implement.

What do you think, what could be a good default value for it? 100-200 ms? Or disabled by default?

I'm asking because if disp. refresh time is 50 FPS, a typical `sleep(5)` after `lv_task_handler` delays the next refresh with 5 ms and reduce the FPS from 20 to 18. 
So we should either:
1. Disable the timeout by default
2. Accept the performance loss
3. Promot different `lv_task_handler` calling mechanism. E.g. [this](https://github.com/lvgl/lvgl/issues/1839#issuecomment-705440508)

embeddedt, 2020-11-03:
Let's disable it by default, since it doesn't affect most users. 100 ms should be a fast enough default and users can always tweak it for their specific needs.

nitz, 2020-11-03:
I very much agree with disabled by default, and even think it'd be worth noting in comments around the setting in `lv_conf.h` that by definition it will reduce performance somewhat, as that's it's intent.

kisvegabor, 2020-11-03:
Alright, I've just checked it and it seems super easy to implement. 
Let's wait for a little while the Kconfig file is merged to make it easier to keep track of the new option.

puzrin, 2020-11-03:
IMO you do strange things, trying to use timer handlers as tasks. Use protothreads if you need yield heavy calculations. Then you will be ok with round-robin for the rest.

kisvegabor, 2020-11-04:
[prtonthread](http://www.dunkels.com/adam/pt/) says:
> Provides blocking wait without full multi-threading or stack-switching

So it seems it's even worse for us because one blocking wait can stop all other treads, isn't it?

embeddedt, 2020-11-04:
I think it means that you can yield within a thread while you are waiting for something (see `PT_WAIT_UNTIL` on that page). Otherwise the library wouldn't be very useful.

puzrin, 2020-11-04:
Protothreads allow "non-blocking implementaion of blocking-like wait". All you need is call each "thread" with high frequency, and YEILD or WAIT_FOR_CONDITION inside (macros).

You will have cooperative activity anyway (current "tasks" are cooperative too). But difference is in more natural algorithms expression.

See https://github.com/speedcontrols/ac_sc_grinder/blob/a17364d99fe81bf8d1e1b73df8a4a1f81e209e20/src/calibrator/calibrator_pid.h#L64-L435 for example - very long logic looks almost natural, thanks to yield-ing.

For sources, see this repos first:

- https://github.com/Akagi201/protothreads
- https://github.com/LarryRuane/protothread
- https://github.com/benhoyt/protothreads-cpp

---

I mean , attempts to enhance cooperative timers with priorities and cpu usage limit look very suspicious (use timers wrong way). Not because impossible, but because result will never reach simplicity of alternate approaches.

Leave periodic timers for small periodic things. There are demands, expessed via timers best way. And route the other demands to protothreads. I would probably advice to remove timer priorities - illusion of corectness.

kisvegabor, 2020-11-05:
I took a closer look to protonthread and in its [article](http://dunkels.com/adam/dunkels06protothreads.pdf), it shows that it uses `__label__` type (Figure 14) which is a [GCC extension](https://gcc.gnu.org/onlinedocs/gcc/Local-Labels.html). 

So we can use protonthread only with GCC?

puzrin, 2020-11-05:
> I took a closer look to protothread and in its [article](http://dunkels.com/adam/dunkels06protothreads.pdf), it shows that it uses `__label__` type (Figure 14) which is a [GCC extension](https://gcc.gnu.org/onlinedocs/gcc/Local-Labels.html).
> 
> So we can use protothread only with GCC?

No. Protothreads can be used with any C99 compiler. The difference is only in % of madness in generated sources. I suggest you inspect my links instead, as more actual info.

"context switch" can be implemented via:

- "switch/while" macros, emulation for "goto variable" - 100% compatible, but real black magic. Will look ugly in debugger.
- "goto variable" - ideal, if compiler supports it (gcc supports).
- setjmp - better compatibility than "goto variable", but slow. May be good mode for debug if "goto variable" not available.

kisvegabor, 2020-11-06:
> No. Protothreads can be used with any C99 compiler. The difference is only in % of madness in generated sources. I suggest you inspect my links instead, as more actual info.

Got it, thanks.

> Prior to land protothreads into LVGL as public task api, recheck, if this should be part or LVGL or separate lib

I'm also thinking about it. For LVGL timers are sufficient and more complex stuff should be independent of LVGL.

> Cleanup timers anyway. Naming those as "tasks" is very confusing. IMO that should be just a periodic timers.

I agree that the name "timer" would be better. Maybe the whole priorities and other complexities can be removed and make this module really just like a simple timer.

kisvegabor, 2020-12-07:
In v8 (`dev` branch) `lv_task` is rewritten to `lv_timer`. It simply runs once so no blocking should occur. 

As it should solve the issue I close this now.

closed 2020-12-07 by kisvegabor

JF002, 2022-03-21:
Hi! First of all : sorry to bring up this old issue :-) 

We are still using LVGL7 in [InfiniTime](https://github.com/InfiniTimeOrg/InfiniTime) and we are currently experiencing this issue : `lv_task_handler` enters into an infinite loop when refreshing the whole display takes too much time. It mostly happens in one app that scrolls 2 labels, and when the strings in those labels are quite long.

We implemented a workaround to prevent the issue by limiting the size of the string.

I also tried to find a fix in LVGL that would prevent this issue from happening in the future (until we integrate LVGL8), and I was wondering if you could provide me with some feedback regarding the solution I'm currently testing. It consists in counting the number of tasks that are waiting for execution at the beginning of `lv_task_handler`. Then, the loop will execute only this specific number of tasks before returning. Other tasks that might have been added in the queue will be executed during the next call to `lv_task_handler`.

[image]

```
Index: src/lv_misc/lv_task.c
IDEA additional info:
Subsystem: com.intellij.openapi.diff.impl.patch.CharsetEP
<+>UTF-8
===================================================================
diff --git a/src/lv_misc/lv_task.c b/src/lv_misc/lv_task.c
--- a/src/lv_misc/lv_task.c	(revision 23430cf20e32294549fff9b2879a9466dacc19bb)
+++ b/src/lv_misc/lv_task.c	(date 1647285586769)
@@ -81,6 +81,9 @@
 
     uint32_t handler_start = lv_tick_get();
 
+    uint32_t size = _lv_ll_get_len(&LV_GC_ROOT(_lv_task_ll));
+    uint32_t count = 0;
+
     /* Run all task from the highest to the lowest priority
      * If a lower priority task is executed check task again from the highest priority
      * but on the priority of executed tasks don't run tasks before the executed*/
@@ -153,6 +156,12 @@
             }
 
             LV_GC_ROOT(_lv_task_act) = next; /*Load the next task*/
+
+           if(count >= size) {
+             end_flag=true;
+             LV_GC_ROOT(_lv_task_act) = NULL;
+           }
+            count++;
         }
     } while(!end_flag);
```

What do you think of this fix? Do you see any drawback or negative side effect it could trigger?

Thanks for your help, and thanks for your work on LVGL !

mircopz, 2022-12-07:
> In v8 (`dev` branch) `lv_task` is rewritten to `lv_timer`. It simply runs once so no blocking should occur.
> 
> As it should solve the issue I close this now.

Hi @kisvegabor 
Since we are still using v7.11, I tried to replicate the behavior of v8 lv_timer in my lv_task_handler.
To avoid any blocking, I edited the run loop as follows:

```c
    /* Run all task from the list*/
    lv_task_t * next;
    do {
        task_deleted             = false;
        task_created             = false;
        LV_GC_ROOT(_lv_task_act) = _lv_ll_get_head(&LV_GC_ROOT(_lv_task_ll));
        while(LV_GC_ROOT(_lv_task_act)) {
            /* The task might be deleted if it runs only once ('count = 1')
             * So get next element until the current is surely valid */
            next = _lv_ll_get_next(&LV_GC_ROOT(_lv_task_ll), LV_GC_ROOT(_lv_task_act));

            if(lv_task_exec(LV_GC_ROOT(_lv_task_act))) {
                /* If a task was created or deleted then this or the next item might be corrupted*/
                if(task_created || task_deleted) {
                    LV_LOG_TRACE("Start from the first task again because a task was created or deleted");
                    break;
                }
            }

            LV_GC_ROOT(_lv_task_act) = next; /*Load the next task*/
        }
    } while(LV_GC_ROOT(_lv_task_act));
```

It is identical to v8 code (replacing timer with task), and in this way all tasks are called only once. Can it work or can there be negative side effects?
Thank you

kisvegabor, 2022-12-11:
It can work, but v7 uses task priorities and they will be ignore with these changes.