lvgl/lvgl #1841: LV_USE_GPU_STM32_DMA2D, gpu_wait, and, interrupts

upbeat27, 2020-10-07:
**Describe the bug**

A couple things that seem like bugs, or at least missing features when using LV_USE_GPU_STM32_DMA2D.

The first is `dma2d_wait`. This function calls `wait_cb` instead of `gpu_wait_cb`, which is confusing. If this is intentional, it should be more clear.

As for `dma2d_wait`, it kind of "overrides" the purpose of `gpu_wait_cb` called elsewhere. If you are waiting right after using DMA2D, you don't need to wait before using it.

There is no way to enable the DMA2D interrupt.
We can't use the cache callback to do it, since DMA2D->CR is written right after.
We can't use the wait callback, since the interrupt should be enabled before the transfer is started.

**To Reproduce**

N/A

**Expected behavior**

There should be some way to enable the DMA2D interrupt.
Useage of `wait_cb` vs `gpu_wait_cb` should be more well defined.

**Screenshots**
N/A

**Additional context**

The interrupt is useful in the context of an OS. I use a semaphore to control access to DMA2D. The interrupt is necessary to release the semaphore.

kisvegabor, 2020-10-12:
Hi,

I think interrupts are not required only `dma2d_wait` needs to be replaced with `gpu_wait_cb`. This way LVGL keeps running until it hits a point where GPU needs to be ready. Here it calls `gpu_wait_cb` to wait for the running GPU operation. 

I've pushed an update to the `dev` branch. You can test it by setting `lv_gpu_stm32_dma2d_wait_cb` as `gpu_wait_cb`.

upbeat27, 2020-10-14:
I think I am just confused about the different callbacks and the places in which they are called.
Now I actually get the best performance if I set `gpu_wait_cb` to a blank function (so `wait_finish` does nothing), and instead wait in `clean_dcache_cb`. This case is without using interrupts or semaphores.

kisvegabor, 2020-10-16:
> I think I am just confused about the different callbacks and the places in which they are called.

So in summary:
- `wait_cb` get called periodically when LVGL waits for something. (right now it means only waiting fo flush ready or gpu). It allows the user to do something while LVGL is idle, e.g. check an I/O, or yield the task etc.
- `gpu_wait_cb` before drawing something LVGL needs to be sure that all previous drawing is finished. With SW drawing it's obvious but the GPU works in the background. So LVGL calls this function to wait until GPU is finished before drawing anything new.
-  `clean_dcache_cb` you need to be sure the cache is written to RAM before flushing. LVGL never calls this function on its own but you can call it with `lv_disp_clean_dcache` when required.  Note that, it might be used in the drivers. 

> wait in clean_dcache_cb. This case is without using interrupts or semaphores.

I think you are just lucky. There are a lot of GPU calls before hitting `clean_dcache_cb` and you should wait for them.

upbeat27, 2020-10-16:
I guess what I mean is, the confusing part is that `wait_cb` is called in the GPU functions, not `gpu_wait_cb`, which is actually called elsewhere.

I don't think it is luck. If you want for the GPU to be finished before using it, it is the same as waiting for it to be finished after using it. The former allows non-GPU code execution until the next GPU call.

kisvegabor, 2020-10-19:
> I don't think it is luck. If you want for the GPU to be finished before using it, it is the same as waiting for it to be finished after using it. The former allows non-GPU code execution until the next GPU call.

Let's say there are 10 GPU calls before finally `disp_flush` is called. You need to wait for all of them to be finished. However, `clean_dcache_cb` needs to be called only once before flushing. (At least as far as I know). Hence if you wait in `clean_dcache_cb` for GPU to be finished you wait only for the 10th GPU call.

upbeat27, 2020-10-19:
With `LV_USE_GPU_STM32_DMA2D`, cache is cleaned and invalidated before each GPU call, and each GPU call waits for the DMA2D operation to be complete. Perhaps the STM32 DMA2D stuff needs some more attention?
BTW is there any specific reason the STM32 DMA2D doesn't just hook into the existing GPU function calls when enabled, but instead uses separate functions with different arguments? I know blending is a special case because of the blend buffer, but I don't know about the rest...

embeddedt, 2020-10-19:
Mainly speed. We found that you can squeeze more performance out of these embedded accelerators in special cases when you are not restricted by a certain callback convention.

kisvegabor, 2020-10-21:
> With LV_USE_GPU_STM32_DMA2D, cache is cleaned and invalidated before each GPU call, and each GPU call waits for the DMA2D operation to be complete. Perhaps the STM32 DMA2D stuff needs some more attention?

Ah, you are right. But note that it's still "dangerous". If GPU is running in the background and meanwhile SW rendering is also working on the same memory artifacts will appear. Therefore before starting SW rendering any pending GPU operation needs to be finished. 
It can be improved though by checking the last working area of GPU and the current area to draw. The GPU needs to be finished only if the two areas intersect.

closed 2020-11-26 by stale[bot]