lvgl/lvgl #1842: RTL Dropdown selection problem

amirgon, 2020-10-08:
**Describe the bug**

On RTL dropdown list, element is correctly selected only when clicking on the text label.
When selecting the empty space of an option, the dropdown list closes, but the wrong element is selected.  

Here is an example trying to select "item 1", but item 2 keeps being selected until clicking on the "item 1" text label.

[image]

**To Reproduce**

```python
scr = lv.obj()
scr.set_style_local_text_font(scr.PART.MAIN, lv.STATE.DEFAULT, lv.font_dejavu_16_persian_hebrew)
scr.set_base_dir(lv.BIDI_DIR.RTL)

def dd_cb(obj, e):
    if e == lv.EVENT.VALUE_CHANGED:
        print("Selected " + str(obj.get_selected()))

dd = lv.dropdown(scr)
dd.set_width(400)
dd.set_options("item 1\nitem 2")
dd.set_event_cb(dd_cb)

lv.scr_load(scr)
```

**Expected behavior**

Should clicking on any point on the list should change the selection to that option.

kisvegabor, 2020-10-12:
Hi,

I've pushed a fix for it.

amirgon, 2020-10-12:
Thanks!

Closing.

closed 2020-10-12 by amirgon