lvgl/lvgl #1844: Something like signal&slots from Qt

DjamesSuhanko, 2020-10-09:
Hello!
In first, thanks for this Amazing library.
My doubt is about emit signals, like I do in Qt. is There something like that? I'm asking because I can get touch event with TFT_eSPI, but I'm not sure if I'm doing some mistake in callback implementation,  because in this one I got no events. If I can send a signal from getTouch(x,y) used by TFT_eSPI, then callback could really receive LV_EVENT_CLICKED, for example. 
Sorry if I'm writing a stupid question (and probably I'm), but I'm thinking something like that:
lv_emit_signal(btn1,LV_EVENT_CLICKED);

Thank you and sorry my terrible english.

embeddedt, 2020-10-09:
You shouldn't need to send a signal directly from the touch driver; LVGL does this for you. All you should need to do is register a callback on the button for `LV_EVENT_CLICKED` and it will get called when you click the button.

Could you share some code?

DjamesSuhanko, 2020-10-09:
Hello! Thanks to answer. I'm 3 days trying that. Three minutes ago I could implement a functional callback! \o/
I'm not sure if it's the right way, but now it's works! I'm reading documentation, examples and so on. I'm using PlatformIO in VSCode with Arduino platform. My board is a ESP32.
Can you say if I'm getting the right way now? My actual code:
`#include <lvgl.h>
#include <TFT_eSPI.h>
#include <stdio.h>
TFT_eSPI tft = TFT_eSPI(); /* TFT instance */
static lv_disp_buf_t disp_buf;
static lv_color_t buf[LV_HOR_RES_MAX * 10];

uint16_t t_x = 0, t_y = 0;

#if USE_LV_LOG != 0
/* Serial debugging */
void my_print(lv_log_level_t level, const char * file, uint32_t line, const char * dsc)
{

    Serial.printf("%s@%d->%s\r\n", file, line, dsc);
    Serial.flush();
}
#endif

static void event_handler(lv_obj_t * obj, lv_event_t event)
{
    if(event == LV_EVENT_CLICKED) {
        printf("Clicked\n");
        for (uint8_t i=0;i<10;i++){
        Serial.println("chucrutes");
        }
    }
    else if(event == LV_EVENT_VALUE_CHANGED) {
        printf("Toggled\n");
    }
}

void lv_ex_btn_1(void)
{
    lv_obj_t * label;

    lv_obj_t * btn1 = lv_btn_create(lv_scr_act(), NULL);

    lv_obj_set_event_cb(btn1, event_handler);
    lv_obj_align(btn1, NULL, LV_ALIGN_CENTER, 0, -40);

    label = lv_label_create(btn1, NULL);
    lv_label_set_text(label, "Button");

    lv_obj_t * btn2 = lv_btn_create(lv_scr_act(), NULL);
    lv_obj_set_event_cb(btn2, event_handler);
    lv_obj_align(btn2, NULL, LV_ALIGN_CENTER, 0, 40);
    lv_btn_set_checkable(btn2, true);
    lv_btn_toggle(btn2);
    lv_btn_set_fit2(btn2, LV_FIT_NONE, LV_FIT_TIGHT);

    label = lv_label_create(btn2, NULL);
    lv_label_set_text(label, "Toggled");

}

/* Display flushing */
void my_disp_flush(lv_disp_drv_t *disp, const lv_area_t *area, lv_color_t *color_p)
{
    uint32_t w = (area->x2 - area->x1 + 1);
    uint32_t h = (area->y2 - area->y1 + 1);

    tft.startWrite();
    tft.setAddrWindow(area->x1, area->y1, w, h);
    tft.pushColors(&color_p->full, w * h, true);
    tft.endWrite();

    lv_disp_flush_ready(disp);
}

/* Reading input device (simulated encoder here) */
bool read_encoder(lv_indev_drv_t * indev, lv_indev_data_t * data)
{
    static int32_t last_diff = 0;
    int32_t diff = 0; /* Dummy - no movement */
    int btn_state = LV_INDEV_STATE_REL; /* Dummy - no press */

    data->enc_diff = diff - last_diff;;
    data->state = btn_state;

    last_diff = diff;

    return false;
}

bool my_input_read(lv_indev_drv_t * drv, lv_indev_data_t*data)
{
    data->point.x = t_x;
    data->point.y = t_y;
    data->state = LV_INDEV_STATE_PR or LV_INDEV_STATE_REL;
    return false; /*No buffering now so no more data read*/
}

void setup()
{

    Serial.begin(9600); /* prepare for possible serial debug */

    lv_init();

#if USE_LV_LOG != 0
    lv_log_register_print_cb(my_print); /* register print function for debugging */
#endif

    uint16_t calData[5] = {331, 3490, 384, 3477, 6};

    tft.begin(); /* TFT init */
    tft.setRotation(0); /* Landscape orientation */

    tft.setTouch(calData);

    lv_disp_buf_init(&disp_buf, buf, NULL, LV_HOR_RES_MAX * 10);

    /*Initialize the display*/
    lv_disp_drv_t disp_drv;
    lv_disp_drv_init(&disp_drv);
    disp_drv.hor_res = 240;
    disp_drv.ver_res = 320;
    disp_drv.flush_cb = my_disp_flush;
    disp_drv.buffer = &disp_buf;
    lv_disp_drv_register(&disp_drv);

    /*Initialize the (dummy) input device driver*/
    lv_indev_drv_t indev_drv;
    lv_indev_drv_init(&indev_drv);
    indev_drv.type = LV_INDEV_TYPE_POINTER;
    indev_drv.read_cb = my_input_read;
    lv_indev_drv_register(&indev_drv);

    /* Create simple label */
    lv_obj_t *label = lv_label_create(lv_scr_act(), NULL);
    lv_label_set_text(label, "Hello Arduino! (V7.0.X)");
    lv_obj_align(label, NULL, LV_ALIGN_CENTER, 0, 0);

    lv_ex_btn_1();
}

void loop()
{
    
    if (tft.getTouch(&t_x, &t_y)){
        Serial.println(t_x);

        lv_signal_t mySignal = LV_EVENT_CLICKED;
        //lv_signal_send(btn1,mySignal,NULL);
        
    }
    lv_task_handler(); /* let the GUI do its work */
    delay(5);
}`

embeddedt, 2020-10-09:
Mostly correct. You should probably call `tft.getTouch` inside `my_input_read`, and return `LV_INDEV_STATE_PR` or `LV_INDEV_STATE_REL` based on whether its return value is true or false.

DjamesSuhanko, 2020-10-09:
Wow, good!!
Thanks a lot for this tip! Closing issue, very glad!

closed 2020-10-09 by DjamesSuhanko