lvgl/lvgl #1846: Build fails when LV_USE_GPU_NXP_PXP is enabled

jupeos, 2020-10-09:
Hopefully you have a CI system that will inform you of this but if not the v7.7.0 fails to build with `LV_USE_GPU_NXP_PXP` enabled.

Looks like some version control text was included in `lv_draw_blend.c`...

```
#if LV_USE_GPU_NXP_PXP
<<<<<<< HEAD
        if (lv_area_get_size(draw_area) >= GPU_NXP_PXP_BLIT_SIZE_LIMIT) {
            lv_gpu_nxp_pxp_blit(disp_buf_first, disp_w, map_buf_first, map_w, draw_area_w, draw_area_h, opa);
            return;
        }
#elif (LV_USE_GPU_NXP_VG_LITE)
            if(lv_area_get_size(draw_area) >= LV_GPU_NXP_VG_LITE_BLIT_SIZE_LIMIT) {

                lv_gpu_nxp_vglite_blit_info_t blit;

                blit.src = map_buf;
                blit.src_width = draw_area_w;
                blit.src_height = draw_area_h;
                blit.src_stride = lv_area_get_width(map_area) * sizeof(lv_color_t);
                blit.src_area.x1 = (draw_area->x1 - (map_area->x1 - disp_area->x1));
                blit.src_area.y1 = (draw_area->y1 - (map_area->y1 - disp_area->y1));
                blit.src_area.x2 = blit.src_area.x1 + draw_area_w;
                blit.src_area.y2 = blit.src_area.y1 + draw_area_h;

                blit.dst = disp_buf;
                blit.dst_width = lv_area_get_width(disp_area);
                blit.dst_height = lv_area_get_height(disp_area);
                blit.dst_stride = lv_area_get_width(disp_area) * sizeof(lv_color_t);
                blit.dst_area.x1 = draw_area->x1;
                blit.dst_area.y1 = draw_area->y1;
                blit.dst_area.x2 = blit.dst_area.x1 + draw_area_w;
                blit.dst_area.y2 = blit.dst_area.y1 + draw_area_h;

                blit.opa = opa;

                if(lv_gpu_nxp_vglite_blit(&blit) == LV_RES_OK) {
                    return;
                }
                /* Fall down to SW render in case of error */
=======
            if(lv_area_get_size(draw_area) >= LV_GPU_NXP_PXP_BLIT_SIZE_LIMIT) {
                lv_gpu_nxp_pxp_blit(disp_buf_first, disp_w, map_buf_first, map_w, draw_area_w, draw_area_h, opa);
                return;
>>>>>>> 4c7a2ae0b7d8f564b22e7774039f6cbe5385926c
            }
#elif LV_USE_GPU_STM32_DMA2D
        if(lv_area_get_size(draw_area) >= 240) {
            lv_gpu_stm32_dma2d_copy(disp_buf_first, disp_w, map_buf_first, map_w, draw_area_w, draw_area_h);
            return;
        }
#endif
```

embeddedt, 2020-10-09:
CI doesn't test every configuration combination, so it was passing. Thanks for the report.

kisvegabor, 2020-10-12:
Hi,

I've fixed it. 

@jozba could you confirm if I did it well?

closed 2020-10-12 by kisvegabor

jozba, 2020-10-29:
Hello Gabor, 
yes, that looks good.

commit c4f8959e (2020-10-12): remove version control text
Fixes #1846