lvgl/lvgl #1848: Nested lv_page drag ability.

iDalink, 2020-10-13:
I make a pair of lv_page, named `a_page` & `b_page`. `a_page` is `b_page` parent.  `a_page` can be dragged vertically, and `b_page`   layout as a line that accept horizontal drag event. They works fine. But when I slide down on `b_page`, `a_page` can not be dragged. Coulld I have an effect than `a_page` can be dragged vertically  though I touch `b_page`?

I am using lvgl6.

```
          |-----------------------------|
          |   a_page                            |
          |                                          |
          |                                          |
          |                                          |
  |---------------------------------------------------------- |
  |---------------------------------------------------------- |        b_page
          |                                          |
          |                                          |
          |                                          |
          |                                          |
          |                                          |
          |                                          |
          |____________________________|
```

embeddedt, 2020-10-13:
LVGL 6 has limited scroll propagation ability. Please upgrade to LVGL 7 if you need this feature.

embeddedt, 2020-11-05:
Closing as there was no response from the OP and the feature is not available in LVGL 6.x.

closed 2020-11-05 by embeddedt