lvgl/lvgl #1849: click event doesn't work for the image object

ali-rostami, 2020-10-14:
**Describe the bug**

Hi, Aa the [documentation ](https://docs.lvgl.io/latest/en/html/widgets/img.html#events)says we should be able to catch the generic events.
But unfortunately, It doesn't work for me and I catch nothing when I touch the image object I created.
**To Reproduce**
This is my code:

```
 img_test = lv_img_create(obj_show_desktop, NULL);
 lv_img_set_auto_size(img_test , false);
 lv_obj_set_size(img_test , 79, 40);
 lv_img_set_src(img_test , &power_status_icon);
 lv_obj_align(img_test , NULL, LV_ALIGN_IN_BOTTOM_RIGHT, -30, -50);
 lv_obj_set_event_cb(img_test , img_test_callback);
```

**Expected behavior**

I need to catch `LV_EVENT_SHORT_CLICKED` in the callback of the img_test.

embeddedt, 2020-10-14:
By default, images do not catch click events. https://github.com/lvgl/lvgl/blob/966b4b11f62e744106a5f4e9f712ab975a70f18c/src/lv_widgets/lv_img.c#L99

You can call `lv_obj_set_click(img_test, true)` to change that behavior.

ali-rostami, 2020-10-15:
Oh, didn't know.
So none of the events are caught by default, yeah?
If so, I think the documentation should be changed based on this.

kisvegabor, 2020-10-16:
> So none of the events are caught by default, yeah?

The non-input device-related events are still sent. E.g. `LV_EVENT_DELETE`

> If so, I think the documentation should be changed based on this.

I agree. this question comes up from time to time. Could you send a PR?

ali-rostami, 2020-10-18:
Thank you guys, @embeddedt &  @kisvegabor 🙏🎉
I sent a PR.

kisvegabor, 2020-11-09:
As the PR is merged I close this issue.

closed 2020-11-09 by kisvegabor