lvgl/lvgl #1850: there are display issue for calendar

ZhaoQiang-b45475, 2020-10-14:
Hi,
I create a calendar with 200*180, it is ok when I run on screen with 480*272.
However, if I change LV_HOR_RES_MAX to 640 and LV_VER_RES_MAX to 480,
there will be display issue (as attached picture), is it a bug? or how could I get the relationship between LV_HOR_RES_MAX /LV_VER_RES_MAX and smallest size for calendar which has no display issue when I set the font size to montserret_12.
[image]

embeddedt, 2020-10-14:
It's happening because the material theme changes padding based on display size. Search for `lv_disp_get_size_category` in `lv_theme_material.c`.

Looks like a bug.

ZhaoQiang-b45475, 2020-10-15:
Will it be fixed soon?

kisvegabor, 2020-10-16:
@ZhaoQiang-b45475 
AS @embeddedt the material theme tries to set the values based on the display size and resolution. 
Be sure to set LV_DPI correctly as well.

closed 2020-11-22 by stale[bot]

embeddedt, 2021-02-12:
While playing around with a custom theme I also ran into this problem. I decided to just disable line wrapping on day numbers and names, since it is broken anyways.

For future readers using newer versions: if your numbers look too squished you need to use a smaller font, increase the padding, or change the DPI.

kisvegabor, 2021-02-15:
>  I decided to just disable line wrapping on day numbers and names, since it is broken anyways.

Good idea!

commit 0f48c62b (2021-02-12): fix(calendar) force no wrapping on day numbers and names

Fixes #1850