lvgl/lvgl #1852: v8 status report

kisvegabor, 2020-10-16:
- **Release date** Planned to November/December
- **News** See the [ROADMAP](https://github.com/lvgl/lvgl/blob/master/docs/ROADMAP.md)
- **Development branch** `dev-v8`
- **Status**
  - [x] Planning features 
  - [ ] Implementation (~80%)
  - [ ] Docs update
  - [ ] Testing
  - [ ] Release

kisvegabor, 2020-10-16:
A more detailed report of the current state:
- Besides [grid](https://github.com/lvgl/lvgl/blob/dev-v8/src/lv_core/lv_grid.c) I implemented [flexbox](https://github.com/lvgl/lvgl/blob/dev-v8/src/lv_core/lv_flex.c) as well.  Both are only a few hundred lines of code.
- grid:
  - supports track sizes px and fr (pct is under consideration)
  - items can be placed to start/end/center/stretch in the cell in both directions
  - support nested grids
- flex:
  - column, column wrap, row, row wrap support
  - px, pct, grow(x) sizes on main axis 
  - start, center, end placement on the cross axis 
  - support nested grids
- scrolling:
  - supported on lv_obj directly
   - elastic scroll
  - automatic scroll chaining (scroll the parent if the obj. can't be scrolled)
  - snap: ensure that when scroll ends the closest "snappable" children will be positioned to the start/center/end of the parent
  - scroll stop: allow scrolling only to the next snap point
- allow setting object size in percentage. Automatically updated when parent size changes.
- lv_components:
  - almost all widgets are rewritten and hugely simplified (e.g. tabview is [210 lines](https://github.com/lvgl/lv_components/blob/master/lv_tabview.c) instead of [1100 lines](https://github.com/lvgl/lvgl/blob/master/src/lv_widgets/lv_tabview.c))
 - still not clear how/where to create examples for the components. Maybe the lv_examples/src/lv_components

What is still missing:
- Some widgets and components need to be rewritten    
- Not all features are tried out
- Docs update

embeddedt, 2020-10-16:
@kisvegabor Looks good! The API is stable enough to start testing now, right?

kisvegabor, 2020-10-16:
Now I continuously merge v7 to dev-v8 but the 2 codebase are quite different now. At some point, we should freeze v7 and make people add features to dev-v8. 

At this point, it's not possible (no examples and docs to the new features and still anything can be changed). 
I believe by the end of October dev-v8 gets stable, and some examples and docs will be added. So at the beginning of November, I think we can make the switch.

---
@embeddedt 
It's stable for testing. Some clean up and refactoring are expected though. Later today I'll create a branch in `lv_sim_eclips_sdl` with the correct submodule versions and  `lv_conf.h`.

embeddedt, 2020-10-16:
We can't expect people to start using v8 while core widgets are missing, so we would need some type of transitionary glue code. Is there a plan in place for that? (I vaguely recall discussing this before but I forget the decision.)

kisvegabor, 2020-10-16:
I've just pushed the project to `dev-v8` in `lv_sim_eclipse_sdl`. 

> We can't expect people to start using v8 while core widgets are missing, so we would need some type of transitionary glue code. Is there a plan in place for that? (I vaguely recall discussing this before but I forget the decision.)

IIRC, the conclusion was that it's not reasonable to keep old and new features (not widgets) together because it'd be very messy. 
And once there are some serious API breaking features (layout, scroll) we are fine with **similar** widgets and exact replacements are not "mandatory". 

Anyway, the goal is to make the transition between major version reasonably smooth. So any idea or suggestion is welcome. 

Note that, the core widgets still have pretty much the same API. However, the components have changed more. 

E.g. now the calendar is reimplemented from a button matrix and its "header" (buttons, year, month name) is a separate object to allow developing different headers. 

Or lv_list is also hugely simplified. It's just a flexbox container, where you can add buttons or texts. 

(The components are still heavily under development )

embeddedt, 2020-10-16:
Okay; I'm trying to build it now. There were a few simple errors which I'll submit a pull request for in a bit.

Where is the new calendar implementation stored? I couldn't find it in `lv_examples` or `lvgl`.

kisvegabor, 2020-10-16:
> There were a few simple errors which I'll submit a pull request for in a bit.

I'm curious about the errors - as it compiles well here. :slightly_smiling_face: 

> Where is the new calendar implementation stored? I couldn't find it in lv_examples or lvgl.

In lv_components. I've noticed that it wasn't added and committed. It should be there now.
 https://github.com/lvgl/lv_components/tree/master/lv_calendar

embeddedt, 2020-10-18:
It looks like the majority of my problems are stemming from the fact that I'm using my v7 configuration file, which has all the old object types enabled. Seemingly their implementations were removed but there are several places in the codebase where they're still being referenced (e.g. themes). In particular the chart and objmask seemed to produce a lot of errors when enabled.

I pushed one simple fix directly to `dev-v8` as it wasn't worth a pull request. I'm working around the rest of the errors by disabling objects/features till they go away. Once I get it running I'll start cleaning things up.

embeddedt, 2020-10-18:
I'm not sure how to go about this as we obviously want some of the old styles to stay. Right now the themes still have styles for all the old object types; what is the equivalent in `lv_components`?

embeddedt, 2020-10-18:
Is `LV_STYLE_CACHE_LEVEL` supposed to be a config option?

kisvegabor, 2020-10-19:
@embeddedt 
Is in it working with the `lv_conf.h` file I've pushed to the Eclipse project? `LV_STYLE_CACHE_LEVEL` is also defined there.

> I'm not sure how to go about this as we obviously want some of the old styles to stay. Right now the themes still have styles for all the old object types; what is the equivalent in lv_components?

I'm still also thinking about it. My current idea is to make the component to add some styles. For example, the tabview header needs a different style than the default button matrix. So the tabview component can create a style that changes borders, padding etc. It keeps colors and other stuff from the theme, just apply the changes it really requires.

What do you think?

embeddedt, 2020-10-19:
> Is in it working with the `lv_conf.h` file I've pushed to the Eclipse project? `LV_STYLE_CACHE_LEVEL` is also defined there.

I will try that file.

> So the tabview component can create a style that changes borders, padding etc. It keeps colors and other stuff from the theme, just apply the changes it really requires.

That should work. We do need to check when reviewing contributed components that they work properly with multiple themes.

kisvegabor, 2020-10-19:
> That should work. We do need to check when reviewing contributed components that they work properly with multiple themes.

Yes, however, if the "gist" of the themes remains it should be fine. I mean if the component style changes only what it really needs the result still should fit into the theme's design. 
Just to mention another example: if the radio button would be a component based on the checkbox it needs to change only the radius to circle. 
  
The other approach would be to have "component themes" but that would be really messy and difficult to handle multiple themes.

puzrin, 2020-10-20:
> I'm still also thinking about it. My current idea is to make the component to add some styles. For example, the tabview header needs a different style than the default button matrix. So the tabview [...]

Inheritance and mixins should be done only for objects which have real relations, not just similar. In other case you can get very fragile dependencies.

See TWBS. They have a lot of constants ant put those to css class descriptions.

puzrin, 2020-10-21:
Is there a possibility to not use themes at all (zero preset)? If user prefer to assign all styles manually.

embeddedt, 2020-10-21:
@puzrin `lv_theme_empty` can be used for this purpose.

kisvegabor, 2020-10-21:
@puzrin 
> Inheritance and mixins should be done only for objects which have real relations, not just similar. In other case you can get very fragile dependencies.

Sorry, but I don't understand what it refers to. I just meant that when you create a tabview it will set its header style
(header = the part with the tab buttons).

puzrin, 2020-10-21:
> Sorry, but I don't understand what it refers to. I just meant that when you create a tabview it will set its header style
> (header = the part with the tab buttons).

I thought you wish to reuse widget "buttons list" an update style. That widget has different semantic meaning, it's better to keep tabs completely independent.

If you need to share some generic props values, create hidden abstraction layer - "generic theme constants" (#define-s), and use those when fill styles with props.

kisvegabor, 2020-11-16:
I'm planning to make tomorrow's release the last v7 release and merge `dev-v8` to `dev`. 

I consider v8's API stable and haven't experienced bugs. The examples are updated (the demos are still WIP) and I'll start to update the docs this week.

The flexbox and grid examples could be particularly interesting: 
https://github.com/lvgl/lv_examples/tree/dev-v8/src/lv_ex_layout

Let me know if you have and objection about the merge.

puzrin, 2020-11-16:
OOP-emu lib is for v9 or "undecided"? That's "must have" (IMO), but big breaking change. Technically - not difficult but boring (a lot of search-replace in editor).

Importance of ooc for me - immediate after v8 new layout with classes.

kisvegabor, 2020-11-16:
I've just realized that if the current is `dev` will merge to master we will need and another v7 release with the current dev (which contains e.g. the Kconfig so it's quite important)

fvanroie, 2020-11-30:
I decided to try out `dev` and see if I could get v8 to compile for my ESP32 Arduino environment in PlatformIO:
```
PLATFORM: Espressif 32 (2.0.0) > WEMOS LOLIN D32 PRO
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
- framework-arduinoespressif32 3.10004.201016 (1.0.4)
 - tool-esptoolpy 1.20600.0 (2.6.0)
 - toolchain-xtensa32 2.50200.80 (5.2.0)
```

 The good news is it worked! Here are the steps I did:

- I switched to the `dev` branch and configured a new `lv_conf.h`
- Included `lv_components` library and configured `lv_comp_conf.h`
- Created a temporary shim with macro's mapping the old to the new api's, where possible
- Masked some parts of the code with `#ifdef LVGL_VERSION_MAJOR != 8`, just to get up-and-running quickly

My first remarks: *(I know it's a development version)*
- `lv_timer.h` is sometimes included from `<lvgl/src/lv_misc/lv_timer.h>` which throws off the compiler. I have replaced these includes by `"lv_timer.h"` 
- lv_components widgets do not use `LV_LVGL_H_INCLUDE_SIMPLE`. I added it to the includes of each lv_component widget
- lv_colorwheel widget declares `lv_colorwheel_get_color()` in the header, but the source file defines `lv_colorwheel_get_rgb()`
- The colorwheel *saturation* gradient, only shows as a semi-arc and not the full circle. I think the black is not supposed to be there:
[image]

Now that I got it to compile and the display works without showstoppers, I will try out some more of the features and see how to migrate the code I have temporarily commented out.

kisvegabor, 2020-12-02:
Thanks for testing!
I saw your PR and trying to make the CI tests sucessfully run. I'll coment there. :slightly_smiling_face:

kisvegabor, 2020-12-02:
> The colorwheel saturation gradient, only shows as a semi-arc and not the full circle. I think the black is not supposed to be there:

I'll take look

> lv_obj_add/clear_flag should invalidate the object when flag = LV_OBJ_FLAG_HI

True, I'll fix it.

fvanroie, 2020-12-02:
Thanks, I've found the *colorwheel saturation gradient* issue and included it in the lv_components PR. Only 5. `lv_obj_add/clear_flag` issue I couldn't get to work yet.

kisvegabor, 2020-12-03:
@fvanroie Okay, thank you!

kisvegabor, 2020-12-08:
We have postponed a v8 a little to add some more features to v8:
- rewrite how inheritance works
- speed up and optimize styles

Now, v8 is in the `dev` branch so I think we should add `dev-v8` again (besides `dev`) until we see the end of the development of the new feature.  @embeddedt  do you agree?

embeddedt, 2020-12-08:
Sounds like a good plan. Are we resetting `dev` to the last v7-related commit then?

kisvegabor, 2020-12-09:
Great! 

> Are we resetting dev to the last v7-related commit then?

Yes, I've just done the following:
- create `dev-v8` from `dev`
- create `dev` from `master`

embeddedt, 2020-12-09:
It looks like you may have also inadvertently closed all the incoming pull requests to `dev`. Was that intentional?

kisvegabor, 2020-12-09:
It happened automatically when the "old" dev branch was deleted. I've reopened both related PRs.

kisvegabor, 2021-02-02:
I'm thinking about a small restructuring and curious about your opinion.

The main question is where to store the examples of `lv_compoennts`:
1. In `lv_example/some_widget`
2. In `lv_components/some_widget/example`

I think the cleanest would be to store the examples as close to the "source" as possible to be user people will find it.
Following this idea the examples from `lv_examples` could be moved to `lvgl`.

So my idea is a layout like this:
```
lvgl
  - src
  - test
  - examples (could be easily reused in tests?)
 
lv_components
 - widgets
   - some_widget
     - src
     - test
     - exampe
     - README.md (the docs of the widget, included in to lvgl's docs too)  

lv_examples renamed to lv_demos to store only large demos.
```

What do you think?

@embeddedt would it make a lot of trouble with the automatic rebuilding of docs and examples?

embeddedt, 2021-02-02:
The restructuring makes sense. It also makes it easier to contribute a widget+example+docs change without having to send pull requests to 2 or 3 different repositories, which lowers the bar to entry.

On the other hand, it means that examples are now in 3 different places instead of 1. I'm not sure how I feel about that. For the docs, I would need to redesign part of the example script to have a flag so that it chooses between `lvgl`, `lv_demos`, and `lv_components`.

----------------

Would you be open to the idea of storing everything in one repository (`lvgl`) and having multiple subdirectories?

* `src` - core library code
* `components` - user-contributed components
* `examples` - component examples and demos

This is how other projects like [Material-UI](https://github.com/mui-org/material-ui) and [React Bootstrap](https://github.com/react-bootstrap/react-bootstrap) do it. They keep the examples in their own directory separate from the core code, but it's all in one repository. This eliminates the issue of having to keep repositories in sync.

kisvegabor, 2021-02-02:
I'm open to the one repo idea!
It also brings people closer to the core code and hopefully makes it less scary to contribute to the core one they had some simple PRs.

What about the docs? I think it's complex enough to keep in its own repo.

embeddedt, 2021-02-02:
> What about the docs? I think it's complex enough to keep in its own repo.

I agree; it matters less if the docs are slightly out of sync since the information usually applies to older/newer versions anyway (and major releases are always in sync).

kisvegabor, 2021-02-02:
I make the release now, and make restructuring in `dev_v8` after that.

kisvegabor, 2021-02-02:
The usual question: naming...
What about this:
```
lvgl
  - examples
    - widgets
      - lv_btn or simply btn
         - lv_btn_example_1.c        
```

kisvegabor, 2021-02-02:
Another thing: can we store the examples of core things and the component  in the same folder?
E.g.
```
examples
  - widgets (core widgets)
    - btn, slider
  - layouts (core layouts)
    - flex, grid
  - components
    - widgets
      - win, tabview
    - layouts
      - custom layouts
    - apps
      - magic menu, or some complex app             
```

or put them in the same folder:
```
examples
  - widgets
    - btn, slider, win, tabview
  - layouts
    - flex, grid, custom layouts
  - apps
    - magic menu, or some complex app             
```

I prefer the latter.

embeddedt, 2021-02-02:
The latter is fine. I think the `lv_` prefix is overkill for the examples since they are already in an `lvgl` repository, as long as the example functions still start with `lv_`.

kisvegabor, 2021-02-02:
Great!
So go for example names like `lv_btn_example_1.c ` and drop `lv_` and `lv_ex_` on folder names.

embeddedt, 2021-02-02:
Precisely.

kisvegabor, 2021-02-05:
Maybe `lv_example_btn_1.c` and `lv_example_btn_1()` would be better auto complate. What do you think?

embeddedt, 2021-02-05:
True; let's go with that.

kisvegabor, 2021-02-12:
dev-v8 is getting very close to begin its public testing. 

Here is what is working:
- class-like behavior
- new scrolling
- new layouts
- updated styles
- reworked widgets
- the former concept of `lv_components` was moved to `lvgl/src/extras` 
- `lv_examples` is moved to `lvgl/examples`
- all examples are reworked and new ones are added
- lv_demo_stress, benchmark and keyboard_and_encoder is working
- simplified and restructured lv_conf.h

what is still missing:
- a few more examples (e.g. for scrolling, styles, get started)
- lv_demo_widgets, printer and music rework 
- update docs
- mono theme
- dark theme
- memory usage and performance test/optimization
- check the compatibility with MicroPython
- update tests, fix warnings
- update `lv_examples` to `lv_demos`
- create a simulator project to easily test v8
- update the automatic build of docs and examples

I think the next release on Tuesday could be the last v7 release and `dev-v8` can be merged to `dev`.

C47D, 2021-02-12:
Will the Kconfig file need updates to catch up with the new `lv_conf.h` isn't? Since we're still doing that manually.

kisvegabor, 2021-02-15:
> Will the Kconfig file need updates to catch up with the new lv_conf.h isn't? Since we're still doing that manually.

Unfortunately yes. But fortunately, we have ~1 major release/year so it doesn't happen too often.

Most of the parts haven't changed just reorganized.

---

There are 2 more things in my mind:
1. I think our release policy can be simplified to have 1 minor release/month and bug fix releases on demand. This way we can merge everything right into master if the tests pass. When v8 is released I will treat adding a correct test background and a reasonable number of test (mainly screenshot compare) at the highest priority so we can be more confident with merging directly into master.
2. Hosting the docs on https://readthedocs.org/. MicroPython hosts the source of the docs in the main repo and seemingly builds it from there. I've never used readthedocs and not sure it fits our case. (With @embeddedt's auto rebuild script our docs is also quite convenient to use) 
What do you think?

embeddedt, 2021-02-15:
I agree with the release policy change. 

Read The Docs seems to be largely a drop-in replacement for GitHub Pages + my script. Since we've effectively built a similar system, how about we leave the docs as they are for now? (Unless you see an advantage to RTD that I didn't.)

kisvegabor, 2021-02-16:
> Read The Docs seems to be largely a drop-in replacement for GitHub Pages + my script. Since we've effectively built a similar system, how about we leave the docs as they are for now? (Unless you see an advantage to RTD that I didn't.)

Okay. I like 2 things in read the docs:
- the nice version selector
- the source of the docs is hosted in the main repo (but I guess we could do it your version too if lvgl is added a submodule to the docs repo)

embeddedt, 2021-02-16:
> the source of the docs is hosted in the main repo (but I guess we could do it your version too if lvgl is added a submodule to the docs repo)

We can do that with mine as well; we can just move the Sphinx things into the `lvgl` repo and adjust the script. Although, if we're moving stuff, maybe we should just look into RTD since lots of projects use it.

kisvegabor, 2021-02-16:
I'm absolutely ok with the current working mechanism too. It knows what we really need. I just mentioned RTD as an opportunity with some nice features. 

You put a lot of energy into improving our docs and it'd be reasonable to drop all these for a minor gain. 

However, if you are curious about readthedocs

embeddedt, 2021-02-25:
Let's stick with the current approach for now, then.

----

Another topic: I'm testing v8 now and found with `lv_example_dropdown_3` the option text gets bolder when an option is focused. Is that another case of the v7 issue where the value is drawn twice?

puzrin, 2021-02-27:
>  1. I think our release policy can be simplified to have 1 minor release/month and bug fix releases on demand. This way we can merge everything right into master if the tests pass.

What is the difference? Do you plan make things more close to "github flow" / "git flow"?

> 2. Hosting the docs on readthedocs.org.

What for? All doc generators are ~ similar. You need reasons to change. For example, "minimize your own hosting".

For example, i would move lvgl.io to gh-pages and so on, to simplify maintenance.

embeddedt, 2021-02-28:
New bug: color palette `LV_COLOR_PALETTE_BLUE_GREY` apparently has no accent colors (I get a black dropdown menu in `lv_example_dropdown_3`).

kisvegabor, 2021-03-01:
> Another topic: I'm testing v8 now and found with lv_example_dropdown_3 the option text gets bolder when an option is focused. Is that another case of the v7 issue where the value is drawn twic

Should be fixed now.

> New bug: color palette LV_COLOR_PALETTE_BLUE_GREY apparently has no accent colors (I get a black dropdown menu in lv_example_dropdown_3).

I've just removed the accent colors at all. We were checking them with our designer and found that they are highly inconsistent. Now the default theme uses the main colors.

---

**Release policy** 

I've read about git flow and github flow again but none of them fit well into our case. But I found gitlab flow what is exactly we need!

In short, it has 2 versions: with production branch or with release branches. We need the release branches.

The release branches are for minor versions and all features and fixes are merged into master. When the "time comes" a new release branch is created from the top of master.  Bug fixes are merged into master but can be cherry-picked to release branches.

Basically, that's it. 

The workflow with scripts and github action can look like this:
- a script like `addfix -b <branch-list> -c <commi-list> -bump` can be used the cherry-pick fixes to branches. `-bump` can automatically bump the version (only patch version number) and tag it. 
- another script like `release minor/major` creates a new release branch from master, and tag it.
- a third script like `update_project -b <branch-list>` updates the lvgl projects. It means the projects should have the same versioning scheme. 
- a github action automatically creates releases when a tag is pushed   

So 3 simple, independent scripts, instead of a monolithic complex one that we have now. 

One more thing: `lv_examples` (renamed to `lv_demos`) , `lv_drivers` and maybe the other "plugin" repos (e.g. `lv_lib_gif`) should follow the minor versions of lvgl to make it straight forward which versions are compatible. 

I vote for a much long release cycle, e.g. make major/minor versions every 3 months. 2.5 months of implementation + 0.5 month testing (in the last 2 weeks features are not merged in `master`).

What do you think?

puzrin, 2021-03-01:
> I vote for a much long release cycle, e.g. make major/minor versions every 3 months. 2.5 months of implementation + 0.5 month testing (in the last 2 weeks features are not merged in master).

Will hotfix releases stay immediate? If you speak about:

- **A**.b.c - new breaking release
- a.**B**.c - new non-breaking feature

I see no problems with 3 month cycles. But as far as i remember, those are already short.

kisvegabor, 2021-03-01:
> Will hotfix releases stay immediate?

Not necessarily immediate, but they could be added at any time.

> I see no problems with 3 month cycles. But as far as i remember, those are already short.

Now it's 2 weeks for minor and major versions.

puzrin, 2021-03-01:
> Now it's 2 weeks for minor and major versions.

Two WEEKS for major versions, breaking everything? Sounds impossible.

I would not care too much about "cycles" until architecture is unstable. But i would declare some "deadlines", when changes should be landed.

- 3 months for "breaking everything", like v8 (can be increased as exclusion, but not more than 6 months).
- 0-4 weeks for hotfixes.

kisvegabor, 2021-03-01:
> Two WEEKS for major versions, breaking everything? Sounds impossible.

Actually, it's only theoretical as we have only ~1 major release per year. So let's consider it 2 weeks/ minor version instead. 

> That's like roadmap file - not strong promise, but intent to follow.

Sounds good.

kisvegabor, 2021-03-01:
> In current state, i'd suggest to focus on setup of "normal" CI testing right after v8, asap.

That's the plan. After the initial issues are fixed, I'll focus on setting up a normal test environment and adding some tests (mainly screenshot compares for first, as they test the whole library from top to bottom)

embeddedt, 2021-03-01:
I'm playing with the v8 examples and found a potential performance issue.

In `lv_example_scroll_3`, adding a new list button triggers 13 flex layout updates. 3 are triggered upon creating the object, 6 are triggered after the image is created, and 4 are triggered after the label is created. Ideally only one should be triggered after all creation is finished. Can this be handled automatically (maybe layout calculation can be deferred till the start of the next rendering cycle), or do we need a flag to skip layout updates?

[29 more replies]

closed 2021-03-16 by kisvegabor