lvgl/lvgl #1853: Unit and functional tests

raulgotor, 2020-10-17:
I was talking with Gabor about contributing with LVGL, and one of the points he was interested in was in writing unit and functional tests. I'm interested on contributing, as I use LVGL in some of my projects. If you'd like, we could discuss the following steps here.

C47D, 2020-10-17:
I'm particularly interested in unit testing because I've used it at work a few times and wanted to learn more about it by contributing to LVGL, there was a discussion about it in #1757.

Do you already have something in mind on how to contribute?

kisvegabor, 2020-10-20:
Thank you @rabbitsaviola and @C47D for showing interest in this topic.

As v8 will be released soon I think it'd be the best to get started with some simple parts that won't change in the new version.
These kind of units are mainly in `lv_misc` folder:
- lv_utils
- lv_mem
- lv_ll
- lv_task
- lv_math
- lv_color
- lv_area

About a year ago I've written a simple test framework with some asserts. I'm not that experienced with unit tests so can't really decide if it's worth using an existing framework or use the current one. 
The current framework can take snapshots from the framebuffer and compare them with a reference image. It will be critical later for more complex tests. 

IMO, the most effective approach would be to get started with the current framework. Later it should be easy to replace the asserts to a new framework if required.

puzrin, 2020-10-20:
Problem consists of 2 steps:

1. Create infrastructure to add an manage tests with ease
2. Add tests upon demans.

(1) was discussed in #1658
(2) the ready case is font decompressor testing - test vectors already available in JS.

kisvegabor, 2020-10-21:
> Create infrastructure to add an manage tests with ease

Do we expect anything else than asserting on test fail?

> the ready case is font decompressor testing - test vectors already available in JS.

True, I added to the list

puzrin, 2020-10-21:
> Do we expect anything else than asserting on test fail?

Sure. There should be configs for (?) cmake, defining how to arrange tests into groups and execute each.

- usually all tests are joined into single output (faster build).
- but we may need different configs (i guess, 1-3 static ones) => set of builds.

Assertions is only 1/3 of things. The rest is:

- machinery how to build/run all.
- mocks.
- coverage report.

Imagine, you prepaired file with unit test. What's next? How to "activate"?

kisvegabor, 2020-10-26:
@puzrin 
I see. So we have 2 options:
1. Set up a professional environment now and rewrite the current (not too many) tests
2. Go with the current test environment and change to another later.

Obviously 1) would be better, however personally I planning to deal with it after v8 is released.  So if anyone has the liking to work on it please let us know. Until that, I think we are better to write tests for the current system.

puzrin, 2020-10-26:
Unfortunately, as not-c-developper, i'm unable to provide concrete trusted examples. To be honest, i hope use your project to learn how to use cmake and so on :).

But i'm absolutely sure, current "home-made" code should be replaced with "professional" asap (after-v8 is nice timeline). For example, linter & coverage reports help a lot at first run of each new test.

puzrin, 2020-10-27:
Asked on local forum for good projects to scrape testing infrastructure from.

- https://github.com/cgreen-devs/cgreen - cmake
- https://github.com/Snaipe/Criterion - meson + ninja

detly, 2020-10-28:
FYI I've used [Unity](https://github.com/ThrowTheSwitch/Unity) (not related to the game engine) with very good results for unit testing C code targeting embedded environments. They do particularly well with broad support for rare platforms eg. DSPs where `CHAR_BIT == 16`, or platforms with partial support for later C standards than C89. I *haven't* used CMock or their other helpers however.

A lot of it comes down to *where* you want to run your tests. On the target hardware? On a simulator? On a reference platform with a known-standards-compliant C compiler and nowhere else?

puzrin, 2020-10-28:
For this concrete project, tests are expected to run automatically on "PC" with continuous integration. That will cover 99.9% of real needs. Running tests on hardware will add a lot of extra efforts with almost zero extra value - possible, but not rationale.

Also, we are not limited with compiler for testing, and can use CPP testing frameworks.

detly, 2020-10-28:
> For this concrete project, tests are expected to run automatically on "PC" with continuous integration. That will cover 99.9% of real needs. Running tests on hardware will add a lot of extra efforts with almost zero extra value - possible, but not rationale.

I was thinking more that it would be useful for a user, like as a sanity check to see if lvgl works out of the box on their selected platform. Having such a goal almost certainly would rule out C++ test frameworks however.

puzrin, 2020-10-28:
No, it's not a real problem which should be solved. Real problem is development & everyday use - testing complicated algorithms, regressions & coverage.

With cmake you still can run some group of tests with C99 and Unity. For example, to check build-ability. But IMO no needs to force use all tests such way.

Crazor, 2020-10-29:
As @puzrin asked me to weigh in on this issue, I'd like to share a quick thought regarding the ability to run tests on embedded targets:
I agree that on first glance, a project like LVGL does not appear to need anything else than a test harness that runs on the developer's machine. However, there will inevitably be portability issues, like different word widths (LVGL appears to use `stdint.h`, which avoids this) or different toolchain feature sets (pragmas and annotation macros spring to mind from previous experience) that need to be addressed when porting to a new target. Having the ability to run all (or most) of the test harness gives the peace of mind that changes that were necessary for one target don't break things on a different target. Those can be obvious problems like breaking the ability to compile on another target, or subtle things like different behaviour because of the aforementioned word width differences mentioned above.
So me personally, I like to be able to run as many of my unit tests on target hardware as possible. That's why I use Unity for testing. It is very bare-bones, however I haven't found any replacement so far that has more features but also makes on-target testing as easy (let alone even possible) as Unity does.

puzrin, 2020-10-29:
> So me personally, I like to be able to run as many of my unit tests on target hardware as possible. That's why I use Unity for testing. It is very bare-bones, however I haven't found any replacement [...]

Everybody wish to have 100% coverage, testing on all available bare meals and so on. Question is "who will pay for that" :). Asking this question to self helps to quickly separate mandatory an optional features :).

Bugracker has stats, which kinds of bugs should be guarded. I doubt % of platform-specific problems is significant. The same about compilers.

Crazor, 2020-10-29:
Sure, that's the tradeoff. My opinion has formed over the years, as I was bitten by stuff that would have been caught by on-target testing. And believe me, nobody wants to pay for on target testing in a corporate setting! Even getting one of each target systems for a long time CI setup raises lots of eyebrows and "unnecessary"-comments :)

puzrin, 2020-10-29:
This does not answer question, what is mandatory for concrete project right now. As i said in https://github.com/lvgl/lvgl/issues/1853#issuecomment-713674320 assertions (unity) is only a small part of required features. Statement "let's use unity" does not helps to understand what to do with the rest.

From my experience, asserts only, wihout code analyser, coverage reports, mocks and tests auto-discovery is a pain. Much more serious than lack of bare metal tests.

About compilers - we can CI gcc & clang only, anyway. So no much need to restrict self with pure C only.

IMO cpp framework + cmake should be ok for next 3-5 years.

embeddedt, 2020-10-29:
@Crazor brings up a good point. I recall that big endian platforms were broken at some point - fortunately I believe that's fixed now, but as none of us use big-endian hardware on a regular basis, it's hard to verify.

I think QEMU will run inside a GitHub action/CI, so we can do some basic testing on other architectures, especially if we have an easy way of getting screenshots.

puzrin, 2020-10-29:
LE/BE testing is not a big deal. As far as i remember, Gem5 has option to select ARM endianess.

I mean, this is still not bare meal issue - can be solved via sim.

kisvegabor, 2020-10-30:
I wonder if we really need to decide the question of platform-independent tests now. I mean we should be fine with tests on PC for first and later add support to different hardwares.

Anyway, I saw at NuttX that they have a huge number of tests on different platforms. See [this PR](https://github.com/apache/incubator-nuttx/pull/2175) for example. I haven't checked how it works yet, but we can get some inspiration from there. 

In lack of experience, I'm not biased to any frameworks, so I trust in your expertise. :slightly_smiling_face:

kisvegabor, 2020-10-30:
Regarding screenshots: probably the easiest way would be to store only the hash of the reference images and compare them with the hash of the fame buffer's content.

detly, 2020-10-30:
> Regarding screenshots: probably the easiest way would be to store only the hash of the reference images and compare them with the hash of the fame buffer's content.

Do you even need to go that far? You could write a "driver" that just keeps whatever LVGL sends to a screen in a screen-sized buffer ie. array, and then compare that array to a reference array.

puzrin, 2020-10-30:
> I wonder if we really need to decide the question of platform-independent tests now. I mean we should be fine with tests on PC for first and later add support to different hardwares.

Hardware tests may have limiation - available compiler. On PC/Sim we have gcc/lvgl/msvc => can use CPP testing frameworks - more comfortable.

IMO reason "may be after 2-3 years we decide to run all written tests on bare metal" - not very impressive to stay with pure C :).

detly, 2020-10-30:
I should point out, I only suggested Unity (and the strategy of broad target testing) in case you hadn't considered it. I don't have a particularly strong opinion on whether you should use it. Also the questions of "whether you use Unity" and "whether you try to have tests that run everywhere LVGL runs " are orthogonal; Unity is nice, I've found it useful, but it doesn't have the range of features that other frameworks have. I sometimes use it even when the deciding factor *isn't* portability.

Regarding targeted testing — @puzrin is quite right, it is extremely unlikely you'd set up even a small subset of your intended targets for Github's CI. Even when I've been working directly on a known embedded target, setting it up in a completely controlled environment, and being paid to do so, it's usually such a massive pain that I don't bother. The reason you *might* want to adopt a policy of allowing it is if you think you'll get a lot of issues opened relating to esoteric targets that you'll want to address but will suck up a lot of your time debugging. If that's the case, allowing bare metal tests mean a reporter can *manually* run tests on their target, send you the results, and it might help cut out some debug time. So it all comes down to whether your crystal ball says you'll spend more time on that than you'd spend implementing super portable tests. For some projects the answer is ~~ask again later~~ yes.

Regarding C++ vs C test frameworks — it only matters if you're expecting to hit issues relating to the relatively minor differences between the way a C++ compiler compiles C and the way a C compiler compiles C (eg. the type of various literals, sizes of enums, ABI consistency).

embeddedt, 2020-10-30:
> it only matters if you're expecting to hit issues relating to the relatively minor differences between the way a C++ compiler compiles C and the way a C compiler compiles C

We get bug reports frequently enough about it that it's probably worth using.

> Do you even need to go that far? You could write a "driver" that just keeps whatever LVGL sends to a screen in a screen-sized buffer ie. array, and then compare that array to a reference array.

Someone suggested (or I found) a framework where you could visually see the difference between the two images (it made a third image with only the different pixels shown). I think that would be extremely useful for seeing discrepancies, if it was easy to set up.

kisvegabor, 2020-10-30:
> Do you even need to go that far? You could write a "driver" that just keeps whatever LVGL sends to a screen in a screen-sized buffer ie. array, and then compare that array to a reference array.

Actually the greatest question here - at least for me -  is how to update reference images. Let's say we have a ref. image about the button and it needs to be updated. So we need to call the test code that generates the button's image, check it manually, and if it's fine overwrite the original image. The question is how to call the test function from a normal application, or make the test to render the images in a window.

It might be easier to manage hashes but with real images, we can output the difference as @embeddedt  suggested. So I don' insist on hashing it's just a possible approach.

puzrin, 2020-10-30:
> Actually the greatest question here - at least for me - is how to update reference images. Let's say we have a ref. image about the button and it needs to be updated. So we need to call the test [...]

1. If button has no custom drawer, you don't need to test it's image. You need to test [lv_obj + styles].
2. Split testing of drawer and "color shrinker". Then test drawer in full RGBA mode for simplicity.
3. Create virtual display driver (mock), with methods for memory access

Probably, framework already has appropriate assertions. If not, it can be extened with this:

```
assert.compare_images(display, rect, fixture path);
```

If assert fails, i does additional action - write to log dir tested area in png format, used fixture and diff image. So:

On first pass, you will get fail, and if recorded image is ok, copy to fixture as right value. Bingo :) !

jozba, 2020-11-01:
Regarding the platform tests - I agree with @puzrin that vast majority of logic is in platform-agnostic algorithms that could be tested easily on x86 PC.  But there's that one small integration part that can make a big head ache. No one will use LVLG on x86. Having correct result image is critical. But if someone is about to explore LVGL port for his particular platform and his out-of-the-box experience is "syntax error", that's not good starting point on the way to convince customer. I would not under-estimate this. 

I have some experience with building automatic multi-platorm test farms and I can confirm that it is realy difficult (and time consuming) process. But we can't pretend or expect that as LVGL is growing, testing will be easy. Trying to add platforms, where each one of them is a separate unique topic with its own issues and traps, especially with HW accelerators included, is a big deal. But I wouldn't completely vanish this idea. At least build test would be a good starting point in this direction.

puzrin, 2020-11-01:
> that could be tested easily on x86 PC

That includes architectures by qemu/gem5 simulators, not just x86.

> No one will use LVLG on x86.

Everybody use it for sdl2 sim. I do most of work without hardware at all.

> Trying to add platforms, where each one of them is a separate unique topic with its own issues and traps, especially with HW accelerators included, is a big deal.

You will be able to do that. Nobody prohibits you write 0.1% of tests (for drivers) with different framework (unity), if you ever need. But propagating this "difficulties" to all tests would be not good idea IMO.

jozba, 2020-11-02:
I didn't mean to propagate HW specific difficulties to main testing process. We could **add** this as a (separate) additional test step to improve quality. Also, we may assume that CPU results are always the same. But once any HW accelerator gets involved, code path is different from CPU rendering, so actually it's not tested at all. The idea is not to test various HW setups or integration, but to have at least build test (easy part), and smoke-test in one reference configuration (more difficult part). I have no idea how many % of uses are in this area.

kisvegabor, 2020-11-02:
Let me summarize how I see it now (listing from the most important):
1. Write a lot of tests for x86. It covers most of the issues.
2. Use qemu/gem5 for benchmarking and run all tests from 1) with hardware-specific tests (e.g. Big endian)
3. Add hardware-specific a few tests in some form to test e.g. GPUs

Do you see it in the same way?

puzrin, 2020-11-02:
~ Yes.

I suggest to do (1) only and verify (2,3) possible in future. cmake and other build/run tools should have "test suite" feature (to group tests).

kisvegabor, 2020-11-03:
Alright, then the first step is to "normal" test environment for PC. 

Are you (somebody with experience in it) interested in getting started with a test environment?

closed 2020-12-06 by stale[bot]