lvgl/lvgl #1854: lv_img rotation bug

lujuntuan, 2020-10-17:
> # Important: issues that don't use this template will be ignored/closed.

**Describe the bug**

I create a lv_img and set source. 
When I set rotation(the actual area > the itself area), The filled area will be displayed.
It looks like a bug in the mosaic feature of the picture

**To Reproduce**
I changed the code as follows, the problem is solved.(lv_img.c ,709 line)
for(; coords_tmp.y1 < zommed_coords.y2; coords_tmp.y1 += zoomed_src_h, coords_tmp.y2 += zoomed_src_h) {
    coords_tmp.x1 = zommed_coords.x1;
    coords_tmp.x2 = zommed_coords.x1 + ext->w - 1;
    for(; coords_tmp.x1 < zommed_coords.x2; coords_tmp.x1 += zoomed_src_w, coords_tmp.x2 += zoomed_src_w) {
        lv_draw_img(&coords_tmp, &clip_real, ext->src, &img_dsc);
    }
}

**Expected behavior**

**Screenshots**

**Additional context**

lujuntuan, 2020-10-17:
This is a bug that appeared after version 7.5.
7.3 and 7.4 is ok.

kisvegabor, 2020-10-19:
Thank you.
I've reverted this part.

commit df49f865 (2020-10-19): fix image transformation
Fixes: #1854
closed 2020-10-19 by kisvegabor