lvgl/lvgl #1857: animation issue

ZhaoQiang-b45475, 2020-10-21:
I use img and animation to implement feature pouring coffee into the cup, then meeting an issue.
I have 62 images, use animation to display image from index 0 to index 61.
I write a lv_anim_exec_xcb_t func to set img src from index 0 to index 61. 
Coffee cup becoming more and more green as the cup fills up.
if keep touching the screen with finger during this process, the coffee cup could display normally.
Please refer to attached zip which have 2 videos to show.
[animimg.zip](https://github.com/lvgl/lvgl/files/5414308/animimg.zip)

kisvegabor, 2020-10-21:
It's very difficult to see the issue from the video.

Can you reproduce it in the simulator too?

ZhaoQiang-b45475, 2020-10-22:
I didn't reproduce it in simulator, I made a compare picture with description in it.
I think it is easy to see the issue.
[image]

kisvegabor, 2020-10-26:
That's very strange.

It'd be important to try it in a simulator to be sure it's not a hardware issue.

What if you don't register any input devices and keep pressing / not pressing the screen? If you still see the issue then probably it's not related to LVGL. Maybe it's a trick hardware issue.

closed 2020-11-26 by kisvegabor