lvgl/lvgl #1858: Stride implementation

kisvegabor, 2020-10-21:
@jozba pointed out that the concept of "stride" is missing from LVGL but it's required for some GPUs.

I've opened this issue to discuss the requirements and need for this feature. 

@jozba could you summarize what we know so far?

jozba, 2020-11-01:
Usual approach to buffer processing is to use buffer width and height in pixels. With limited resources in embedded world, it makes sense to save every possible byte of memory. Also, most CPUs have byte access to any address. However, when HW accelerators are to be involved, new limitations arise here, which are the tax for offloading the CPU. 

1) Stride. HW accelerators (GPUs, VPUs, blitters, LCD controllers etc.) usually require buffers to meet some width requirements. This is called stride. For instance, HW can require a buffer stride to be multiple of 16 pixels. E.g. if buffer real data width is 100 px, allocated width has to be 112 px in order to be accessed correctly by HW. One may look at this as 12 pixels in each line is wasted in this case, but that's the cost for HW not being forced to access or process non-burst-sized chunks of memory. Some accelerators require even height stride (e.g. GPU working in tiled mode of 4x4 squares require height to be multiple of 4 px). 

If all SW algorithms use +stride to move "one line up", instead of +width, the buffer can be safely processed also by HW accelerators. In case of no stride requirements, stride can be configured to be "multiple of 1 px", meaning width = stride. 

This will affect also web image generator and other tools, that will need to generate c-arrays that meet the stride requirement, and also add information to image descriptor strucure, what is real data width, and what is actual stride. 

2) Alignment. Due to performance and design reasons, some HW blocks require the buffer starting address to be at some particular boundary. E.g, on 16 or 32 byte boundary. Even if there is no such requirement, aligned memory access is faster. 

For this reason, it would be good to have buffer allocator, that can platform specific. It could be used for allocating buffer that will be accessed by HW and will ensure that buffer will meet stride and alignment requirements.

kisvegabor, 2020-11-05:
Thanks for the explanation. For the display sizes, the width criteria usually meet by default. I.e. all comment widths (240, 320, 480, 800) are multiple of 16. 

However, it's a real issue with images, they can have any size. 

What if we simply want to fill an area with a color. E.g. the area is (7;1) (21;88). Is there some kind of requirements here as well?

jozba, 2020-11-05:
Buffer width of multiple 16px was just an example. It can be even 32px. But it doesn't matter if we have a generic approach to buffers that takes into account the stride and real width.

I have not seen any size limitations for features like fill, blend etc. yet. Actually this would render them useless, if they would be capable of drawing only from 16 px boundaries. So only buffer start address and stride has requirements, but operations inside the buffer can have any area. 

In case you need to draw to child area that starts on address e.g. 0x1111_1112, and from offset [7, 1], where child width doesn't meet requirements, as well as starting pointer, you need to provide frame buffer (screen) related coordinates and pointer. So your FB starts at e.g. 0x1110_0000 (perfectly fits requirements) and screen-relative coordinate of drawing is e.g. [51, 66] (not limited at all).

In fact, **stride concept is used on several places in LVGL** already :-). When rendering uses ``disp_buf_tmp += disp_w;`` instead of something like ``disp_buf_tmp += child_window_size;`` - thats the main idea. Width of interest is still child width, but for buffer Y computation the stride value is used, in this case screen width. This need to be expanded into generic way for all (drawing) buffers, so any buffer could be accessed by HW accelerator.

kisvegabor, 2020-11-09:
Thanks.

So, normally LVGL draws into the buffer(s) you provide when you initialize the display driver. There are 2 requirements:
- aligned start: can be managed when the buffers are allocated. LVGL might offer some support for this.
- aligned line width: 
  - [Here](https://github.com/lvgl/lvgl/blob/85c5e786a4a41257d03a701828626ad77514b941/src/lv_core/lv_refr.c#L410) when the number of lines (height) of the drawing buffer is calculated we can ensure that `w` is aligned. 
  - Besides [real area size](https://github.com/lvgl/lvgl/blob/85c5e786a4a41257d03a701828626ad77514b941/src/lv_core/lv_refr.c#L453-L456) we need to store the stride in `lv_disp_buf_t`
  - When the drawing algorithms want to go to the next line of the buffer using `stride` instead of `lcv_area_get_widht(&disp_buf->area)`

If that's all then it seems doable and not too complicated. Just a bunch of refactoring and testing.

kisvegabor, 2020-11-26:
@jozba can you confirm my above comment?

jozba, 2020-11-27:
Hi Gabor, 

sorry for late answer. I think this change will invole more places to update. 

1) stride need to be added to all images/buffers that could be accessed by hw. So not only `lv_disp_buf_t`, but also into [lv_img_header_t](https://github.com/lvgl/lvgl/blob/85c5e786a4a41257d03a701828626ad77514b941/src/lv_draw/lv_img_buf.h#L136). 
2) With this in mind, web based image generator need to be updated as well with new configurable stride item. 
3) any buffer handling like [lv_img_buf_alloc](https://github.com/lvgl/lvgl/blob/85c5e786a4a41257d03a701828626ad77514b941/src/lv_draw/lv_img_buf.h#L201) or [_lv_mem_buf_get](https://github.com/lvgl/lvgl/blob/85c5e786a4a41257d03a701828626ad77514b941/src/lv_misc/lv_mem.c#L486) if the buffer could be used by HW. (Or eventually use `lv_img_buf_alloc` instead of generic `_lv_mem_buf_get` in algorithms).
4) your last point perfectly fits - all algorithms need to go for stride, instead of width, when going to next line. This is the point I see most complex, as there are multiple places that need to be changed, and this also involves much more testing, than previous points.

kisvegabor, 2020-12-01:
Clear, thank you. 

We already have an extensive list for v8 but in v9 I'm planning some visual/rendering updates anyway so it fits well into v9.
I've just added it to the [ROADMAP](https://github.com/lvgl/lvgl/blob/master/docs/ROADMAP.md). 

I close this issue now. Let's warn up this topic when we start to work on v9. If meanwhile there are any remarks or ideas feel free to comment here.

nicusorcitu, 2022-12-08:
@kisvegabor Is the stride implementation going to be done for v9?
Without stride concept, the vg-lite can not handle images not aligned to 16px.
Thanks, Nicu

kisvegabor, 2022-12-08:
Yes, I prepared the new function in v9 to have a stride parameter too. 

There is something that I couldn't decide: the conventional unit for stride is byte or pixel?

nicusorcitu, 2022-12-08:
The definition would say to be in bytes:
https://learn.microsoft.com/en-us/windows/win32/medfound/image-stride
The GPU also use the pitch value in bytes.

BUT
The stride value in bytes is dependent of the LV_COLOR_DEPTH, so the value will be different on 16/24/32 bits color depth for the same pixels value. The width is in pixels for example, we might keep it the same for stride in the LVGL code. Any calculation invoking stride will be much easy with the value in pixels.

So my vote is for pixels value.

kisvegabor, 2022-12-11:
I also think that the more conventional byte unit would be better, but there is a tricky case:  imagine there is an indexed 2 bit format image with 10 pixel width. In this case the stride is 2.5 byte if there is no padding on the line ends.

kisvegabor, 2022-12-12:
> So in this case, the stride should be 16 px * 2 bits = 4 bytes.

I think it should be 3 bytes (2.5 bytes rounded up)

This problem exists only if there is no stride on "bit level" formats. For example with an indexed 1 bit format and 9 pixel width, padding to 2 bytes (16 bits) mean 77% overhead. 

I'm not sure we need to support these tightly packed formats though. 77% is a lot but with a 9x10 image it just 20 bytes instead of 12. So +8 bytes. Not a big deal.

nicusorcitu, 2022-12-12:
I just checked the VG-Lite doc, and for a linear buffer, the source image in case it is indexed (1, 2 or 4 bits per pixel) needs to be aligned at a stride of 8 bytes. So is not 4, but 8 bytes :)

> I think it should be 3 bytes (2.5 bytes rounded up)
>
I do not understand why a round up value can help on stride requirements. 3 bytes at 2bits/px means 12px. Usually the GPUs stride alignments are to 8px or 16px.

The stride alignment applies on the source image. It will be done only once when the image is copied on memory buffer. It should not affect the performance.

kisvegabor, 2022-12-12:
> I do not understand why a round up value can help on stride requirements. 3 bytes at 2bits/px means 12px. Usually the GPUs stride alignments are to 8px or 16px.

I meant this only theoretically. So if we have stride, all fractional bytes are rounded up at least to the nearest neighbor. In a worse case to the 4th or 8th neighbor.  

> Usually the GPUs stride alignments are to 8px or 16px.

However without GPU due to the fact that there is stride we should round up to at least to the next neighbor. 

> The stride alignment applies on the source image. It will be done only once when the image is copied on memory buffer. It should not affect the performance.

I'm not worried about the performance but the memory usage.

nicusorcitu, 2022-12-13:
But should not be so many cases, the input images (source images) are not so many and usually those are not kept in internal RAM but on SDRAM or FLASH (if available).

Honestly the stride in pixels would help me better than the one in bytes. At the end is up to you to pick the reference value.

kisvegabor, 2022-12-13:
Okay, let's keep both option in mind and we will see if any remarks come up.

Until that I'll keep the pixel unit.

nicusorcitu, 2023-04-10:
Hi @kisvegabor is the stride going to be included in lvgl v9?

kisvegabor, 2023-04-11:
Yes, definitely. I've prepared all the SW image draw function to support it. We need to add stride the generated images too, but the image decoding/management will be also updated in v9, so we can take care of it as well.

Stride is required only on for the source buffer, right? (So the image). Or shall we consider it on other places too? Maybe on the layer's buffer?

nicusorcitu, 2023-04-11:
For liniar format, only the Source image requires memory alignament and stride.
For tiled format, the alignament and stride is requires for destination also (layer buffer).
________________________________
De la: Gabor Kiss-Vamosi ***@***.***>
Trimis: Tuesday, April 11, 2023 12:41:03 PM
Către: lvgl/lvgl ***@***.***>
Cc: Nicușor Cîțu ***@***.***>; Comment ***@***.***>
Subiect: [EXT] Re: [lvgl/lvgl] Stride implementation (#1858)

Caution: EXT Email

Reopened #1858<https://github.com/lvgl/lvgl/issues/1858>.

kisvegabor, 2023-04-12:
Okay, it's not a big problem to handle stride on the destination buffer too.

kisvegabor, 2023-10-26:
Stride support seems working so I close this issue.

closed 2023-10-26 by kisvegabor