lvgl/lvgl #1860: Can't build LVGL without including `lv_conf.h`

C47D, 2020-10-22:
> # Important: issues that don't use this template will be ignored/closed.

**Describe the bug**

When trying to build LVGL wihout `lv_conf.h` it fails because `lv_gpu_nxp_pxp_osa.c` and `lv_gpu_nxp_pxp.c` includes it. I'm building LVGL wihout `lv_conf.h` because I'm working on Kconfig support (configuration on `lv_conf_internal.h`).

**To Reproduce**

Pull [this](https://github.com/C47D/lvgl/tree/esp-idf-aware) branch of my local LVGL repo and try to build it.

**Expected behavior**

LVGL being able to compile without `lv_conf.h` included.

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Additional context**

I've locally fix it (see 903ea8bd) with the following pattern but I don't know if it's the best way to do it:

``lv_gpu_nxp_pxp.c``:
```c
#if defined __has_include
#  if __has_include("lv_conf.h")
#   include "lv_conf.h"
#  else
#   include "lv_conf_internal.h"
#  endif
#endif
```

kisvegabor, 2020-10-22:
Hi,

I've just fixed it. Inside LVGL `lv_conf_internal.h` should be included. 

@C47D I also added [lv_conf_kconfig.h](https://github.com/lvgl/lvgl/commit/c03160abc2ca26805513d2e8c6f4b92e10a6a91a) to handle special Kconfig options.

closed 2020-10-22 by kisvegabor

C47D, 2020-10-23:
`lv_gpu_nxp_pxp_osa.h` also includes `lv_conf.h`directly, I guess it should be also updated?

kisvegabor, 2020-10-26:
I've just updated it as well.

sharpcall, 2021-03-19:
This error is still there. Just did a clone and here:

/lvgl/lvgl/examples/widgets/chart# gcc lv_example_chart_
gcc: error: lv_example_chart_: No such file or directory
gcc: fatal error: no input files
compilation terminated.
root@quotestreamer1:/home/programmer/ibkrstreaming/lvgl/lvgl/examples/widgets/chart# gcc lv_example_chart_
lv_example_chart_1.c   lv_example_chart_2.c   lv_example_chart_4.c   lv_example_chart_6.c
lv_example_chart_1.py  lv_example_chart_3.c   lv_example_chart_5.c
root@quotestreamer1:/home/programmer/ibkrstreaming/lvgl/lvgl/examples/widgets/chart# gcc lv_example_chart_5.c
In file included from ../../../src/misc/lv_log.h:16,
                 from ../../../lvgl.h:25,
                 from lv_example_chart_5.c:1:
../../../src/misc/../lv_conf_internal.h:40:14: fatal error: ../../lv_conf.h: No such file or directory
   40 | #    include "../../lv_conf.h"                 /*Else assume lv_conf.h is next to the lvgl folder*/
      |              ^~~~~~~~~~~~~~~~~
compilation terminated.

embeddedt, 2021-03-20:
@sharpcall You need more components besides the LVGL repo; I suggest looking into one of the [PC simulator projects](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html) as an example of how to set things up.

silverdragon17, 2022-03-30:
Hi,
I'm having this error.

[image]

I've checked the lv_conf_internal.h file and it's as per updated code mentioned above. How do I solve it?

kisvegabor, 2022-03-30:
Are you using the latest version of LVGL?

Sarah-C, 2022-07-23:
It's back again.

I've just run the new "factory" build for Lilygo T-QT...

                from C:\Users\sarah\Documents\Arduino\sketches\T-QT\factory\factory.ino:5:
c:\users\sarah\documents\arduino\libraries\lvgl\src\lv_conf_internal.h:41:18: fatal error: ../../lv_conf.h: No such file or directory
         #include "../../lv_conf.h"                /*Else assume lv_conf.h is next to the lvgl folder*/
                  ^~~~~~~~~~~~~~~~~
compilation terminated.

kisvegabor, 2022-07-24:
Have you done [these](https://docs.lvgl.io/master/get-started/platforms/arduino.html#configure-lvgl) steps?

commit 200d3aa2 (2020-10-22): fix build with out lv_conf.h
Fixes: #1860