lvgl/lvgl #1861: How to use it with Arduino IDE?

albertomercurio, 2020-10-22:
Hello,
I installed this library from Arduino IDE, like the deprecated version lv_arduino. There are not examples, but i can see a .ino file in the directories of the library called LVGL_Arduino.ino. But when i try to compile it, it returns
```
In file included from /home/alberto/Arduino/libraries/lvgl/src/../src/lv_misc/lv_log.h:16:0,
                 from /home/alberto/Arduino/libraries/lvgl/src/../lvgl.h:26,
                 from /home/alberto/Arduino/libraries/lvgl/src/lvgl.h:17,
                 from /home/alberto/Arduino/libraries/lvgl/examples/LVGL_Arduino/LVGL_Arduino.ino:1:
/home/alberto/Arduino/libraries/lvgl/src/../src/lv_misc/../lv_conf_internal.h:24:101: fatal error: ../../lv_conf.h: No such file or directory
compilation terminated.
exit status 1
Errore durante la compilazione per la scheda ESP32 Dev Module.
```

kisvegabor, 2020-10-26:
Hi,

Have you followed the guide here?
https://docs.lvgl.io/latest/en/html/get-started/arduino.html#configure-lvgl

albertomercurio, 2020-10-26:
> Hi,
> 
> Have you followed the guide here?
> https://docs.lvgl.io/latest/en/html/get-started/arduino.html#configure-lvgl

Yes.

kisvegabor, 2020-10-27:
And do you really have `lv_conf.h` next to the lvgl folder in the Arduino's library folder?

albertomercurio, 2020-11-03:
Yes.

kisvegabor, 2020-11-04:
Please send [tree view](https://www.codeproject.com/KB/tree/FileSystemTreeView/screenshot.gif) of the folders where lv_conf.h is located. 

And try to add this to lv_conf.h: `#error "seen by the compiler"`. Do you the error on compile?

benjaminSchilling33, 2020-11-20:
@kisvegabor I have the same problem.
I added the #error directive but it is not shown in the compilation output.

kisvegabor, 2020-11-21:
Thanks for the update!

closed 2020-11-21 by kisvegabor

daxliniere, 2021-11-16:
> Hi,
> 
> Have you followed the guide here? https://docs.lvgl.io/latest/en/html/get-started/arduino.html#configure-lvgl

This link goes to the docs for 7.11.0-dev instead of the actual latest.
Also, there are no instructions for v8.0 on how to configure demos, yet I get compilation error: **lv_demo_conf.h: No such file or directory. compilation terminated.**

LVGL looks great and I can't wait to start using it! 😃

kisvegabor, 2021-11-17:
The URL has been changed meanwhile, see the latest version here: https://docs.lvgl.io/master/get-started/arduino.html

`lv_demos` should work the same way as lvgl. Have you copied `lv_demo_conf.h` next to the `lv_demos` folder?

Hrishikesh46779, 2021-11-28:
I have added lv_config.h in arduio directory but still not working.

c:\users\hrishikesh\documents\arduino\libraries\lvgl\src\lv_conf_internal.h:41:100: fatal error: ../../lv_conf.h: No such file or directory

Hrishikesh46779, 2021-11-28:
C:\Users\Hrishikesh\Documents\Arduino\libraries\lv_examples\src\lv_demo_widgets\lv_demo_widgets.c:1225:17: error: too few arguments to function 'lv_calendar_header_dropdown_create'
How to fix?

kisvegabor, 2021-11-30:
Are you sure you use the latest lvgl with the latest lv_demos. There was a change in the calendar widget and it seems the 2 repos are not in sync.

Hrishikesh46779, 2021-12-01:
I am installing lvgl and lv_examples libraries from arduino IDE.

Hrishikesh46779, 2021-12-01:
Now after downloading the both libraries from github i got another error-
undefined reference to `lv_demo_widgets'

kisvegabor, 2021-12-02:
You should use lvgl 8.1.0 too.