lvgl/lvgl #1869: LVGL as ESP-IDF component

C47D, 2020-10-26:
# All enhancement requests or questions should be directed to the Forum.

We use GitHub issues for development related discussions.
Please use the [forum](https://forum.littlevgl.com/) to ask questions.

We've been using LVGL as an ESP-IDF component on the lv_port_esp32 repo, that's accomplished by adding some "especial" files to it so it. This issue is to keep track of the changes done to LVGL so it can be used directly as a component. The Kconfig file also should work for other projects like Zephyr and Nuttx, so any feedback is welcome.

- [ ] Create an Kconfig file with all (or most) of the LVGL configuration options.
- [ ] lv_conf_kconfig: Update as necessary.
- [ ] lv_conf_internal: Include special ESP-IDF header files when working with the ESP-IDF framework. See #1864 
- [ ] lv_conf_internal: Update as necessary.
- [ ] Create an CMakeLists.txt file for (ESP-IDF) component registration. See #1868

C47D, 2020-10-26:
@kisvegabor Some LVGL configuration options depends on another options being enabled, e.g. `LV_SPINNER_DEF_ARC_LENGTH` depends on `LV_USE_SPINNER` being enabled. Should we hide `LV_SPINNER_DEF_ARC_LENGTH` in the menuconfig interface when `LV_USE_SPINNER` is not enabled? (same for all other options)

embeddedt, 2020-10-26:
Yes; I think that would make the most sense. Otherwise the Kconfig interface will be cluttered with options which don't change anything.

C47D, 2020-10-26:
Nice, I will send a draft PR later today with the Kconfig file.

kisvegabor, 2020-10-26:
I agree, it'd be better to hide them.

C47D, 2020-10-28:
@kisvegabor in `lv_conf_internal.h` I manually added `lv_color_hex` when using the `CONFIG_LV_COLOR_TRANSP` value, is this correct? If so, how can we add `lv_color_hex` when using the `lv_conf_checker.py` script?

```c
/*Images pixels with this color will not be drawn (with chroma keying)*/
#ifndef LV_COLOR_TRANSP
#  ifdef CONFIG_LV_COLOR_TRANSP
#    define LV_COLOR_TRANSP lv_color_hex(CONFIG_LV_COLOR_TRANSP)
#  else
#    define  LV_COLOR_TRANSP    LV_COLOR_LIME         /*LV_COLOR_LIME: pure green*/
#  endif
#endif
```

Also to let you know that I've tested the new Kconfig with the latest `dev` and it seems to be working as expected.

kisvegabor, 2020-10-29:
Can we handle `LV_COLOR_TRANSP` similarly to theme colors in `lv_conf_kconfig.h`?

C47D, 2020-10-30:
@kisvegabor I've pushed c2094b9 to handle `LV_COLOR_TRANSP`, it defines `CONFIG_LV_COLOR_TRANSP`, so `lv_conf_internal.h` doesn't need to change to handle that option.

C47D, 2020-11-18:
Hi @kisvegabor @embeddedt 
Should we keep this issue open for feedback or can we close it?

kisvegabor, 2020-11-18:
I suggest closing this and let people opening issues for specific issues.

I think one more step remained. We need to let people know how LVGL can be used as an ESP component. Similar to the [Arduino docs here](https://docs.lvgl.io/latest/en/html/get-started/arduino.html).

Carlos, do you have time and energy to write a few lines about it?

C47D, 2020-11-18:
I can do something about it at night.

kisvegabor, 2020-11-26:
As all tasks are done here so let's close this issue. 

Thank you @C47D for the great work! :clap: :clap: :clap:

closed 2020-11-26 by kisvegabor