lvgl/lvgl #1871: Problem with lv_img_set_offset_y in ver 7

ali-rostami, 2020-10-27:
**Describe the bug**

This issue is opened as a place to continue [this discussion](https://forum.lvgl.io/t/problem-with-lv-img-set-offset-y-in-ver-7/3330) in the forum.
This is the description:
In ver 6, I had no problem with lv_img_set_offset_y, but in ver 7, it seems that something has changed about it.
consider this picture as the source of the img object.
[image]

offset
Each part has a size of 100 * 50.
In ver 6, if I wanted to display part 7, I had to write sth like this:

lv_img_set_offset_x(img_power_status, 100);
lv_img_set_offset_y(img_power_status, 100);
But now this doesn’t work and instead shows part 6. and seems if I want to show it correctly I should write it like this:

lv_img_set_offset_x(img_power_status, 100);
lv_img_set_offset_y(img_power_status, -100);
Is it a bug or something else?

we got to the place that I expect from this code:

```
lv_img_set_offset_x(img, 170);
lv_img_set_offset_y(img, 50);
```

to display for me rectangle number 6 but I see number 8.
Now the question is that is my expectation wrong?-As I said in ver 6 it worked like this expectation-
You say @kisvegabor that you wrote these lines and see the number 6 rectangle?

kisvegabor, 2020-10-28:
Hi Ali,

In v7 it works well with this (shows 6):
```c
  LV_IMG_DECLARE(a);
  lv_obj_t * img = lv_img_create(lv_scr_act(), NULL);
  lv_img_set_src(img, &a);
  lv_obj_set_size(img, 170, 50);
  lv_img_set_offset_x(img, -170); //Make the image start at -170, so at 0 the 2nd column will start 
  lv_img_set_offset_y(img, -50);  //Make the image start at -5, so at 0 the 2nd row will start
```

This part has changed a little since v6 so some incompatibility might happen.

ali-rostami, 2020-10-28:
Common😎, that's what I'm trying to tell you.
In ver 6 we should do this:
```
  lv_img_set_offset_x(img, 170); 
  lv_img_set_offset_y(img, 50); 
```
Right? 
If yes, I didn't see anywhere refer to this.
Anyway, it seems O.K and I think we can close the comment If you say.

kisvegabor, 2020-10-28:
> Commonsunglasses, that's what I'm trying to tell you.

I see, but honestly don't know without immersing in how it worked in v6 :smile: 

It looks consistent in v7 with the negative coordinates. :slightly_smiling_face: 

> If yes, I didn't see anywhere refer to this.

Actually, I didn't know it's different hence it's not noted in the migration notes.
When v8 is released testing will have high priority so we will detect similar changes automatically.

closed 2020-10-28 by kisvegabor