lvgl/lvgl #1872: recoloring of hands when using the hands image

uraich, 2020-10-27:
> # Important: issues that don't use this template will be ignored/closed.

**Describe the bug**

According to the description in the lvgl documentation the hands of the gauge widget will be re-colored to the color defined with  lv_gauge_set_needle_count. This is not the case.

**To Reproduce**

Just run lv_ex_gauge_2.c from the lv_examples/src/lv_ex_widgets/lv_ex_gauge

**Expected behavior**

I expect the result shown in  lv_examples/src/lv_ex_widgets/lv_ex_gauge/lv_ex_gauge_2.png. However, all 3 hands are black.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**

Add any other context about the problem here.

uraich, 2020-10-27:
Sorry, I think the problem is that the documentation is not clear. In fact, if you define a style with the set_image_recolor_opa and add it to LV_GAUGE_PART_MAIN
then the gauge works as expected

kisvegabor, 2020-10-28:
Hi,

Thanks for reporting this.

I fixed the followings:
- lvgl: https://github.com/lvgl/lvgl/commit/85c80148804af2c0d8d7652100ad789292d29be8
- lv_examples: https://github.com/lvgl/lv_examples/commit/6f46b57ec3dec976f4cd1c0cfb7f73b2a3ea5049
- docs: https://github.com/lvgl/docs/commit/d633032da663e92740a56ca2e76f154d5c6091d4

uraich, 2020-10-28:
Hi Gabor,
I updated my code and now I get a compilation error on the esp32 port:
../../lib/lv_bindings/lvgl/src/lv_misc/lv_txt.h:135:48: error: unknown type name 'va_list'
I had to add
#include <stdarg.h> to fix this

kisvegabor, 2020-10-29:
Thanks, I've added this include. 

Otherwise, was the hand coloring  working?

uraich, 2020-10-30:
Yes, I modified the Python code according to your modified C code and both versions work fine. Btw: I almost finished to convert all widget examples in the lvgl docs to their micropython equivalents staying as close as possible to the C originals. I will check how to include them into the docs and make a pull request as soon as I have finished (of course only if you agree).I still have to figure out how you manage to execute your programs within sphinx and include the result on the html pages.Sent from my Samsung Galaxy smartphone.

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/lvgl/lvgl/issues/1872#issuecomment-719051271",
"url": "https://github.com/lvgl/lvgl/issues/1872#issuecomment-719051271",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

embeddedt, 2020-10-30:
> I still have to figure out how you manage to execute your programs within sphinx and include the result on the html pages.

**TL;DR** Magic. :slightly_smiling_face: There's quite a bit of complication behind the scenes because I have yet to find a way to compile C in the browser without a server. Fortunately, we don't have this limitation with MicroPython, so it would make more sense to use the online simulator than to limit ourselves to an iframe. That means that all you should really need to do is push your example .py files to `lv_examples`. After that, I can just make the directive include a link to the MicroPython version as well if it is available.

-------------------------------

I made a Sphinx extension that adds a new custom directive (`lv_example`). When that directive is [used in an example file](https://raw.githubusercontent.com/lvgl/lv_examples/master/src/lv_ex_widgets/lv_ex_btnmatrix/index.rst) (each widget includes all of its examples), it injects a 320x240 iframe and the code snippet into that part of the HTML.

Then there is an additional layer of complexity because we need the examples to be compiled from C to HTML and stored in the docs repository itself. For this I used a GitHub action. Every time a change is made to `lv_examples/master` GitHub rebuilds Emscripten versions of all the examples and pushes them to `lv_examples/gh-pages` (which can be viewed online [here](https://lvgl.github.io/lv_examples/)).

The final step is to have the docs repository check out the appropriate commit of `gh-pages`, which I ended up doing by modifying [`conf.py`](https://github.com/lvgl/docs/blob/a273b67673d0d0ec4e45adfc4a3f810057c51a15/conf.py#L231-L251).

uraich, 2020-10-30:
Thanks for the explanation. I will have a close look into it to understand the magic.In fact there is no need to execute both, the C and the micropython version as the result will be the same anyway.I had some difficulty with the imgbtn example because the images are included in the C source code. I modified the C image sources such that they write the pixel values to a binary file, which I can easily read with my micropython code.Another problem were the colors, which are not defined in the python bindings. I added a lv_colors.py file which defines them.Sent from my Samsung Galaxy smartphone.

TL;DR Magic. 🙂 There's quite a bit of complication behind the scenes because I have yet to find a way to compile C in the browser without a server. Fortunately, we don't have this limitation with MicroPython, so it would make more sense to use the online simulator than to limit ourselves to an iframe. That means that all you should really need to do is push your example .py files to lv_examples. After that, I can just make the directive include a link to the MicroPython version as well if it is available.

kisvegabor, 2020-10-30:
@uraich 
>  Btw: I almost finished to convert all widget examples in the lvgl docs to their micropython equivalents staying as close as possible to the C originals

Wonderful!

> In fact there is no need to execute both, the C and the micropython version as the result will be the same anyway

Theoretically, it's possible to have different MP examples so it'd be better to conceptually support "real" MP builds. Of course only if it's feasible with a reasonable amount of work. 

>  I added a lv_colors.py file which defines them.

@amirgon What do you think about `color.py` to add the very basic 16 colors?

embeddedt, 2020-10-30:
For standalone MP examples I think a manually-added link to the online simulator is sufficient. The simulator is superior to the iframe approach anyways.

uraich, 2020-10-30:
Sorry, where do I find color.py?

amirgon, 2020-10-30:
> In fact there is no need to execute both, the C and the micropython version as the result will be the same anyway

One advantage of running the Micropython examples is that you can let the user interactively change the example code online and see how it affects the results. Like the "Try it yourself" buttons on https://www.w3schools.com/.
@embeddedt  Perhaps it could be possible to embed the simulator in the doc pages, such that Micropython examples would become interactive?

> What do you think about `color.py` to add the very basic 16 colors?

If the basic colors don't change a lot, that's a good idea. You can open a PR to add it under [lib](https://github.com/lvgl/lv_binding_micropython/tree/master/lib).  

Another option is to add static inline functions in LVGL that return the colors [defined by the color macros](https://github.com/lvgl/lvgl/blob/a2b4bbcd2a997a874ab924fa0dd31820964233ea/src/lv_misc/lv_color.h#L37-L54). If such functions are available they would automatically be available on Micropython and would not cost anything on the C side (cause they are static inline).   
The advantage of that is that we won't need to maintain the colors in two separate files (`color.py` and `lv_color.h`).

embeddedt, 2020-10-30:
> Perhaps it could be possible to embed the simulator in the doc pages, such that Micropython examples would become interactive?

I was thinking of having a "Try it yourself" link that would go to a normal online simulator URL. Does that work?

amirgon, 2020-10-30:
> I was thinking of having a "Try it yourself" link that would go to a normal online simulator URL. Does that work?

Sure that would work too.
On the other hand, an embedded simulator would save that extra click and display the Micropython code as part of the doc page without the need for another indirection.

embeddedt, 2020-10-30:
There isn't any extra work to embed it vs link to it (since I already embed the C examples). The main concern I have with embedding the full simulator is that there's nowhere near enough space on the page, even on my 27" monitor. My other concern is that there could be 4 Emscripten instances active on a single widget page at one time (e.g. 2 Python and C examples per widget).

amirgon, 2020-10-30:
> The main concern I have with embedding the full simulator is that there's nowhere near enough space on the page, even on my 27" monitor.

That would require changing the layout of an embedded simulator to be a bit more compact

> My other concern is that there could be 4 Emscripten instances active on a single widget page at one time (e.g. 2 Python and C examples per widget).

No idea about that.  
If that's an issue, then your suggestion of "Try it yourself" button is good enough.

kisvegabor, 2020-11-02:
The "Try it yourself" also help to make the simulator more popular. 

> Another option is to add static inline functions in LVGL that return the colors defined by the color macros. If such functions are available they would automatically be available on Micropython and would not cost anything on the C side (cause they are static inline).

I think `lv_color_red()` for a constant color is a little bit odd. These colors are standard so they shouldn't really change at all.
So I'd prefer the custom `py` file.

uraich, 2020-11-21:
When implementing the "Try it yourself" link, how can I add micropython modules like lv_colors.py defining the colors or imagetools.py needed by some examples?

I started uploading the micropython examples to github:
https://github.com/uraich/lv_examples/tree/mpy_examples
For the moment only the styles and widget/arc examples have been uploaded. Can you please check if this is the way you want them?
I also added header.py and footer.py. You can then concatenate the files:
cat header.py example.y footer.py > runnable.py to get the example ready to be executed on your PC + SDL or your ili9341 board.

amirgon, 2020-11-21:
> When implementing the "Try it yourself" link, how can I add micropython modules like lv_colors.py defining the colors or imagetools.py needed by some examples?

In the [online simulator](https://sim.lvgl.io/v7/micropython/ports/javascript/bundle_out/index.html), you can import file from the web.
So if the python modules are on GitHub ([under lib](https://github.com/lvgl/lv_binding_micropython/tree/master/lib) for example) you simply import them.  

You tell Micropython where to look for modules by setting the `sys.path` global, so code on the simulator usually starts like this:

```python
import imp, sys
sys.path.append('https://raw.githubusercontent.com/littlevgl/lv_binding_micropython/master/lib')
```

Specifically regarding imagetools.py, it is not supported on the simulator because unfortunately the javascript micropython port does not implement the `native` and `viper` code emitters. But you probably don't need imagetools.py for the micropython examples.

Regarding `lv_colors.py`, you are welcome to open a PR to add it under `lv_binding_micropython/lib`.
I noticed that you defined a different global variable for each color. Instead, I suggest defining a single "lv_colors" global object with colors as members. The advantages are less pollution of the global namespace and providing the ability for the user to introspect available colors (with `dir(lv_colors)`, `hasattr`, etc.).

> I also added header.py and footer.py. You can then concatenate the files:
> cat header.py example.y footer.py > runnable.py to get the example ready to be executed on your PC + SDL or your ili9341 board.

I think the goal is to run them on the [Micropython online simulator](https://sim.lvgl.io/v7/micropython/ports/javascript/bundle_out/index.html), so the header and footer are not needed.  If I remember correctly the simulator can automatically add a header when it's given some parameter in the URL.

@embeddedt 
- Do you remember the URL parameters we can give the simulator? I don't remember where it's documented.
- Not all of @uraich scripts can run on the simulator because it is not up to date. Could you update it? Maybe it's worth updating it automatically with a GitHub action?

embeddedt, 2020-11-21:
> Maybe it's worth updating it automatically with a GitHub action?

This link should update with `master`: https://lvgl.github.io/lv_micropython/bundle_out/index.html

sim.lvgl.io isn't set up to auto-update yet because communicating across repositories is somewhat more complex. I didn't feel like digging into that right away, possibly tomorrow.

> Do you remember the URL parameters we can give the simulator?

https://github.com/lvgl/lvgl/issues/1320#issuecomment-586746524

uraich, 2020-11-22:
For me as an end user of lv_micropython, running the examples on the simulator is only a goody. What I really need is examples that show me how to implement an lvgl micropython program on my bare metal device. For this reason I think that header and footer can be helpful.
I tried to write some micropython code for a lolin t-watch 2020 without having used neither the C nor the micropython version of lvgl before. When experimenting with this I saw that I had to figure out calling sequences be analysing lv_mpy.c and that the few micropython example programs were not sufficient for me. For this reason I tried to produce a micropython version of all C example programs with a very good learning effect and plenty of questions to all of you, which you all patiently answered. Thanks!
I will now try to run my examples on the simulator one by one.
Concerning imagetools.py: This is used in the style, img and imgbtn examples, which also use external png images. For the moment I don't know how I could implement these on the simulator.

uraich, 2020-11-22:
After addition of lv_colors to the lv_bindings lib I just tried to run lv_ex_style_1.py in the simulator and it works as expected.

amirgon, 2020-11-23:
> What I really need is examples that show me how to implement an lvgl micropython program on my bare metal device

Obviously the end goal is to run it on a device.
For each device/display you would need different code to register the HW specific drivers, but the rest of the code (where you actually create and interact with the GUI) is not HW specific and would be exactly the same on all devices and on the simulator.

It's a good idea to add a section to the documentation that explains/gives examples of registering different drivers on different devices, but I don't think this should be part of the widget/style examples.
The widget/style examples should be independent from which platform / display you are using, so there is no point in adding HW specific prefix/suffix to every example.

> Concerning imagetools.py: This is used in the style, img and imgbtn examples, which also use external png images. For the moment I don't know how I could implement these on the simulator.

For some reason, lodepng (PNG decoding library) does not work on the simulator.
@embeddedt any idea why? Here is [an example](https://sim.lvgl.io/v7/micropython/ports/javascript/bundle_out/index.html?script_direct=c7b456f7a5ddcd6120858e898937b39bfd2810c5).  
 It gets stuck for a long period of time and then gives an out-of-memory error: `MemoryError: memory allocation failed, allocating 166914 bytes`.  The same code works well on the unix or ESP32 port.

@uraich instead of decoding PNG, you can use raw images in the examples. Raw images work well on the simulator. 
[Here is an example script that uses raw image](https://sim.lvgl.io/v7/micropython/ports/javascript/bundle_out/index.html?script_direct=f7c4554d14dd7fbb9ac663285b5b9a75a4320fa4).

uraich, 2020-11-23:
Of course you are right saying that there should be no hardware specific code in each example. The driver registration is done in the init_gui script and you need a few lines of code to import it and a few lines of code at the end of the script for an endless loop. I therefore separated the programs into a header, the main code and a footer. The main code will be the example, while we could provide the header and footer in the documentation of the HW specific part. In order to run the examples on the hardware all you would need  is uploading the init_gui script and concatenating header,example and footer. 
I started to prepare the PR and you can see it in https://github.com/uraich/lv_examples/tree/mpy_examples
For the moment only the examples for lv_ex_style and lv_ex_arc have been uploaded but I guess that by the end of the week I should be able to upload the others.
Thanks for the example using the raw image. I will try to re-write the image examples such that they run on the simulator.

closed 2020-12-20 by uraich