lvgl/lvgl #1874: Widget type reflection support

puzrin, 2020-10-29:
Current widgets have no reflection support. It's impossible to detect "object type" by generic `lv_obj_t*` pointer.

This cause "hacks" like in themes https://github.com/lvgl/lvgl/blob/master/src/lv_widgets/lv_btn.c#L95, with hardcoded global constants. Why global? Because if you add foreign widgets, those constants still have to be globally unique.

What is suggested:

1. Create explicit type id, and store it in each `lv_obj` struct.
2. To protect from collisions, add `register_widget_type()` method, and throw/message if duplicate found. I don't know if this can be checked in compile time. But runtime check is still much better than nothing.
   - you can call it lazily on first create, if user forgot that. Or error on attempt of unregistered widget create.
3. In worst case, if 3rd-par widgets have collision - use guards to redefine type value via `-D`

That will cost +1 byte per `lv_obj`, but will allow to implement some "impossible" things:

- Implement generic fabric for any-widget-create.
- Implement delegated event listener (global one on root node, instead of local on each item).

Now this can be hacked via analyse of widget-specific class (check if exists), assigned by theme. But that's not normal to hack reflection via theme's kludge, instead of implicit type announce.

kisvegabor, 2020-10-30:
Actually, there is a mechanism to get the widget type. Every widget can handle `LV_SIGNAL_GET_TYPE` and put their type into a list. It works for the ancestor types too. So finally for `lv_slider` you get a list like this: "lv_slider", "lv_bar", "lv_obj".

It's important, because if asserts are enabled API functions check the type of the widget. E.g if you pass a button to `lv_slider_set_value` it will assert. However, widgets API functions intensively use the ancestor type's functions. E.g.
```c
void lv_slider_set_value(lv_obj_t_ * slider, int value) 
{
  lv_bar_set_value(slider, value);  //Slider has obj+bar+slider type so it won't assert
} 
```

The current solution works well but it's not user-friendly. To add a new type for the object you need:
1. Save the current signal function (it will be the ancestor signal function)
2. Write you the new signal function and handle `LV_SIGNAL_GET_TYPE`
3. Call the ancestor signal function too

So I can see 2 ways:
1. Use the current mechanism
2. Get rid of getting the ancestor types and use 1 byte as you suggested (at the cost of losing this type of assert)

> Implement delegated event listener (global one on root node, instead of local on each item).

Do you mean using event bubbling and in the ancestors' event handler check the type of the sender object?

puzrin, 2020-10-30:
> Actually, there is a mechanism to get the widget type. Every widget can handle `LV_SIGNAL_GET_TYPE` and put their type into a list.

I'd prefer something more useable :). 1 byte seems reasonable price for simplification.

> (at the cost of losing this type of assert)

Type value will be at struct start. Seems it can be asserted when required. Do i miss something?

The only disadvanage - such primitive implemenation does not allow to check directly full inheritance chain. But who cares?

> Do you mean using event bubbling and in the ancestors' event handler check the type of the sender object?

Yes. When you sit of top, you receive a lot of events, from known and unknown widgets.

embeddedt, 2020-10-30:
> The only disadvanage - such primitive implemenation does not allow to check directly full inheritance chain. But who cares?

It matters right now. In @kisvegabor's example above, `lv_slider_set_value` calls `lv_bar_set_value`. If there is no inheritance data available, when `lv_bar` sees a slider as the object type, it will assert (since it doesn't know about that type).

kisvegabor, 2020-10-30:
Yes, exactly as @embeddedt says. So we need to choose between 
1. simple type settings
2. dropping this "widget type assert"

Or what about this: 
When `lv_bar` extends `lv_obj_t` with the bar's data (range, value, etc) it also adds a `type` field. When `lv_bar_set_value` is called it checks the `type` filed among the bar's data.

If a slider is created from the bar, another type of field is used among the slider's data, independently from the bar's `type` filed. 

For example:
```c
obj->ext->bar.value;
obj->ext->bar.range;
obj->ext->bar.type;
obj->ext->slider.foo;
obj->ext->slider.bar;
obj->ext->slider.type;
```
Hence every widget can check the types it knows. It's an internal data and widget can do whatever they wants with it.

puzrin, 2020-11-01:
After thinking more, it seems all "problems" are caused by "missed" abstraction layer. In terms of CPP - missed class definition (current struct-s extend data only, without methods and other class info).

For example https://github.com/lvgl/lvgl/blob/master/src/lv_core/lv_obj.c#L299-L300 `event_cb` & `signal_cb` are class-specific methods, not needed in every instance object. That saves us 8 bytes, and adding 4 bytes of class pointer is not a problem. This also solves initial issue with types resolve.

Reading:

- https://www.cs.rit.edu/~ats/books/ooc.pdf
- http://ooc-coding.sourceforge.net/
- https://github.com/CObjectSystem/COS
- https://github.com/ShmuelFine/COOP
- more: https://stackoverflow.com/questions/351733/how-would-one-write-object-oriented-code-in-c

kisvegabor, 2020-11-05:
It's a good idea to make the signal and design callbacks (event_cb is set by the user) depending on the type of the widget. 
It still requires knowing all the ancestor types to call their callbacks too. It might be hardcoded though, e.g.
```c
lv_res_t lv_slider_slignal(...) 
{ 
   lv_bar_signal(...)
}
```
This way when a signal is sent only the "top type" needs to be known.

One more thing here: the user needs to be able to change (extend) the signal or design function. Although, it is possible to make it required to register a new widget type with the custom signal/design callback. 

Saving 8 bytes would be great. It's about 8% saving. (every 13th object is "free").

puzrin, 2020-11-05:
> One more thing here: the user needs to be able to change (extend) the signal or design function. Although, it is possible to make it required to register a new widget type with the custom signal/design callback.

I suggest follow formal logic:

1. All types of inheritance (classes, prototypes, ...) are already known and investigated
2. Method of extension depends directly on inheritance type.

So, in case of classes - way to customize handlers is to create inherited class with overridden virtual methods. Some libs provide automatic class registration.

---

What i expect:

- Change existing "freestyle" :) approach to one of existing formal inheritance type.
- If "classes" are ok - inspect existing "Object Oriented C" libs and see what we can use.

kisvegabor, 2020-11-06:
The main requirement here is to make it possible for ancestor types to also react to signals. E.g. when pressed `obj`, `bar` and `slider` also needs to be modified.

Maybe it's only my ignorance, but I don't see how it can be solved in an intuitive way in higher-level languages.

puzrin, 2020-11-06:
Mmm... what's wrong with standard CPP-like inheritance? If you inherit class, you get all parent behaviors by default (including message handlers). Then you can replace or extend existing methods. Seems i don't understand your question. Could you rephrase?

If you are completely lost, take a look at this lib sources for start http://ooc-coding.sourceforge.net/. It's code looks human-friendly.

kisvegabor, 2020-11-09:
> Mmm... what's wrong with standard CPP-like inheritance?

Nothing actually, if the signal functions directly call the ancestor type's signal function.  Simply it can mean that before changing the bar's type to slider, slider get's the current signal function, saves it, and calls it in `slider_signal()`. See the current implementation [here](https://github.com/lvgl/lvgl/issues/1874#issuecomment-722301696) and [here](https://github.com/lvgl/lvgl/blob/master/src/lv_widgets/lv_slider.c#L258).

So in summary, if can tell only the "final" type:
- advantages
  - save 2 x 4 bytes per object (signal and design function) and store them per type.
  - simple, easy to understand by the user
- disadvantages
  - we lose the opportunity on asserting on wrong widget type in API functions
  - loose 1 byte per widget (store the type)
  - widget types can be registered

For API something like this seems suitable:
```c
uint8_t my_type = lv_widget_regsiter_type(signal_cb, design_cb);
```

---

ooc seems really cool !

puzrin, 2020-11-09:
> Nothing actually, if the signal functions directly call the ancestor type's signal function. Simply it can mean that before changing the bar's type to slider, slider get's the current signal [...]

Let's stay aside "byte issues". Advantage is, that any OOP-emulaion lib simplifies this process for "methods". It's the same struct-based "inheriance" + steroids.

> For API something like this seems suitable:
> 
> ```c
> uint8_t my_type = lv_widget_regsiter_type(signal_cb, design_cb);
> ```

With "emulated classes" no need for callbacks register. You will call widget's `.on_signal()` or `.on_draw()` method "direclty". And since class info exists in every struct, proper function will be invoked.

> ooc seems really cool !

Yeah. I'm not sure, how "modern" is it, so i suggested several libs + book (already filtered all garbage). Sorry, this may take some time to read, but that seems minimal possible set to understand situation well.

kisvegabor, 2020-11-11:
> With "emulated classes" no need for callbacks register. You will call widget's .on_signal() or .on_draw() method "direclty". And since class info exists in every struct, proper function will be invoked.

I didn't get it. How would `bar.on_signal()` know what function to call if we don't tell `bar` type uses `lv_bar_signal`?

puzrin, 2020-11-11:
That depends on concrete lib.

For virtual methods, each class has vtable + struct with method names (same for all class chain). Instead of `bar.on_signal()` you will have to write something like 

```c
*BarClass(*lv_obj->class_ref)->virtual->on_signal(param1, param2, param3)
// or macro
INVOKE_VIRTUAL(lv_obj, on_signal, param1, param2, param3)
```

That's why classes should be "registered" (initialized prior to use) - to fill vtables.

I'm not sure i'm qualified enough to explain you right all possibilities. I would not like to be responsible for telling your wrong things in area of fundamental programming paradigms. It would be better if you dig libs and book from https://github.com/lvgl/lvgl/issues/1874#issuecomment-720069263. That will take 1-2 full days + 1-2 weeks pause to let brain structurize info in background.

kisvegabor, 2020-11-12:
> *BaseClass(*lv_bar_obj->class_ref)->virtual->on_signal(param1, param2, param3)

Okay, I understand all these but please "demonstrate" this: get the `bar`'s signal (superclass of slider) from a `slider` object. 

```c
*BaseClass(*lv_slider_obj->class_ref)->virtual->on_signal(param1, param2, param3)
```
will give the `slider_cb`.

puzrin, 2020-11-12:
> Okay, I understand all these but please "demonstrate" this: get the `bar`'s signal (superclass of slider) from a `slider` object.

`.super` call may be needed ONLY from concrete class implementation (not from outer code). So, probably, will be:

```c
*SliderClass(*lv_slider_obj->class_ref)->(*BarClass)super->virtual-> on_signal()
// OR
*BaseClass(*lv_slider_obj->class_ref)->super->virtual->on_signal()
```

Each class know it's parent (has .vtable AND .super).

Is this what you asked?

kisvegabor, 2020-11-13:
With `super` it's ok, but `virtual` assumes that the function pointer is stored in every instance. 
With C++ terms we need a `static` function (the function is specific to the class and the same for every instance).

So your example should look like this:
```c
*BarClass(*SliderClass(*lv_slider_obj->class_ref)->super)->on_signal();
```

Anyway, I think we are on the same page now and it can be implemented with acceptable effort.

But why we want to change the current method? :smile:  What I see as an advantage of the new approach:
- easier to add new types
- cleaner what "type" means (always the top-level type)
- save 8 types per widget
- faster to get the type

The only disadvantage is the loss of assertation on invalid types. 

@embeddedt do you agree it's worth it?

puzrin, 2020-11-13:
> With super it's ok, but virtual assumes that the function pointer is stored in every instance.

No. "Virtual" means, other method of Base class can use this function, but it will be implemented only in child classes. For example `Figure->Square()` Figure can be triangle, circle an so on. You can operate with area size in base class, if method is vitual. 

But, for our case, most important fact is all those functions are invoked via vtables of class descriptor. You an have 100 instances, but descriptor is only one (for each unique class). vtables are cheap.

> But why we want to change the current method?

- Using well-known OOP programming patterns. More easy & less to learn. DOM-related things are more easy to describe via OOP.
- More universal experience. When user learns ooc lib - this can be reused in other projects. When user learns lvgl-specific things - this can be reused only in lvgl projects.

In short - more user-friendly :)

> save 8 types per widget

Bytes? Then 4 - save 8 for virtual methods, and add 4 for class descripor pointer. Anyway, this is not main point for switch to more "formal" OOP. I provided memory info as proof "this will be not worse".

> The only disadvantage is the loss of assertation on invalid types.

Types check is one of declared features in many ooc (object oriened c) libs.

kisvegabor, 2020-11-16:
I've examined some OOC libs and found a possible solution to make LVGL more "class"-like but still not sure it's worth it. 
Independently we can make `signal` and `design` functions public to make them directly available without getting and storing them. 

So here is the idea:
```c
/* BASE OBJ */
typedef struct {  
  //data fields
}lv_obj_t;

//to create a base obj:
lv_obj_t * lv_obj_create(lv_obj_t * parent) 
{
  lv_obj_t * obj = add_child(parent, sizeof(lv_obj_t));
  lv_obj_constructor(obj); //Init the created children
  obj->... //Further init
}

/* BUTTON */
typedef struct 
{ 
   lv_obj_t base;    //Ancestor
   //btn data
}lv_btn_t;

//to create a btn:
lv_obj_t * lv_btn_create(lv_obj_t * parent) 
{
  lv_obj_t * btn = add_child(parent, sizeof(lv_btn_t));
  lv_obj_constructor(btn); //call the ancestor's constructor
  btn->...;//Further init
  ...
}

/* CUSTOM OBJ TYPE */
typedef struct {
  lv_btn_t base;
  //custom data
}super_btn_t

lv_super_btn_t * superbtn = suprtbtn_craete(parent)
{
  lv_obj_t * superbtn = add_child(parent, sizeof(super_btn_t));
  lv_btn_constructor(superbtn); //call the ancestor's constructor which will call obj_constructor too
}
```

Now, when you create a `btn` and want to add some extra data you can use `lv_obj_allocate_ext_attr(btn, sizeof(super_btn_ext_t));` . It's a custom method (I mean not coming from an existing paradigm) but simple and dynamic. (means extra members can be added any time after creation).

If we try to move closer to CPP and use an existing object-oriented C library or create our own I feel that things get even more complicated.

See detailed answers. Sorry for being so rejective but wouldn't like to take this complex task unless it's really required and brings true benefits.

> Using well-known OOP programming patterns. More easy (for methods) & less to learn (because everyone is familiar with OOP fundamentals). 

It's only a concern if we really try to "simulate" a lot of complex OOP patterns. Now it's not the case. We just need a simple inheritance-like mechanism.  

> DOM-related things, by nature, are more easy to describe via OOP. I'm not big fan of OOP, but it would be not rationale to reject it everywhere.

We have no problems with DOM implementation in C either. It knows what we need there were no issues for missing features in it. 

> More universal experience. When user learns ooc lib - this can be reused in many other projects. When user learns lvgl-specific things - this can be reused only in lvgl projects.

OOC libs are also specific IMO. It's not a statistic but never heard anybody who ever used one. So it shouldn't be that common either. An there a lot of commands, patterns, limitations, and issues to learn there too.. 

> note, in events at some moment you added lv_obj pointer to emulate this. Can be counted as one of hand-made compensation for "missed" classes.

It's the same in OOP libs too because C has no language support for this.

> in total - at least 3 kludges + missed reflections. As consequence of missed "full-featured & standard" OOP layer.
In short - switch to ooc lib will make everything more simple & user-friendly :). At almost zero cost (+1 memory lookup for virtual method call).
 
The main problem is it's neither "user-friendly" nor "standard". It doesn't use Cpp notions but custom commands and macro-magic to simulate object-oriented behavior.

embeddedt, 2020-11-16:
In my opinion it's better to make an extensible (object-oriented) C++ wrapper for the C API, instead of trying to turn C into C++.

puzrin, 2020-11-16:
> Now, when you create a `btn` and want to add some extra data you can use `lv_obj_allocate_ext_attr(btn, sizeof(super_btn_ext_t));` . It's a custom method (I mean not coming from an existing paradigm) but simple and dynamic. (means extra members can be added any time after creation).

I don't like this pattern. IMO it came from lack of destructor and attempt to free memory with "what we have".

> If we try to move closer to CPP and use an existing object-oriented C library or create our own I feel that things get even more complicated.

What problem do you feel? With oop-convention user can call `malloc()` directly, because it has destructor to place custom `free()`.

> It's only a concern if we really try to "simulate" a lot of complex OOP patterns. Now it's not the case. We just need a simple inheritance-like mechanism.

I would be happy to use any "simple inheritance-like mechanism", but i don't know such. Right now we have only data inheritance with almost everything else manual:

- no simple method inheritance
- no destructors
- no type reflection
- more costy data model (some methods references in each object)

I suggested not abstract oop, but libs which solve all that in C-specific way.

> We have no problems with DOM implementation in C either. It knows what we need there were no issues for missing features in it.

1. Delegated listeners (via bubbled evens). Not exist, and would be inconvenient without type reflection.
2. You operate in scope "data is ok". But problem is "even if data is ok, methods are not".

> OOC libs are also specific IMO. It's not a statistic but never heard anybody who ever used one.

Probably, that may be caused by specifics of your work. In embeddeds people tend to do "all project by single person". That cause reduced communications (new information propagates slower, when people feel self-sufficient).

For PC take a look at `GObject` - it's very well known. I did not mentioned it before, because it's too fat for small embededs (but more featured).

> It's the same in OOP libs too because C has no language support for this.

This was used in different context: "instead of apply everything at once [as oop-like known things], lvgl spreads pleasure for several years"

> The main problem is it's neither "user-friendly" nor "standard". It doesn't use Cpp notions but custom commands and macro-magic to simulate object-oriented behaviour.

With current compiler and resources restrictions, i don't know better alternatives :(. I'll be ok to use anything else, if it implements methods inheritance / destructors / reflections in simple way.

> In my opinion it's better to make an extensible (object-oriented) C++ wrapper for the C API, instead of trying to turn C into C++.

IMO you misunderstand the root of problem. It's not because i don't like C. It's because some kinds of tasks are more easy to describe in oop-style. Many projects on PC use GObject and stay with C. That's not attempt to turn C into CPP. That's attempt to find suitable instrument for  task.

kisvegabor, 2020-11-17:
I don't know if a proverb like this exists in English, but in Hungarian, we say "You put the bug in my ears", which means what you said made me thinking and can stop thinking about it. :laughing: 

So, using an external lib is definitely not an option. If we really wanna generalize LVGL in this regard, we need full control and customization on such a core thing. 

You are probably more familiar with it but I've just checked that JS uses a different concept for inheritance, called composition. I'm still not sure, but it also could be an interesting direction. 
A good summary: https://www.youtube.com/watch?v=wfMtDGfHWpA

puzrin, 2020-11-17:
> So, using an external lib is definitely not an option. If we really wanna generalize LVGL in this regard, we need full control and customization on such a core thing.

I did not insisted to use external dependencies. Refs to libs was a simple way to explain bulk of complicated things.

If summarize my point very short:

1. LVGL has direct and indirect "problems", caused by missed inheritance layer.
2. IMO this MUST be solved somehow.
3. There are ready solutions for C, useable "as is" or patched or as example (no need to reinvent everything from scratch).

> You are probably more familiar with it but I've just checked that JS uses a different concept for inheritance, called composition. I'm still not sure, but it also could be an interesting direction.

JS uses prototype-based inheritance. Don't waste time for it. JS provides unlimited possibilities to shout into leg for novices :)

IMO single-parent inheritance should be ok in LVGL for many years, if there is simple API how to inherit "widget". CPP-like classes will be enough.

puzrin, 2020-11-17:
BTW, moving to "more classic inheritance" will probably make not needed more fields of lv_obj_t:

- `ext_attr` (just add  attrs in extended class, only when needed)
-  `user_data` - the same
- `ext_draw_pad` - can be done with `.get_draw_pad()` class method (+ override on demand for concrete class)
- may be something else

I don't point such things as main argument, because don't like micro-optimizations in architecture decisions. But tendency is "with good architecture everything become better, magically".

embeddedt, 2020-11-17:
It seems to me that there are several different paths forward being proposed here. We should probably choose one of them before diving too deeply into specifics. :slightly_smiling_face:

I would like to seriously consider the option of simply using C++ with restrictions, as I mentioned below.

------------------------------------------------------

**Keep the current system.**

Pros:
* Little to no API change.

Cons:
* Complex to extend/subclass existing widgets without awkward workarounds (like saving an ancestor signal function and calling it later).
* Extra properties on objects that aren't always used.

I think we all agree that this isn't a good solution.

**Use C++, but not the STL (i.e. "C with classes").**

Pros:
* No need to reinvent the wheel (or adapt an existing library) for inheritance.
* Makes it easier to customize existing widgets without modifying their code.
* Removes the need for users to make a C++ wrapper themselves.
* We can get rid of monolithic event/signal handlers (and perhaps merge events and signals) in favor of individual methods for each event (e.g. `onClick`, `onDelete`, etc.).

Cons:
* Templates can get very confusing, very quickly (at least to me).
* We are inevitably going to get pull requests for features that depend on the STL.
* Anyone using LVGL in an environment that only supports C is stuck using v8.
* C++ has gained a reputation for causing bloated code because of templates and the STL.

Personally I feel like this may be the best solution, provided that we measure code size impact.

**Use a custom, C-based inheritance wrapper.**

Pros:
* A quick solution to the problem without needing to refactor the entire library.
* Doesn't prevent C users from using the library.

Cons:
* I am unsure of the runtime cost.
* We are essentially remaking C++.

embeddedt, 2020-11-18:
Because of the similarity between C and C++ one could even start the conversion in parallel with mainline development, starting by converting the widgets to C++, then switching over the core APIs later. That way we could determine if this is feasible or not. We'd also need to see whether the parser @amirgon uses for MicroPython can handle C++.

Another thing we need to consider is exceptions. Personally I find them indispensable for debugging purposes, but I know they have some amount of runtime overhead associated. We would need to decide whether or not we are using them.

amirgon, 2020-11-18:
> whether the parser @amirgon uses for MicroPython can handle C++

[No it doesn't](https://github.com/eliben/pycparser/tree/master#user-content-13which-version-of-c-does-pycparser-support:~:text=pycparser%20aims%20to%20support%20the%20full,patches%20to%20support%20more%20are%20welcome.).

C++ API would mean a rewrite of the Micropython bindings. I don't think I would want to do that.  

If you are going to change LVGL to C++ instead of C (even without STL), you might lose Micropython support unless someone takes this very heavy task of rewriting the Micropython bindings.

liebman, 2020-11-18:
What about a C compatibility layer that would implement something close to the current API and map this to the C++ implementation? (no I've not thought this thru completely, just a thought as it would make migrating apps to the new version easier as well)

puzrin, 2020-11-18:
I'd be happy to use CPP, but no sure other users will be as happy as me :). So tried to suggest something useable for current limits. Of cause, if team decide switch to CPP, i will prepare completely different proposal about API.

@embeddedt 

Some unsorted notes to your posts:

- Those crazy ooc libs also support exceptions, if needed.
- IMO, for lvgl needs cpp templates are not needed too much.
- If you worry about STL, see ETL https://www.etlcpp.com/ (but i have no idea why STL may be needed).
- Runtime cost of ooc lib is ~ zero
- Modern CPP compilers produce code as small as C, if used properly (see ETL as example)

I do not suggest anything, only clarify: some of "cons" you mentioned are not fatal.

---

One more strange idea: instead of kill C version in favour of CPP implicit - create lvgl_cpp and see who survives.

- C repo will continue to exist for situation when breaking changes needed (major release). But without active updates
- CPP repo - for main activity
- since only "widgets" need "classes", most of code an stay in C for simple sync

---

- Probably, need poll with clear question: "Will you stop use LVGL if we switch to CPP?"
- May be worth ask Zephyr devs what they thing about that.

puzrin, 2020-11-18:
@amirgon it's possible to keep c-like wapper, IMO. CPP features can be demanded in 2 places:

- Widgets, because those need better inheritance support
- Optionally, public API for more smart syntax (can be wrapped to legacy style with some efforts).

amirgon, 2020-11-18:
> @amirgon it's possible to keep c-like wrapper

It would need to be C, not just C-like.
For Micropython we use the entire LVGL public API, including widgets. Anything that is included from lvgl.h.
I don't mind if there is "another" API which is C++ (or Rust or whatever), but my opinion is that we should keep the C API.
Anything that is not exposed on the C API would not be accessible from Micropython.

embeddedt, 2020-11-18:
@amirgon C++ is pointless unless the widgets are actually written in C++, so we would need to "switch things around", so to speak, and have a C++ core with a C wrapper around it. Not sure how that would work.

kisvegabor, 2020-11-21:
Even if most of the people already use or could use Cpp we can not "drop" the minority who needs to use C. The example of the Micropython binding shows that non-trivial use cases of LVGL are better with C. In short, C is compatible with "everything", Cpp isn't.

I agree that the current method could be improved, but I think we should avoid using Cpp as the default option.  
So I vote for custom Cpp-like inheritance in C. IMO, using existing libs for this purpose is risky because:
1. it's too important to rely on external libs
2. it shouldn't too difficult to implement our version by getting inspiration from existing libs.  

BTW, not sure we really need Cpp-like inheritance. I haven't examined all the options yet but maybe we could be good with something like [mixin](https://en.wikipedia.org/wiki/Mixin)s too.

embeddedt, 2020-11-21:
I suppose if we have object inheritance in C it's not impossible to make C++ class wrappers for the C objects.

kisvegabor, 2020-11-21:
> I suppose if we have object inheritance in C it's not impossible to make C++ class wrappers for the C objects.

I think we can construct the new "system" to support it. We don't need to recreate Cpp. We have goals and sure mimicking Cpp is the best way to reach them. Although we haven't written an explicit list of goals yet :sweat_smile: Probably it should be the next step. 

As we are a little bit far from the original topic I suggest opening a new issue for better inheritance support.

puzrin, 2020-11-21:
Done: #1919

@kisvegabor need to understand your general opinion & plans. A least, seems we have agreement about "somethings should be done with inheritance".

closed 2020-11-21 by puzrin

puzrin, 2020-11-22:
> BTW, not sure we really need Cpp-like inheritance. I haven't examined all the options yet but maybe we could be good with something like [mixin](https://en.wikipedia.org/wiki/Mixin)s too.

Mixins are good when you already have classes and memory. Without those, mixins are just a dream, useable only for trending yotube videos, where authors are not responsible for their recommendations :).