lvgl/lvgl #1876: hi，How to rotate a image，I want to implement a rotating dashboard pointer

zq190kg, 2020-10-30:
[image]
I want to achieve the above effect。
use CANVAS &lv_canvas_transform? thks.

embeddedt, 2020-10-30:
You can probably use a gauge with a [custom needle image](https://docs.lvgl.io/latest/en/html/widgets/gauge.html#needle-image) as well, but you would need to have a background image behind it as I don't think we can recreate the metallic effect.

----------------------
We use GitHub issues for development related discussions.
Please use the [forum](https://forum.littlevgl.com/) to ask questions.

It's possible to sign in to the forum with your GitHub account... you don't need to make a separate account.

closed 2020-10-30 by embeddedt

zq190kg, 2020-10-31:
Sorry, I used to use GitHub less. Thank you for your advice，I'll use the forum next time，https://forum.lvgl.io/