lvgl/lvgl #1878: [Performance] _lv_disp_refr_task takes a long time

mark12q, 2020-11-02:
I created the simplest task, but it took 201ms.
 
`int main(void)
{
  STM32_init();

  usrAppConfig();

  lv_obj_t* scr = lv_scr_act();
  for(;;)
  {
    lv_task_handler();
  }
}`

Here is the refresh task code，running time took 201ms

`void _lv_disp_refr_task(lv_task_t * task)

{

    _**uint32_t t1 = lv_tick_get();**_

    LV_LOG_TRACE("lv_refr_task: started");

    uint32_t start = lv_tick_get();

    uint32_t elaps = 0;

    disp_refr = task->user_data;

#if LV_USE_PERF_MONITOR == 0

    /* Ensure the task does not run again automatically.

     * This is done before refreshing in case refreshing invalidates something else.

     */

    lv_task_set_prio(task, LV_TASK_PRIO_OFF);

#endif

    /*Do nothing if there is no active screen*/

    if(disp_refr->act_scr == NULL) {

        disp_refr->inv_p = 0;

        return;

    }

    lv_refr_join_area();

    lv_refr_areas();

    /*If refresh happened ...*/

    if(disp_refr->inv_p != 0) {

        /* In true double buffered mode copy the refreshed areas to the new VDB to keep it up to date.

         * With set_px_cb we don't know anything about the buffer (even it's size) so skip copying.*/

        if(lv_disp_is_true_double_buf(disp_refr)) {

            if(disp_refr->driver.set_px_cb) {

                LV_LOG_WARN("Can't handle 2 screen sized buffers with set_px_cb. Display is not refreshed.");

            }

            else {

                lv_disp_buf_t * vdb = lv_disp_get_buf(disp_refr);

                /*Flush the content of the VDB*/

                lv_refr_vdb_flush();

                /* With true double buffering the flushing should be only the address change of the

                 * current frame buffer. Wait until the address change is ready and copy the changed

                 * content to the other frame buffer (new active VDB) to keep the buffers synchronized*/

                while(vdb->flushing);

                lv_color_t * copy_buf = NULL;

#if LV_USE_GPU_STM32_DMA2D

                LV_UNUSED(copy_buf);

#else

                copy_buf = _lv_mem_buf_get(disp_refr->driver.hor_res * sizeof(lv_color_t));

#endif

                uint8_t * buf_act = (uint8_t *)vdb->buf_act;

                uint8_t * buf_ina = (uint8_t *)vdb->buf_act == vdb->buf1 ? vdb->buf2 : vdb->buf1;

                lv_coord_t hres = lv_disp_get_hor_res(disp_refr);

                uint16_t a;

                for(a = 0; a < disp_refr->inv_p; a++) {

                    if(disp_refr->inv_area_joined[a] == 0) {

                        uint32_t start_offs =

                            (hres * disp_refr->inv_areas[a].y1 + disp_refr->inv_areas[a].x1) * sizeof(lv_color_t);

#if LV_USE_GPU_STM32_DMA2D

                        lv_gpu_stm32_dma2d_copy((lv_color_t *)(buf_act + start_offs), disp_refr->driver.hor_res,

                                                (lv_color_t *)(buf_ina + start_offs), disp_refr->driver.hor_res,

                                                lv_area_get_width(&disp_refr->inv_areas[a]),

                                                lv_area_get_height(&disp_refr->inv_areas[a]));

#else

                        lv_coord_t y;

                        uint32_t line_length = lv_area_get_width(&disp_refr->inv_areas[a]) * sizeof(lv_color_t);

                        for(y = disp_refr->inv_areas[a].y1; y <= disp_refr->inv_areas[a].y2; y++) {

                            /* The frame buffer is probably in an external RAM where sequential access is much faster.

                             * So first copy a line into a buffer and write it back the ext. RAM */

                            _lv_memcpy(copy_buf, buf_ina + start_offs, line_length);

                            _lv_memcpy(buf_act + start_offs, copy_buf, line_length);

                            start_offs += hres * sizeof(lv_color_t);

                        }

#endif

                    }

                }

                if(copy_buf) _lv_mem_buf_release(copy_buf);

            }

        } /*End of true double buffer handling*/

        /*Clean up*/

        _lv_memset_00(disp_refr->inv_areas, sizeof(disp_refr->inv_areas));

        _lv_memset_00(disp_refr->inv_area_joined, sizeof(disp_refr->inv_area_joined));

        disp_refr->inv_p = 0;

        elaps = lv_tick_elaps(start);

        /*Call monitor cb if present*/

        if(disp_refr->driver.monitor_cb) {

            disp_refr->driver.monitor_cb(&disp_refr->driver, elaps, px_num);

        }

    }

    _lv_mem_buf_free_all();

    _lv_font_clean_up_fmt_txt();

#if LV_USE_PERF_MONITOR && LV_USE_LABEL

    static lv_obj_t * perf_label = NULL;

    if(perf_label == NULL) {

        perf_label = lv_label_create(lv_layer_sys(), NULL);

        lv_label_set_align(perf_label, LV_LABEL_ALIGN_RIGHT);

        lv_obj_set_style_local_bg_opa(perf_label, LV_LABEL_PART_MAIN, LV_STATE_DEFAULT, LV_OPA_COVER);

        lv_obj_set_style_local_bg_color(perf_label, LV_LABEL_PART_MAIN, LV_STATE_DEFAULT, LV_COLOR_BLACK);

        lv_obj_set_style_local_text_color(perf_label, LV_LABEL_PART_MAIN, LV_STATE_DEFAULT, LV_COLOR_WHITE);

        lv_obj_set_style_local_pad_top(perf_label, LV_LABEL_PART_MAIN, LV_STATE_DEFAULT, 3);

        lv_obj_set_style_local_pad_bottom(perf_label, LV_LABEL_PART_MAIN, LV_STATE_DEFAULT, 3);

        lv_obj_set_style_local_pad_left(perf_label, LV_LABEL_PART_MAIN, LV_STATE_DEFAULT, 3);

        lv_obj_set_style_local_pad_right(perf_label, LV_LABEL_PART_MAIN, LV_STATE_DEFAULT, 3);

        lv_label_set_text(perf_label, "?");

        lv_obj_align(perf_label, NULL, LV_ALIGN_IN_BOTTOM_RIGHT, 0, 0);

    }

    static uint32_t perf_last_time = 0;

    static uint32_t elaps_max = 1;

    if(lv_tick_elaps(perf_last_time) < 300) {

        elaps_max = LV_MATH_MAX(elaps, elaps_max);

    }

    else {

        perf_last_time = lv_tick_get();

        uint32_t fps = 1000 / (elaps_max == 0 ? 1 : elaps_max);

        elaps_max = 1;

        uint32_t fps_limit = 1000 / disp_refr->refr_task->period;

        if(fps > fps_limit) fps = fps_limit;

        uint32_t cpu = 100 - lv_task_get_idle();

        lv_label_set_text_fmt(perf_label, "%d FPS\n%d%% CPU", fps, cpu);

        lv_obj_align(perf_label, NULL, LV_ALIGN_IN_BOTTOM_RIGHT, 0, 0);

    }

#endif

    LV_LOG_TRACE("lv_refr_task: ready");

    _**uint32_t t2 = lv_tick_get();

    printf("refresh v8 time:%d\n", t2 - t1);**_

}`

amative-weitai, 2020-11-02:
_lv_disp_refr_task tooks too long time. when _lv_indev_read_task or anim_task is about to be executed, _lv_disp_refr_task is still in execution. _lv_indev_read_task or anim_task cannot be executed in time. the page doesn't slide smoothly. animation is not smoothly.

kisvegabor, 2020-11-02:
> t2 - t1 equals 201ms, the _lv_disp_refr_task took 201ms.

Is this ~200 ms reasonable?
- How fast is your MCU
- What is the resolution of your display?
- How do you communicate with the display?
- How complex the created UI is?

embeddedt, 2020-11-02:
How much time is spent in `flush_cb`, and what is the size of your display buffer compared to your display?

amative-weitai, 2020-11-02:
> > t2 - t1 equals 201ms, the _lv_disp_refr_task took 201ms.
> 
> Is this ~200 ms reasonable?
> 
> * How fast is your MCU
> * What is the resolution of your display?
> * How do you communicate with the display?
> * How complex the created UI is?

MCU rate is 120Mhz.
The display resolution is 390*390.
Double buffer is filled directly. Not GPU, not dma.

> > t2 - t1 equals 201ms, the _lv_disp_refr_task took 201ms.
> 
> Is this ~200 ms reasonable?
> 
> * How fast is your MCU
> * What is the resolution of your display?
> * How do you communicate with the display?
> * How complex the created UI is?

MCU rate is 120Mhz.
The display resolution is 390*390.
Double buffer is filled directly. Not GPU, not dma.
The created UI is simply. It only gets active screen.

amative-weitai, 2020-11-02:
> How much time is spent in `flush_cb`, and what is the size of your display buffer compared to your display?
flush_cb takes ~8ms. The display resolution is 390*390. Double buffer is filled directly.

amative-weitai, 2020-11-02:
> > t2 - t1 equals 201ms, the _lv_disp_refr_task took 201ms.
> 
> Is this ~200 ms reasonable?
> 
> * How fast is your MCU
> * What is the resolution of your display?
> * How do you communicate with the display?
> * How complex the created UI is?

MCU rate is 120Mhz.
The display resolution is 390*390.
Double buffer is filled directly. Not GPU, not dma.
The created UI is simply. It only gets active screen.

embeddedt, 2020-11-02:
Where are you calling `lv_tick_inc`? With nothing on the screen it should barely take any time to run `_lv_disp_refr_task`.

amative-weitai, 2020-11-02:
> Where are you calling `lv_tick_inc`? With nothing on the screen it should barely take any time to run `_lv_disp_refr_task`.

I don't calling `lv_tick_inc`. Should `lv_tick_inc` be called?

embeddedt, 2020-11-02:
Unless you have `LV_TICK_CUSTOM` enabled in `lv_conf.h`, you need to be calling `lv_tick_inc` every 5-10 ms.

amative-weitai, 2020-11-02:
> Unless you have `LV_TICK_CUSTOM` enabled in `lv_conf.h`, you need to be calling `lv_tick_inc` every 5-10 ms. It should not be called in the same loop as `lv_task_handler`; you should use another thread or interrupt handler.

I use timer generate interrupt 1ms. There should be no problem.

amative-weitai, 2020-11-03:
> Unless you have `LV_TICK_CUSTOM` enabled in `lv_conf.h`, you need to be calling `lv_tick_inc` every 5-10 ms. It should not be called in the same loop as `lv_task_handler`; you should use another thread or interrupt handler.

`lv_tick_inc` will be called once in 1ms by timer's callback. I don't think tick is a problem.

```
void HAL_TIM_PeriodElapsedCallback(TIM_HandleTypeDef *htim)
{
  /* USER CODE BEGIN Callback 0 */

  /* USER CODE END Callback 0 */
  if (htim->Instance == TIM1) {
    lv_tick_inc(1);
    HAL_IncTick();
    //RunTimeCounterINC();
  }
  /* USER CODE BEGIN Callback 1 */

  /* USER CODE END Callback 1 */
}

/**
 * You have to call this function periodically
 * @param tick_period the call period of this function in milliseconds
 */
LV_ATTRIBUTE_TICK_INC void lv_tick_inc(uint32_t tick_period)
{
    tick_irq_flag = 0;
    sys_time += tick_period;
}

/**
 * Get the elapsed milliseconds since start up
 * @return the elapsed milliseconds
 */
uint32_t lv_tick_get(void)
{
#if LV_TICK_CUSTOM == 0

    /* If `lv_tick_inc` is called from an interrupt while `sys_time` is read
     * the result might be corrupted.
     * This loop detects if `lv_tick_inc` was called while reading `sys_time`.
     * If `tick_irq_flag` was cleared in `lv_tick_inc` try to read again
     * until `tick_irq_flag` remains `1`. */
    uint32_t result;
    do {
        tick_irq_flag = 1;
        result        = sys_time;
    } while(!tick_irq_flag); /*Continue until see a non interrupted cycle */

    return result;
#else
    return LV_TICK_CUSTOM_SYS_TIME_EXPR;
#endif
}
```

BesitzeRuf, 2020-11-03:
@amative-weitai  Hi, LVGL has a [benchmark](https://github.com/lvgl/lv_examples) demo, please provide results from it or provide code of your GUI and functions related to LCD driver. After that it will be possible to analyze your issue. There are a lot of things, that may affect "performance"

amative-weitai, 2020-11-04:
> @amative-weitai Hi, LVGL has a [benchmark](https://github.com/lvgl/lv_examples) demo, please provide results from it or provide code of your GUI and functions related to LCD driver. After that it will be possible to analyze your issue. There are a lot of things, that may affect "performance"

> @amative-weitai Hi, LVGL has a [benchmark](https://github.com/lvgl/lv_examples) demo, please provide results from it or provide code of your GUI and functions related to LCD driver. After that it will be possible to analyze your issue. There are a lot of things, that may affect "performance"

The 'ZZ.7z' zip is demo code. The development environment is stm32cubeide. If you just view the code, you only need to use a text edit

> @amative-weitai Hi, LVGL has a [benchmark](https://github.com/lvgl/lv_examples) demo, please provide results from it or provide code of your GUI and functions related to LCD driver. After that it will be possible to analyze your issue. There are a lot of things, that may affect "performance"

[image]

> @amative-weitai Hi, LVGL has a [benchmark](https://github.com/lvgl/lv_examples) demo, please provide results from it or provide code of your GUI and functions related to LCD driver. After that it will be possible to analyze your issue. There are a lot of things, that may affect "performance"

amative-weitai, 2020-11-04:
> @amative-weitai Hi, LVGL has a [benchmark](https://github.com/lvgl/lv_examples) demo, please provide results from it or provide code of your GUI and functions related to LCD driver. After that it will be possible to analyze your issue. There are a lot of things, that may affect "performance"

The 'ZZ.7z' zip is demo code. The development environment is stm32cubeide. If you just view the code, you only need to use a text editor.

The video has a phenomenon. 
The first page is a clock. The second page is a list. 
The first page refresh the hour minute second hand every second. At this time, the refresh task execution time is ~300ms. 
The first page switches to the second page from left to right by animation,but there is no animation effect. Because the refresh task execution takes too long ting, the animation task is difficult to be executed. When the animation task is executed, the tick difference is very big, so the page is only moved twice, without animation effect.
On the second page, the list doesn't slide very smoothly. When the list is sliding, the refresh task will be executed, and the task a cannot be executed in time, so the sliding is not very smooth
The above is my analysis, my analysis may be wrong, please help me check if my code has any problems.

amative-weitai, 2020-11-04:
> @amative-weitai Hi, LVGL has a [benchmark](https://github.com/lvgl/lv_examples) demo, please provide results from it or provide code of your GUI and functions related to LCD driver. After that it will be possible to analyze your issue. There are a lot of things, that may affect "performance"

[ZZ.zip_83.zip](https://github.com/lvgl/lvgl/files/5486667/ZZ.zip_83.zip)
[image]
[image]
The 'ZZ.zip_83.zip' zip is demo code. The development environment is stm32cubeide. If you just view the code, you only need to use a text editor.

The first page is a clock. The second page is a list. 
The first page refresh the hour minute second hand every second. At this time, the refresh task execution time is ~300ms. 
The first page switches to the second page from left to right by animation,but there is no animation effect. Because the refresh task execution takes too long ting, the animation task is difficult to be executed. When the animation task is executed, the tick difference is very big, so the page is only moved twice, without animation effect.
On the second page, the list doesn't slide very smoothly. When the list is sliding, the refresh task will be executed, and the task a cannot be executed in time, so the sliding is not very smooth
The above is my analysis, my analysis may be wrong, please help me check if my code has any problems.
Thank you very much.

BesitzeRuf, 2020-11-04:
Thank for sharing, I will recommend you to:
1) try execute your code with enabled optimization level -O3
2) decrease color depth from 32 to 16 bits (it is configured via symbol **LV_COLOR_DEPTH** in lv_conf.h header file
3)  Try to use internal SRAM for draw buffers (from which you will need to flush data to the video/visible buffer using DMA) instead of external PSRAM

amative-weitai, 2020-11-04:
> Thank for sharing, I will recommend you to:
>
> 1. try execute your code with enabled optimization level -O3
> 2. decrease color depth from 32 to 16 bits (it is configured via symbol **LV_COLOR_DEPTH** [...]

Thank you for your suggestion. I'll try it.

closed 2020-11-26 by kisvegabor