lvgl/lvgl #1879: checkbox has position issue

ZhaoQiang-b45475, 2020-11-03:
Hi, 
I create a checkbox and set its position with (0, 0), using font with lv_font_montserrat_48, then the widget is beyond the screen. please refer my code and capture as below:

        ui->screen = lv_disp_get_scr_act(NULL);
	ui->screen_cb0 = lv_checkbox_create(ui->screen, NULL);
	lv_checkbox_set_text(ui->screen_cb0, "checkbox");
	lv_obj_set_pos(ui->screen_cb0, 0, 0);
	lv_obj_set_style_local_text_font(ui->screen_cb0, LV_CHECKBOX_PART_BG, LV_STATE_DEFAULT, &lv_font_montserrat_48);

[image]

upbeat27, 2020-11-03:
Did you try setting the font before setting the position?

ZhaoQiang-b45475, 2020-11-04:
Hi, 
I try setting position after font, it works, thanks!

kisvegabor, 2020-11-04:
Just a quick explanation of why it happens: when you change the font, the height of the checkbox is expanded symmetrically to both directions.

As it's normal and the proposed workaround should work, so I close this issue.

closed 2020-11-04 by kisvegabor