lvgl/lvgl #1882: Dark bar for indicator when tabview location is set to LV_TABVIEW_TAB_POS_BOTTOM

acumartini, 2020-11-05:
As per @embeddedt 's suggestion I am moving this forum post to an Issue: https://forum.lvgl.io/t/dark-bar-for-indicator-when-tabview-location-is-set-to-lv-tabview-tab-pos-bottom/3613/4

**Describe the bug**
I really like the dark bar at the bottom of the tabview that holds the sliding tab indicator when the location is set to LV_TABVIEW_TAB_POS_TOP. It would be nice to have duplicate behavior for LV_TABVIEW_TAB_POS_BOTTOM. In fact I think I see the dark bar at the bottom fo the tabview when in the bottom position and it appears to be pushing the tabview up by roughly 4 pixels, so this might be considered a BUG report.

**To Reproduce**
```
lv_obj_t *tv = lv_tabview_create(pg, NULL);
lv_tabview_set_btns_pos(tv, LV_TABVIEW_TAB_POS_BOTTOM);

lv_obt_t *tb = lv_tabview_add_tab(tv, "TEST TAB");
```

**Expected behavior**
Rendering and stylization of tabview in all position should be the same with respect to the dark indicator background coloring and location. In addition, respective tabview dimensions should be the same for TOP and BOTTOM position in terms of total “height” of the rendered widget.

**Screenshots**
[image]

kisvegabor, 2020-11-08:
Hi,

Actually, it's a bottom border on tab buttons background. It really looks odd when the buttons are on the bottom. :slightly_frowning_face: 

You can get rid of it with 
`lv_obj_set_style_local_border_width(tabview, LV_TABVIEW_PART_TAB_BG, LV_STATE_DEFAULT, 0);`

acumartini, 2020-11-09:
I mean we prob can move the overlaying tab_view obj down when in `LV_TABVIEW_TAB_POS_BOTTOM` mode to make it symmetrical with the top view?

embeddedt, 2020-11-09:
We'd need to dynamically adjust the object's style to permanently fix it; the border side would need to be adjusted whenever `lv_tabview_set_btns_pos` is called.

I think this would fix the problem for you:

```c
lv_obj_set_style_local_border_side(tabview, LV_TABVIEW_PART_TAB_BG, LV_STATE_DEFAULT, LV_BORDER_SIDE_TOP);
```

acumartini, 2020-11-11:
@embeddedt Confirmed definitely fixed the styling. Thanks!

kisvegabor, 2020-11-11:
Finally I foxed it by removing the border from the theme.

commit 43771fa2 (2020-11-11): fix(theme_material): remove the bottom border from tabview header

If the tabs were placed to the bottom the bottom border looked like gap on the bottom

fix #1882
closed 2020-11-11 by kisvegabor