lvgl/lvgl #1884: issues for arc knob

ZhaoQiang-b45475, 2020-11-06:
Hi,
below are my codes, I met two issues:
1. I have trouble to drag the knob to an angle between 0 and 90 degree.
2. when I drag the knob, the start angle are modified to 0 suddenly

       //Write codes screen
	ui->screen = lv_disp_get_scr_act(NULL);

	//Write codes screen_arc0
	ui->screen_arc0 = lv_arc_create(ui->screen, NULL);

	lv_obj_set_pos(ui->screen_arc0, 195, 80);
	lv_obj_set_size(ui->screen_arc0, 100, 100);
	// lv_arc_set_angles(ui->screen_arc0, 0, 160);
	lv_arc_set_bg_angles(ui->screen_arc0, 0, 360);
	// lv_arc_set_rotation(ui->screen_arc0, 0);
	lv_arc_set_adjustable(ui->screen_arc0, true);

kisvegabor, 2020-11-09:
Answer on the Forum:
https://forum.lvgl.io/t/use-the-knob-to-adjust-the-arc-to-realize-a-time-modifier/3600

ZhaoQiang-b45475, 2020-11-10:
As capture as below, I have set the bg_angle_start = 0 and bg_angle_end=360, but I couldn't drag the knob to the angle between 0 and 90.
[image]

kisvegabor, 2020-11-10:
I could reproduce the issue and will work on it tomorrow.

ZhaoQiang-b45475, 2020-11-11:
The code is as below, meanwhile, I attached a video zip to show my issue.
[VID_20201111_113738.zip](https://github.com/lvgl/lvgl/files/5521594/VID_20201111_113738.zip)
 
 /*Create an Arc*/
  lv_obj_t * arc = lv_arc_create(lv_scr_act(), NULL);
  lv_arc_set_bg_angles(arc, 0, 360);
  lv_arc_set_angles(arc, 270, 270);
  lv_obj_align(arc, NULL, LV_ALIGN_CENTER, 0, 0);
  lv_arc_set_adjustable(arc, true);

kisvegabor, 2020-11-11:
Are you sure you have updated LVGL?

> lv_arc_set_angles(arc, 270, 270);

This line is not required because the angles are controlled dragging:

See:
[image]

ZhaoQiang-b45475, 2020-11-12:
I test with b19fb8620f9d59d86580e380796f78d52b5a373f, it could be reproduced on my side.

kisvegabor, 2020-11-12:
b19fb86 is 6 days old but the fix was added 2 days ago:

[image]

ZhaoQiang-b45475, 2020-11-15:
Thanks, it works.

kisvegabor, 2020-11-16:
Great! :slightly_smiling_face:

closed 2020-11-16 by kisvegabor