lvgl/lvgl #1886: lv_label_set_text_fmt unable to format number correctly

zephray, 2020-11-06:
> # Important: issues that don't use this template will be ignored/closed.

**Describe the bug**

I am trying to run the lv_ex_get_started_1.c example, however the text of button is always Button: 0.

It turns out that the code lv_label_set_text_fmt(label, "Button: %d", cnt) will always set the label to Button: 0 regardless of the number of cnt. After tracing down into the implementation of _lv_txt_set_text_vfmt, it turns out that the va_list ap is being used twice, first time to determine the length, second time to print the text. However, in the second time, all values read by vsnprintf would be 0.

I have implemented a simple fix myself to resolve the issue:

```diff
@@ -483,7 +483,10 @@ void _lv_txt_cut(char * txt, uint32_t pos, uint32_t len)
 char * _lv_txt_set_text_vfmt(const char * fmt, va_list ap)
 {
     /*Allocate space for the new text by using trick from C99 standard section 7.19.6.12 */
-    uint32_t len = lv_vsnprintf(NULL, 0, fmt, ap);
+    va_list ap_copy;
+    va_copy(ap_copy, ap);
+    uint32_t len = lv_vsnprintf(NULL, 0, fmt, ap_copy);
+    va_end(ap_copy);
     char * text = 0;
 #if LV_USE_ARABIC_PERSIAN_CHARS
     /*Put together the text according to the format string*/
```

I am new to the lvgl, I am quite surprised that this is not working. I am suspecting there must be something wrong with my enviornment but I am not really sure how this could work on other environments.

**To Reproduce**

lv_ex_get_started_1.c is not working for me. My platform is my own simulator with FreeRTOS on Linux (Ubuntu 20.04 LTS + gcc 9.3.0). Changing lv_label_set_text_fmt(label, "Button: %d", cnt) to lv_label_set_text_fmt(label, "Button: %d", 100) also only prints Button: 0

**Expected behavior**

lv_label_set_text_fmt(label, "Button: %d", 100) should set the label to Button: 0

**Screenshots**

If needed I can provide one, but this should be fairly clear

**Additional context**

embeddedt, 2020-11-06:
Good catch. Can you send a pull request?

There are many different implementations of `va_list` so I wouldn't doubt that some don't suffer from this issue.

kisvegabor, 2020-11-09:
Hi,

I've just created and merged the PR.

closed 2020-11-09 by kisvegabor