lvgl/lvgl #1887: lv_obj_report_style_mod is much slower in v7

embeddedt, 2020-11-08:
@kisvegabor `lv_obj_report_style_mod` takes significantly longer to run on v7 compared to v6.

This can be reproduced with LVGL `dev`, STM32F7 with full caching and DMA2D using the following code:

```c
    lv_style_t style;
    lv_style_init(&style);

    for(int i = 0; i < 800; i++) {
        lv_obj_t * obj = lv_obj_create(lv_scr_act(), NULL);
        lv_obj_add_style(obj, LV_OBJ_PART_MAIN, &style);
    }
    dprintf("start time: %d\n", lv_tick_get());
    lv_obj_report_style_mod(&style);
    dprintf("end time: %d\n", lv_tick_get());
```

The `lv_obj_report_style_mod` call takes about 150ms to run.

This was originally reported on [the forum](https://forum.lvgl.io/t/seeing-much-slower-style-refresh-performance-on-v7-vs-v6/3662?u=embeddedt). There are 338 objects being refreshed in their case; my test case uses quite a few more, but even with 300 it takes 50ms for me.

Do you have an idea why invalidation takes so long?

kisvegabor, 2020-11-10:
Hi,

I've debugged it and found what happens:
1. When a style changes the cache of the style list of the widget is invalidated 
2. After that the "extended draw area" is recalculated. E.g. if the new style has shadow LVGL needs to allow drawing out of the widgets too and needs to invalidate a larger area if something changes. 
3. While getting the "extended draw area" some style properties are got and the first get triggers a cache update. The update executes ~30 style querries. For 300 objects it's ~10,000 queries.  If one query takes 500 instruction on 200 MHz (2.5 us), 10,000 takes 25 ms. (close to the measured 50 ms)

This patch disables the use of the cache. Could you try it? 
[a.patch.zip](https://github.com/lvgl/lvgl/files/5516552/a.patch.zip)

BTW, how much time did it take to create 800 objects?

ripe909, 2020-11-10:
I tried the patch and performance was worse... the style report for 338 objects went from 143481 ms to 197722 ms after the patch.

kisvegabor, 2020-11-10:
On the forum, you reported 143481 microseconds but now 143481 milliseconds. I hope it's microseconds. :slightly_smiling_face: 

As a quick workaround, you can solve the issue by simply not calling `lv_obj_report_style_mod` and call `lv_obj_invalidate(lv_scr_act())`. 
If you change only colors a simple redraw is enough. `lv_obj_report_style_mod` handles if size changes that require layout recalculation or something similar. 

To solve the real issue an important question: do you use `layout`, `fit`, or lists?

ripe909, 2020-11-10:
ah yes, lol, it is uS, I changed it.  I will try the redraw approach now.

embeddedt, 2020-11-10:
It seems that we really need more efficient style queries then. Even if we postponed the cache update to when the object actually gets redrawn, it would still take 50ms to redraw the object, which caps the maximum FPS at 20 (and that's on powerful hardware like STM32/i.MX RT).

kisvegabor, 2020-11-10:
> using lv_obj_invalidate(lv_scr_act()) the update only takes 30uS.

Huh. :sweat_smile: Good to hear that.

>  I don't have layout or fit but I do have 4 dropdown lists

Then mainly what is the type of those 338 objects?

> It seems that we really need more efficient style queries then. 

I estimate that the planned new style structure will be 10..15 times faster. It makes ~10..15 ms from 141 ms.

 > Even if we postponed the cache update to when the object actually gets redrawn, it would still take 50ms to redraw the object, which caps the maximum FPS at 20 

Only if all the 300 objects are drawn. The main needs for updating the cache directly on style change are:
- update layouts: Needed if padding or margin changes. It's important because if the user gets the position of an object it should be already correct after the style change.
- update extra draw area: Actually it can be refreshed when first needed (on drawing or area invalidation). So `obj->ext_draw_pad`  can be marked invalid and calculated only when needed. I expect a huge speed up from this.

@ripe909 
Please measure the speed if you comment out  [this line](https://github.com/lvgl/lvgl/blob/master/src/lv_core/lv_obj.c#L4014).

ripe909, 2020-11-10:
most of the objects are base objects (squares) with a few text labels and lines.  The base objects have styles that add borders, radii, etc.  Similar to a spreadsheet.

With the patch still applied and commenting that line, the update is: 159935 uS

kisvegabor, 2020-11-12:
So you have more labels on each square? If so, it can be really slow to "recalculate" all the labels.

kisvegabor, 2020-12-07:
We will ensure to run this mechanism faster the new engine of styles are added. (Most probably in v8.)

closed 2020-12-07 by kisvegabor

ripe909, 2021-03-17:
I see that v8 is now released.  Is this style issue addressed?

kisvegabor, 2021-03-18:
It's not released yet, just available for testing. :slightly_smiling_face: 

I made some comparisons on PC and v8 was 15 times faster with a style change report on 100 buttons.

However, if you know that only colors were changed and a simple redraw is enough an `lv_obj_invalidate(lv_scr_act())` will be the fastest.