lvgl/lvgl #1888: Textarea bug when I use Arabic

lilengqiang, 2020-11-09:
> # Important: issues that don't use this template will be ignored/closed.

**Describe the bug**

When I add characters to thetextarea, Arabic does not connect, but when I delete one, it is ok. I had enable LV_USE_ARABIC_PERSIAN_CHARS and LV_USE_BIDI.

**To Reproduce**
First
    lv_obj_t *ta = lv_textarea_create(lv_scr_act(), NULL);

    static lv_font_t font_set;
    lv_freetype_font_init(&font_set, "/system/res/font/arial.ttf", 36, 36);

    lv_obj_set_style_local_text_font(ta, LV_LABEL_PART_MAIN, LV_STATE_DEFAULT, &font_set);

    lv_textarea_set_text(ta, "");
    lv_textarea_add_text(ta, "ض");
    lv_textarea_add_text(ta, "ض");
    lv_textarea_add_text(ta, "ض");

Second

    lv_obj_t *ta = lv_textarea_create(lv_scr_act(), NULL);

    static lv_font_t font_set;
    lv_freetype_font_init(&font_set, "/system/res/font/arial.ttf", 36, 36);

    lv_obj_set_style_local_text_font(ta, LV_LABEL_PART_MAIN, LV_STATE_DEFAULT, &font_set);

    lv_textarea_set_text(ta, "");
    lv_textarea_add_text(ta, "ض");
    lv_textarea_add_text(ta, "ض");
    lv_textarea_add_text(ta, "ض");

    lv_textarea_add_text(ta, "ض");
    lv_textarea_del_char(ta);

**Expected behavior**
[image]

**Screenshots**
First Result
[image]

Second Result
[image]

**Additional context**

kisvegabor, 2020-11-10:
Hi,

I've investigated the issue and seems and it looks complicated. :slightly_frowning_face: 
The problem is now we can't get the unconnected text from the connected.

When a character is inserted into the text area the text is not processed to handle connections. So you have the raw text after inserts. It'd a simple bug.

On delete, the text is processed, and the original text replaced by the connected forms. In your case after 4 inserts, the text area still had the raw text, an on delete the algorithm handled the connections. 

But if we added connection handling on insert, the algorithm couldn't process it correctly when a new character is added/deleted because it expects a raw text.

I see 2 solutions:
1. Store the raw text too in labels
2. Modify the connection algorithm to recognize the raw characters from the connected forms too.

I'd prefer the second because it's more universal and solves the real underlying problem. 

The problem is [here](https://github.com/lvgl/lvgl/blob/a07583331e38579255ad4c2ad2abd89c4bfa3f24/src/lv_misc/lv_txt_ap.c#L144) when the processed characters are get from the text instead of the raws. 
So the solution would be to run a loop on the `ch_enc` and replace the connected characters in it with the raw forms.

Due to the fact the I don't know the Arabic language it'd be very difficult for me to fix it. Could you help with the above-mentioned converter algorithm?

To see the issue:
```c
  lv_obj_t * label = lv_label_create(lv_scr_act(), NULL);
  lv_label_set_text(label, "ضضض");   //ch_enc[0] = 0x636;
  lv_label_set_text(label, lv_label_get_text(label));   //ch_enc[0] = 0xfebf but should be 0x636
```

What I imagined in [_lv_txt_ap_proc](https://github.com/lvgl/lvgl/blob/a07583331e38579255ad4c2ad2abd89c4bfa3f24/src/lv_misc/lv_txt_ap.c#L144) is something like:
```c
    while(j < txt_length) {
        ch_enc[j] = _lv_txt_encoded_next(txt, &i);
        ch_enc[j] = to_raw(ch_enc[j]);
        j++;
}
```

lilengqiang, 2020-11-16:
Sorry. I don't know the Arabic language either.

kisvegabor, 2020-11-16:
Let's leave this issue open for a while. Hopefully, we will get some help from native speakers.

ashrafkamel5, 2020-11-19:
@kisvegabor 
hi, 
I have added a fix for the problem. but now I lost, 
where should I insert the _lv_txt_ap_proc or the lv_label_set_text(ext->label, label_txt); at the text area file?

ashrafkamel5, 2020-11-19:
hi,
I have added a fix for the problem. but now I am lost,
where should I insert the `lv_txt_ap_proc `or the `lv_label_set_text(ext->label, label_txt);` at the text area file?
for the test

lilengqiang, 2020-11-19:
@kisvegabor 
I pulled the latest code.
When I enabled LV_USE_ARABIC_PERSIAN_CHARS, there was an Segmentation fault using btnmatrix

kisvegabor, 2020-11-19:
@ashrafkamel5 
> I have added a fix for the problem. but now I am lost,
where should I insert the lv_txt_ap_proc or the lv_label_set_text(ext->label, label_txt); at the text area file?
for the test

Could you share your fix? I can handle its integration because there some places to modify and test.

@lilengqiang 
> When I enabled LV_USE_ARABIC_PERSIAN_CHARS, there was an Segmentation fault using btnmatrix

I've just fixed it.

ashrafkamel5, 2020-11-19:
ok, I am just afraid of it being a complete failure, but here it is.
I have been thinking about your problem statement and approach to solve it
and wasn't a fan of iterating the whole text every time to replace the forms letters with the raw.
I read Mr. Hamid's algorithm and added a small modification to it.
so now it recognizes both of the raw and forms letters with no extra looping or memory.

in function `lv_ap_get_char_index`
I added a check for the forms beside the raw which Mr. Hamid implemented.
my code is between `//start modification` and `//end` comments
``` c
static uint32_t lv_ap_get_char_index(uint16_t c)
{
    for(uint8_t i = 0; ap_chars_map[i].char_end_form; i++) {
        if(c == (ap_chars_map[i].char_offset + LV_AP_ALPHABET_BASE_CODE))
            return i;
        //start modification
        else if (c == ap_chars_map[i].char_end_form                                                 //is it an End form
            || c == (ap_chars_map[i].char_end_form + ap_chars_map[i].char_begining_form_offset)     //is it a Begining form  
            || c == (ap_chars_map[i].char_end_form + ap_chars_map[i].char_middle_form_offset)       //is it a middle form
            || c == (ap_chars_map[i].char_end_form + ap_chars_map[i].char_isolated_form_offset))    //is it an isolated form
        {
            return i;
        }
        //end
    }
    return LV_UNDEF_ARABIC_PERSIAN_CHARS;
}
```

ashrafkamel5, 2020-11-21:
Hi Mr @kisvegabor ,
any updates?

kisvegabor, 2020-11-21:
@ashrafkamel5 
I've just tried your update and it worked perfectly. Thank you very much.

I've also modified the required parts of the library and the issue seems to be solved.

closed 2020-11-21 by kisvegabor

ashrafkamel5, 2020-11-22:
Wow, that is great to hear <3

commit 07c63b8a (2020-11-21): fix(textarea): support Arabic letter connections

fixes #1888
commit bf08ff44 (2020-12-01): Update Kconfig. (#1917)

* Kconfig: Add missing _HEX to LV_COLOR_TRANSP option.

* lv_conf_checker: Define LV_CONF_SKIP if CONFIG_LV_CONF_SKIP is defined.

* fix(indev): disabled object shouldn't absorb clicks but let the parent to be clicked

* Update README.md

* fix(arabic): support processing again alraedy processed texts with _lv_txt_ap_proc

* fix(textarea): support Arabic letter connections

fixes #1888

* Fix #1924: incorrect date on changelog

December 1st is two weeks from 7.7.2 release.

* Kconfig: Fix LV_CONF_MINIMAL related options.

* fix(arabic): support Arabic letter connection [... cut]