lvgl/lvgl #1892: add support for large image

incity, 2020-11-10:
[image]
i saw the lv_img_header_t just has 11 bits of width and height(2047 pixels), but something i need a big image to show in a page.

embeddedt, 2020-11-10:
A workaround would be to split it into multiple images. This might be more efficient anyways if your display is significantly smaller than the total image size.

kisvegabor, 2020-11-10:
To better understand the requirements please tell us what your use case is.
Why do you need so large images?

incity, 2020-11-11:
We need to generate a bitmap based on user input, and show on the screen, user may input many words and will make the width of bitmap long.

incity, 2020-11-11:
I'm curious why you limit the length of the image to just 11 bits, that's just 2047 pixels, but a image with 2048 pixels is not so large, it's still a common case i think. in android, we often do a long screen shot.

incity, 2020-11-11:
Even users may use lvgl to display a large poster, which can be frustrating if it doesn't.

kisvegabor, 2020-11-11:
The limitation exists to have a small 4 byte header.

Actually the 2 reserved bit could be added to width and height resulting 4096px max size. 

If we want more we needed a larger header too.

Note that, if you convert the images to C array, you can modify the `lv_img_header_t`. E.g. 
```c
typedef struct {

    uint32_t cf : 5;          /* Color format: See `lv_img_color_format_t`*/
    uint32_t always_zero : 3; /*It the upper bits of the first byte. Always zero to look like a
                                 non-printable character*/

    uint32_t reserved : 2; /*Reserved to be used later*/

    uint32_t _w : 11; /*Placeholders*/
    uint32_t _h : 11; /*Placeholders*/
    uint32_t w; 
    uint32_t h; 
} lv_img_header_t;
```

incity, 2020-11-12:
maybe you can add a  macro to control it?

kisvegabor, 2020-11-13:
`bin` images (Binary format in the converter) use the same format so switching between different modes would be messy. :slightly_frowning_face: 

If we can either add another 4 bytes to the header or use the 2  reserved bits in width and height. 
IMO once we breaking changing adding 4 bytes would be better to allow more options in the future.

kisvegabor, 2020-11-26:
I've added it to the ROADMAp as an idea. 
https://github.com/lvgl/lvgl/blob/master/docs/ROADMAP.md

closed 2020-11-26 by kisvegabor