lvgl/lvgl #1893: Add a function to remove a series from a chart

Rabdf, 2020-11-11:
A function to remove a series from a chart (counterpart to lv_chart_add_series ()) is missing yet.

upbeat27, 2020-11-11:
I'm gonna piggy back on this and suggest a function for toggling series visibility.
Something like `lv_chart_show_series(bool show)`. Default would be `true` of course.
An example: https://www.flotcharts.org/flot/examples/series-toggle/index.html
Basically if there are a lot of series, it can be useful to hide some in order focus in on what you are interested in.
Right now you have to clear the series, and re-set all the data in order to achieve a similar result.

kisvegabor, 2020-11-13:
I agree with both suggestions. 
Do you have time and interest to contribute to these feature?

Rabdf, 2020-12-02:
I think I could write a function `lv_chart_remove_series ()`.
Adding a function like `lv_chart_show_series (bool show)` might be difficult as there is no flag attached to a series. Maybe we could change the color instead (to "invisible"); to make the series visible again one has to change the color to the one used before (or any other color).
Is there a kind of tutorial for code changes (how to create a fork, does anyone manage function names etc., and how to insert the changes into master)?

embeddedt, 2020-12-02:
> Is there a kind of tutorial for code changes (how to create a fork, does anyone manage function names etc., and how to insert the changes into master)?

Of course! https://docs.lvgl.io/latest/en/html/contributing/index.html#pull-request

There's also a [coding style guide](https://github.com/lvgl/lvgl/blob/master/docs/CODING_STYLE.md), but you can probably pick up the most important things just by looking at the functions. I will complain in the code review if something is noticeably wrong. :wink:

Rabdf, 2020-12-03:
#1948 adds `lv_chart_remove_series ()`.

Rabdf, 2020-12-07:
#1952 adds `lv_chart_hide_series (..., bool hide)`.

Rabdf, 2020-12-07:
#1952 merged into #1948

kisvegabor, 2021-01-04:
The fieature is added

closed 2021-01-04 by kisvegabor