lvgl/lvgl #1894: Image problem in tabview

senlinding, 2020-11-11:
The picture size is the same as the screen size, but after the picture in the tabview is displayed, there will be a blank space under the picture when you drag upwards.

The version: 7.7.0.
[image]

[image]

    lv_obj_t *tabview;
    tabview = lv_tabview_create(lv_scr_act(), NULL);
    /*Add 3 tabs (the tabs are page (lv_page) and can be scrolled*/
    lv_obj_t *tab1 = lv_tabview_add_tab(tabview, "Tab 1");
    lv_obj_t *tab2 = lv_tabview_add_tab(tabview, "Tab 2");
    lv_obj_t *tab3 = lv_tabview_add_tab(tabview, "Tab 3");

    lv_tabview_set_btns_pos(tabview, LV_TABVIEW_TAB_POS_NONE);
    lv_page_set_scrollbar_mode(tab1, LV_SCROLLBAR_MODE_OFF);

    /*Add content to the tabs*/
    lv_obj_t * img = lv_img_create(tab1, NULL);
    lv_img_set_src(img, "S:theme1.png");
    lv_img_set_pivot(img, 0, 0);

    
    lv_obj_t * label = lv_label_create(tab1, NULL);
    lv_label_set_text(label, "This the first tab\n\n"
                             "If the content\n"
                             "of a tab\n"
                             "become too long\n"
                             "the it \n"
                             "automatically\n"
                             "become\n"
                             "scrollable.");

    label = lv_label_create(tab2, NULL);
    lv_label_set_text(label, "Second tab");

    label = lv_label_create(tab3, NULL);
    lv_label_set_text(label, "Third tab");

kisvegabor, 2020-11-12:
Just remove the paddings:
```c
lv_obj_set_style_local_pad_all(tab1, LV_PAGE_PART_SCROLLABLE, LV_STATE_DEFAULT, 0);
lv_obj_set_style_local_pad_all(tab1, LV_PAGE_PART_BG, LV_STATE_DEFAULT, 0);
```

senlinding, 2020-11-13:
> ```c
> lv_obj_set_style_local_pad_all
> ```

Thanks!

closed 2020-11-13 by senlinding