lvgl/lvgl #1895: v6 lv_list incorrect deletion of buttons

nazar01, 2020-11-11:
I use v6 at the moment and I encountered a bug with incorrect deletion of buttons from list added to a group. Steps to reproduce:
1) Create lv_list with buttons
2) Add it to a group
3) Delete buttons
4) Click on the list
5) Get `Error: Invalid object` and `Error: lv_list_set_btn_selected`

I can reproduce it with lvgl simulator with v6.1.2
```c
//in main.c
// some code before
lv_obj_t *list;
void btn_event_cb(lv_obj_t * obj, lv_event_t event)
{
    lv_obj_t* selected_btn = lv_list_get_btn_selected(list);
    if(selected_btn != NULL)
        lv_obj_del(selected_btn);
}

int main(int argc, char ** argv)
{
//some code
    list = lv_list_create(lv_scr_act(), NULL);
    lv_list_add_btn(list, NULL, "");
    lv_group_t *group = lv_group_create();
    lv_group_add_obj(group, list);
    lv_obj_t *btn = lv_btn_create(lv_scr_act(), NULL);
    lv_obj_set_event_cb(btn, btn_event_cb);
//some code
}
```
Faulty behavior: after clicking on `btn` and clicking on the `list` I receive an error.
Correct behavior: no errors

The guilty code that calls `lv_list_set_btn_selected` with incorrect object is located in `lv_res_t lv_list_signal(lv_obj_t * list, lv_signal_t sign, void * param)` here:
```c
/*Else select the clicked button*/
else {
    /*Mark the last clicked button (if any) as selected because it triggered the focus*/
    lv_list_ext_t * ext = lv_obj_get_ext_attr(list);
    if(ext->last_clicked_btn) {
        lv_list_set_btn_selected(list, ext->last_clicked_btn);
        ext->last_clicked_btn = NULL;

    } else {
        if(ext->last_sel) {
            /* Select the last used button */
            lv_list_set_btn_selected(list, ext->last_sel); //HERE
        } else {
            /*Get the first button and mark it as selected*/
            lv_list_set_btn_selected(list, lv_list_get_next_btn(list, NULL));
        }
    }
}
```
ext->last_sel isn't NULL'ed after the button is deleted. By the way, list size is also incorrect when buttons are being deleted using lv_obj_del. Calling lv_list_remove keeps list size correct, but doesn't fix described bug. Here is a patch I used to quick-fix the problem:
```c
// in lv_list.c in function lv_list_remove()
while(e != NULL) {
    if(count == index) {
        if(e == ext->last_sel)            //added
            ext->last_sel = NULL;         //added
        if(e == ext->last_clicked_btn)    //added
            ext->last_clicked_btn = NULL; //added
        lv_obj_del(e);
        ext->size--;
        return true;
    }
    e = lv_list_get_next_btn(list, e);
    count++;
}
```

kisvegabor, 2020-11-13:
Hi,

Good catch, thank for the report. It affects v7 as well.

In v7 I fixed but could you send a PR for v6?

nazar01, 2020-11-13:
I can send PR on monday. Also your commit looks a bit suspicious: shouldn't new lines be just before lv_obj_del inside "if"?

kisvegabor, 2020-11-13:
> I can send PR on monday. 

Great, thanks.

> Also your commit looks a bit suspicious: shouldn't new lines be just before lv_obj_del inside "if"?

I don't get it. It still seems good to me. But maybe I'm just blind to my own mistake :slightly_smiling_face: 
```c
{
    LV_ASSERT_OBJ(list, LV_OBJX_NAME);

    lv_list_ext_t * ext = lv_obj_get_ext_attr(list);
    uint16_t count = 0;
    lv_obj_t * e   = lv_list_get_next_btn(list, NULL);
    while(e != NULL) {
        if(e == ext->last_sel_btn) ext->last_sel_btn = NULL;
        if(e == ext->act_sel_btn) ext->act_sel_btn = NULL;

        if(count == index) {
            lv_obj_del(e);
            return true;
        }
        e = lv_list_get_next_btn(list, e);
        count++;
    }
    return false;
}
```

nazar01, 2020-11-13:
You're iterating over lv_list buttons and if current button(called "e") is last_sel_btn, you set last_sel_btn to NULL. As last_sel_btn is supposed to be in the list it will be NULL'ed always except when you "return true" earlier. So if index of button being removed from list(called "index") is greater or equal than index of last_sel_btn then last_sel_btn is always NULL'ed. I think intended behaviour is to NULL last_sel_btn only if it is the button being deleted.

kisvegabor, 2020-11-13:
Ah, see it now, thanks. 
Fixed

kisvegabor, 2020-11-18:
As the issue is fixed in both v6 and v7 I close this issue.

Thank you for your contribution!

closed 2020-11-18 by kisvegabor