lvgl/lvgl #1896: LV_IMG_CF_ALPHA_8BIT image can not be zoom.

incity, 2020-11-12:
when i zoom a canvas with LV_IMG_CF_ALPHA_8BIT, the canvas will show anything.

Before zoom:

[image]

After zoom:

[image]

```
#define MASK_WIDTH 100
#define MASK_HEIGHT 50

void lv_demo_widgets(void)
{
    /* Create the mask of a text by drawing it to a canvas*/
    static lv_opa_t mask_map[MASK_WIDTH * MASK_HEIGHT];

    /*Create a "8 bit alpha" canvas and clear it*/
    lv_obj_t * canvas = lv_canvas_create(lv_scr_act(), NULL);
    lv_obj_set_pos(canvas, 100, 100);
    lv_canvas_set_buffer(canvas, mask_map, MASK_WIDTH, MASK_HEIGHT, LV_IMG_CF_ALPHA_8BIT);
    lv_canvas_fill_bg(canvas, LV_COLOR_WHITE, 255);

    /*Draw a label to the canvas. The result "image" will be used as mask*/
    lv_draw_label_dsc_t label_dsc;
    lv_draw_label_dsc_init(&label_dsc);
    label_dsc.color = LV_COLOR_BLACK;
    lv_canvas_draw_text(canvas, 5, 5, MASK_WIDTH, &label_dsc, "Text with gradient", LV_LABEL_ALIGN_CENTER);
    lv_img_set_zoom(canvas, 512);

    return;
}
```

incity, 2020-11-12:
And for the backgroud color should be WHITE, but why it shows BLACK, and the text color should be BLACK but WHITE.
if i use LV_IMG_CF_TRUE_COLOR_ALPHA, it will be OK.

kisvegabor, 2020-11-13:
Hi,

zoom and rotation works only if the whole image is available in memory. However `LV_IMG_CF_ALPHA_..` and `LV_IMG_CF_INDEXED_...` reads the image line by line. See [here](https://docs.lvgl.io/latest/en/html/widgets/img.html#transformations).

`LV_IMG_CF_ALPHA_` have one color with alpha. This color is set by `recolor` style property.

incity, 2020-11-13:
so how can i zoom a 8-bit color depth image?

embeddedt, 2020-11-13:
I think the canvas needs to be in true color mode.

kisvegabor, 2020-11-16:
There is no way to directly zoom 8 bit images. You really need to use canvas. But instead of 8bit + canvas, it's better to simply use an ARGB image.

kisvegabor, 2020-11-26:
As it's not planned to support transforming non-RGB images I close this issue.

closed 2020-11-26 by kisvegabor