lvgl/lvgl #1897: add LV_EVENT_COORD_CHG event.

incity, 2020-11-12:
add a LV_EVENT_COORD_CHG event to help applications to listen the object's position and size change event.
 in my requirement, i need to show the position of a draggable object on a textarea.

embeddedt, 2020-11-12:
You can use a custom signal handler for this.

https://github.com/lvgl/lvgl/blob/43771fa2f71d5323ef9dffc1ff0aee326d59f5ce/src/lv_core/lv_obj.c#L810-L814

incity, 2020-11-13:
but if so i need to save old signal handler for all objects(different kind of objects)

incity, 2020-11-17:
@kisvegabor please think about this~

kisvegabor, 2020-11-18:
Hi,

I was also thinking about it but found that it's too low level for the user. As a reference I  checked Observers in JS:
- [IntersectionObserver.observe()](https://developer.mozilla.org/en-US/docs/Web/API/MutationObserver/observe)
- [MutationObserver.observe()](https://developer.mozilla.org/en-US/docs/Web/API/IntersectionObserver/observe)
- [ResizeObserver.observe()](https://developer.mozilla.org/en-US/docs/Web/API/ResizeObserver/observe)

incity, 2020-11-18:
yes, but it's more better to give the users an ability(interface) to custom it.

incity, 2020-11-18:
[image]
https://android.googlesource.com/platform/frameworks/base/+/a175a5b/core/java/android/view/View.java

kisvegabor, 2020-11-19:
I'm afraid providing access to it will let the user to make infinite loops and performance costy stuff.

I haven't found a similar event in the link either.

embeddedt, 2020-11-19:
+1, I think it's better to keep it hidden; there are ways of solving this without needing an event.

incity, 2020-11-20:
in Android, users can listen the size changed event, in windows, there is also an WM_SIZE message.  
@embeddedt  yes, i know, i can solve it by override the signal handler and save the old one, but it seems sloppy.

embeddedt, 2020-11-20:
It's sloppy, yes, but exposing an event handler will cause other harder-to-diagnose problems when users start overflowing their stack because of recursive coordinate changes. At least with the signal handler you have to think about the architecture more before.

incity, 2020-11-20:
change the size in the size changed handler will cause infinite loops, maybe user should know that, It's just a simple logic.

kisvegabor, 2020-11-21:
My experience is that if something can cause issues it will cause issues. We need to keep in mind there is a lot of novice user of LVGL who has problems even with the basics of programming. By exposing such low-level control we will end up getting a lot of issues about it. 

I'm ok to expose something coordinate change-like this to people but it should be something directly good for.  As the observers [here](https://github.com/lvgl/lvgl/issues/1897#issuecomment-729524641).

embeddedt, 2020-12-28:
The only other option here is to not allow the event to be recursive, i.e. if the size changes inside the event handler, it will not be called again. This adds a code size penalty for tracking event recursion, however.

kisvegabor, 2020-12-29:
I'm still in favor of adding dedicated "observers" instead of exposing low-level things (if really needed). Besides being error-prone it's still difficult to use.

I suggest waiting if other requests arrive on this topic. And if we will see how/what to develop.

incity, 2020-12-29:
position changed event and size changed event is very common to application(users).

embeddedt, 2021-01-23:
Let's close it for now and see if any new requests come in for this feature.

closed 2021-01-23 by embeddedt