lvgl/lvgl #1902: Taps and containers names don't apply Arabic processing

ashrafkamel5, 2020-11-15:
# Description
the documentation said that only text inputs and static texts are those which don’t process Arabic words.
but that is the result.
[image]

and can someone explain why such a limitation exists if the processing is already implemented?
, is there a walkaround for this?
, and If I wanted to add it myself, can someone guide me on how to start?

hamidrm, 2020-11-15:
Hi @ashrafkamel5,

Some of your words (العصر، المغرب و...)  seems to be shown correctly. Is it right?
Otherwise, How did you convert your font? And what was your characters range for this?

> and can someone explain why such a limitation exists if the processing is already implemented?
, is there a walkaround for this?

This limitation can be removed. you can follow the [Arabic processing pull request](https://github.com/lvgl/lvgl/pull/1409) to start.

kisvegabor, 2020-11-16:
@ashrafkamel5 
Is the issue that the letters are not connected correctly (incorrect forms) or the direction (RTL/LTR) is incorrect? 
Can you send a minimal example to create only the tab view and describe the expected result?

In the meanwhile, we have another Arabic-related issue with the Text area and apparently, we have a possible solution but need someone who knows the language. Could you take look at this? https://github.com/lvgl/lvgl/issues/1888

ashrafkamel5, 2020-11-16:
in the forum, alsaleem00 suggested to me a solution that worked for the taps name but not the container [here](https://forum.lvgl.io/t/tab-name-doesnt-apply-arabic-processing/3715/9?u=ashrafkamel5)
he suggested calling the function that processes the words by myself like so,
```
   char ctx_text[40];
    _lv_txt_ap_proc(your_text, ctx_text);
```
and then pass the string to the item I desire,
which worked fine for the taps but not the container which is more than enough for me till now.
[image]

@kisvegabor 
> Is the issue that the letters are not connected correctly (incorrect forms) or the direction (RTL/LTR) is incorrect?

yes, it was an incorrect form. the direction is fine.
> In the meanwhile, we have another Arabic-related issue with the Text area and apparently, we have a possible solution but need someone who knows the language. Could you take look at this? #1888 

for sure I would.
 just read it and I am a native speaker. but I don't know how the fix would go really, didn't get your description very well, and I didn't make any pull request before, so a little anxious actually.

@hamidrm 
> This limitation can be removed. you can follow the [Arabic processing pull request](https://github.com/lvgl/lvgl/pull/1409) to start.

should I read your pull request or the discussion behind it which matters?
I don't know really how would I investigate your code. 🤷‍♀️

ashrafkamel5, 2020-11-16:
Mr @hamidrm 
I would appreciate it if you elaborated more on how would I remove the limitation. 
as it appears that Mr @kisvegabor  is asking for a similar fix for the text area which I also would need in a couple of days in my current project.

another thing is that I don't really understand the structure of the font .C file generated from the converter.
is this would be a problem?
and where can I learn more about the font file structure and elements?

ashrafkamel5, 2020-11-16:
and why every problem asked here is marked as a not-template ?
can you fix this, please?

kisvegabor, 2020-11-17:
I've pushed a fix to handle Arabic texts in button matrix (the tabs are made from a button matrix): 4903bdb9
Can you confirm if it works well?

Regarding the text area issue, I'd be happy to tell more about it, but please discuss it in #1888 to keep the discussions clean.

ashrafkamel5, 2020-11-21:
@kisvegabor  
yes, now they work without calling the `_lv_txt_ap_proc` directly,
a Similiar work needed for the Drop down objects also :D 

but the container name still doesn't process arabic words and doesn't accept the output of the `_lv_txt_ap_proc` function,
don't know why.

kisvegabor, 2020-11-21:
> yes, now they work without calling the _lv_txt_ap_proc directly,

Great!

For the drop down please open a new issue with a short code snippet to see the issue. Please also add the expected text.

ashrafkamel5, 2020-11-22:
ok, Thanks

closed 2020-11-22 by ashrafkamel5