lvgl/lvgl #1904: linemeter artefacts

estermann, 2020-11-16:
I have a problem with linemeter. lv_linemeter_set_value doesn't redraw correctly if data jumps (for example from 100 to 0). if i use the following code only the the lines until "test_data" and the next line are set correctly (to blue). The following lines show artefacts or are blue too. lvgl version is 7.7.1. plattform is esp32. please help.

`
#include <lvgl.h>  
#include <Ticker.h>
#include <TFT_eSPI.h> 
#define LVGL_TICK_PERIOD 20
Ticker tick; /* timer for interrupt handler */
TFT_eSPI tft = TFT_eSPI(); /* TFT instance */
lv_disp_buf_t disp_buf;
lv_color_t buf[LV_HOR_RES_MAX * 10];

static void lv_tick_handler(void) { // Interrupt driven periodic handler
  lv_tick_inc(LVGL_TICK_PERIOD);
}

void my_disp_flush(lv_disp_drv_t *disp, const lv_area_t *area, lv_color_t *color_p)
{
    uint32_t w = (area->x2 - area->x1 + 1);
    uint32_t h = (area->y2 - area->y1 + 1);
    tft.startWrite();
    tft.setAddrWindow(area->x1, area->y1, w, h);
    tft.pushColors(&color_p->full, w * h, true);
    tft.endWrite();
    lv_disp_flush_ready(disp);
}

int test_data=0;
lv_obj_t * temp0_meter;

void data_update_task(lv_task_t * task){
    lv_linemeter_set_value(temp0_meter, test_data);                  //Set the current value
}

void setup()
{
  Serial.begin(115200);
  
  lv_init();
  tft.begin(); /* TFT init */
  tft.writecommand(TFT_MADCTL); //displayrotation landscape
  tft.writedata(0b10100100);  //displayrotation landscape
  
  lv_disp_buf_init(&disp_buf, buf, NULL, LV_HOR_RES_MAX * 10);
  lv_disp_drv_t disp_drv;
  lv_disp_drv_init(&disp_drv);
  disp_drv.hor_res = 160;
  disp_drv.ver_res = 128;
  disp_drv.flush_cb = my_disp_flush;
  disp_drv.buffer = &disp_buf;
  lv_disp_drv_register(&disp_drv);
  tick.attach_ms(LVGL_TICK_PERIOD, lv_tick_handler); //Initialize the graphics library's tick
  
  temp0_meter = lv_linemeter_create(lv_scr_act(), NULL);
  lv_linemeter_set_range(temp0_meter, 0, 100);                   //Set the range
  lv_linemeter_set_scale(temp0_meter, 240, 10);                  //Set the angle and number of lines
  lv_obj_set_size(temp0_meter, 120, 120);
  lv_obj_align(temp0_meter, NULL, LV_ALIGN_CENTER, 0, 0);
  lv_task_create(data_update_task, 100, LV_TASK_PRIO_MID, NULL);
}

void loop()
{
   static long lasttime_lv_task=0;
   static long test_data_update=0;
   
   if (millis()-lasttime_lv_task>5) {
      lv_task_handler(); 
      lasttime_lv_task=millis();
   }
   
   if((millis()-test_data_update)>20){ 
     test_data++;
     if (test_data>=99)test_data=1;
     test_data_update = millis();
   }
          
}`

kisvegabor, 2020-11-26:
Hi,

I could reproduce the issue and will investigat it tomorrow.

kisvegabor, 2020-11-27:
I've fixed it in `master`.

closed 2020-11-27 by kisvegabor

estermann, 2020-11-27:
thank you very much. I will try it this we

estermann, 2020-11-27:
[20201127_225340.zip](https://github.com/lvgl/lvgl/files/5609663/20201127_225340.zip)
srry but it still doesn't work as expected. see attached video

kisvegabor, 2020-12-01:
It seems like a display driver issue to me.

commit 4853eacf (2020-11-27): fix(linemeter): fix invalidation when setting new value

The old complex but more optimal invalidation was buggy in some cases

fixes #1904
commit 6c62a8ee (2020-11-27): fix(linemeter): fix invalidation when setting new value

The old complex but more optimal invalidation was buggy in some cases

fixes #1904