lvgl/lvgl #1905: lvgl tailoring

zdp1994, 2020-11-16:
Hello, I have been studying lvgl recently. I am currently trying to tailor lvgl, but according to the section 3.11 and 3.12 of the document "LittlevGL Documentation", there is still 700KB at the end. Do you have any more detailed suggestions?

kisvegabor, 2020-11-16:
Have you disabled the unused fonts?

zdp1994, 2020-11-16:
yes , i has set “#define LV_FONT_MONTSERRAT_12    0” ,and so on. I saw an introduction that lvgl can be cut to 64kb, How did you do it？
[lv_config.txt](https://github.com/lvgl/lvgl/files/5546095/lv_config.txt)

zdp1994, 2020-11-16:
The macro control method for selecting screen size is no longer applicable

embeddedt, 2020-11-16:
The best way to profile size is to run your final ELF through `<cross-compile prefix>nm --print-size --size-sort`; the largest functions/symbols will be put at the bottom.

zdp1994, 2020-11-17:
I don’t quite understand what you mean, is it to modify the compile instructions of the makefile?

embeddedt, 2020-11-17:
You can use `nm` to find what functions are taking up the most space; you would need to run this as a command on a terminal.

zdp1994, 2020-11-18:
The above are all ways to reduce the ROM. My current demo RAM reaches 3.5M when running. How can I reduce the RAM size?
Also, I hope that the module of lvgl can be optional and configurable, just need to set the macro in lv_conf.h? What modules are necessary, can you provide a reference for a minimum configuration file?

kisvegabor, 2020-11-19:
The main RAM usage of LVGL consists of 2 things:
- LV_MEM_SIZE
- the size of display and frame buffers

You can also use `nm` to show RAM usage:

```
functions:  nm -S --size-sort my_program.elf -t d | grep -P "^[0-9]+ [0-9]+ [tT]" | tail -20
const data:  nm -S --size-sort my_program.elf -t d | grep -P "^[0-9]+ [0-9]+ [rR]" | tail -20
RAM usage  nm -S --size-sort my_program.elf -t d | grep -P "^[0-9]+ [0-9]+ [bBdD]" | tail -20
```

zdp1994, 2020-11-25:
It is planned to carry out the optional configuration of LVGL module according to the actual situation, but I have been busy with other things recently. It is estimated that this matter will not be pushed forward until next week

zdp1994, 2020-11-26:
I tried to run " nm -S --size-sort mypro -td | grep -P "^[0-9]+ [0-9]+ [bBdD]" | tail -20",
 it shows 00307832 00001764 d scenes, 
what do scenes mean, is it a screen or a scene?
 English translation into Chinese has a little semantic error

embeddedt, 2020-11-26:
> what do scenes mean, is it a screen or a scene?

`scenes` is a variable in your code somewhere. Off the top of my head the columns are arranged like this:
```
 address   size  sec var.name
========|========|=|==========
00307832 00001764 d scenes
```

zdp1994, 2020-11-26:
According to the lvgl website, lvgl can be cut to less than 2M at least and RAM is less than 100K. Have you tried it here?
 Is it really possible?

embeddedt, 2020-11-26:
There's no way LVGL uses 2 megabytes of flash unless you have other assets like images or large fonts, or unless you have optimization disabled. You should check in `nm` what the culprit is.

zdp1994, 2020-11-27:
On the official website of lvgl-"https://docs.lvgl.io/latest/en/html/intro/index.html",
There is such a description of the characteristics of lvgl
"Scalable to operate with little memory (64 kB Flash, 16 kB RAM)"
You mean lvgl cannot be less than 2M, so is the description on the official website wrong? How big can it be at least?

embeddedt, 2020-11-27:
> You mean lvgl cannot be less than 2M, so is the description on the official website wrong

LVGL should never be 2 megabytes. It should not be larger than 2 megabytes. I have never seen it take up more than 400KB and that is in an unoptimized scenario.

zdp1994, 2020-11-27:
I am using a 720*1280 screen, and compiled the source code of lvgl into a dynamic link library of lib.so. The lib.so file itself reaches 4.6Mb.
When running lvgl, the memory is far more than 2M
Can I refer to your makefile file?

zdp1994, 2020-11-27:
I used the 7.4.0 version of the code
I found an extra Kconfig file in the latest version of 7.7. Is there any introduction to this file?How to use the old version?

embeddedt, 2020-11-27:
On Linux there is a lot of overhead caused by dynamic libraries, etc., so there is probably quite a bit of memory being used for that which LVGL is not using.

kisvegabor, 2020-11-27:
You can try this project with it's Makefile: https://github.com/lvgl/lv_sim_eclipse_sdl

zdp1994, 2020-11-30:
This is the makefile I referred to before. I tried to compile without using the dynamic link library. The compiled project is indeed much smaller, but the running memory is still relatively large. I will continue to try to reduce it.
I am now thinking about how to control whether the module is compiled in a form similar to kconfig. I found out that the kconfig file has been added to the latest version. How do you plan to use it?

embeddedt, 2020-11-30:
> I am now thinking about how to control whether the module is compiled in a form similar to kconfig.

You can do this with `lv_conf.h`.

> I found out that the kconfig file has been added to the latest version. How do you plan to use it?

This is just for better compatibility with NuttX, Zephyr, and ESP-IDF. It is essentially just exposing the settings from `lv_conf.h` in a manner that is more usable with these OSes.

zdp1994, 2020-12-01:
Thank you, I used lv_config for tailoring before. I see that there are more kconf files in the latest version, thinking that this updated and better configuration method. Now I am working hard to reduce the running memory of my project.

zdp1994, 2020-12-16:
In my project, RAM has reached 3M+, the following are some parameters
The screen size is 720*1280,
Single buffer display,
# define LV_MEM_SIZE (64U * 1024U)
Mainly RAM is consumed on the fbp variable
fbp = (char *)mmap(0, screensize, PROT_READ | PROT_WRITE, MAP_SHARED, fbfd, 0);
screensize=720*1280*4
I tried banning all obj, banning all fonts, and turning off the display. At the end, there is still 700kb of RAM

zdp1994, 2020-12-16:
By the way, I would like to ask a small question or suggestion, how to change the color of the displayed text?
Is there such a wrapper function that can realize different color display?
void lv_label_set_text_color(lv_obj_t * label, const char * text,in color)
Is there a similar function?

embeddedt, 2020-12-16:
Technically `mmap` is not consuming RAM; Linux has already reserved that RAM for the framebuffer. Also keep in mind that Linux platforms will show much larger RAM usage than bare-metal, simply because of the added layers of complexity.

> how to change the color of the displayed text?

Change the [`text_color`](https://docs.lvgl.io/latest/en/html/overview/style.html#text-properties) property of `LV_LABEL_PART_MAIN`'s style.

zdp1994, 2020-12-17:
> Technically `mmap` is not consuming RAM; Linux has already reserved that RAM for the framebuffer. Also keep in mind that Linux platforms will show much larger RAM usage than bare-metal, simply because of the added layers of complexity.
> 
Yes, pure mmap will not cause an increase in memory.
But using "memset(fbp, 0, screensize);"
Or "memcpy(&fbp[location], (uint32_t *)color_p, (act_x2-act_x1 + 1) * 4);" After the assignment operation, the memory will increase

zdp1994, 2020-12-17:
The other is based on the previously provided method
functions: nm -S --size-sort lvgldemo -t d | grep -P "^[0-9]+ [0-9]+ [tT]"
const data: nm -S --size-sort lvgldemo -t d | grep -P "^[0-9]+ [0-9]+ [rR]"
RAM usage nm -S --size-sort lvgldemo -t d | grep -P "^[0-9]+ [0-9]+ [bBdD]"

But the value I got after adding up the three is very different from the memory consumption value obtained by top
Are there other possible places that consume memory?

embeddedt, 2020-12-17:
> Or "memcpy(&fbp[location], (uint32_t *)color_p, (act_x2-act_x1 + 1) * 4);" After the assignment operation, the memory will increase

I'm still not convinced that the system is actually using more RAM, but okay. :slightly_smiling_face: 

> Are there other possible places that consume memory?

Many. As I've mentioned Linux adds quite a bit of overhead. The C library, dynamic libraries, etc. could all be using more memory.

zdp1994, 2020-12-21:
> I'm still not convinced that the system is actually using more RAM, but okay. 🙂
> Many. As I've mentioned Linux adds quite a bit of overhead. The C library, dynamic libraries, etc. could all be using more memory.

I found a new method to detect memory consumption. The data obtained by this method is more detailed and more realistic. You can refer to the following URL
https://www.golinuxcloud.com/check-memory-usage-per-process-linux/

After analysis, if the optimization is not performed, the memory PSS used is 2000kb, of which lvgl-related only accounts for about 600kb, and the others are mainly SDL2.
b6dff000-b6eb8000 r-xp 00000000 103:0d 8710 /usr/lib/libSDL2-2.0.so.0.8.0
Size: 740 kB
KernelPageSize: 4 kB
MMUPageSize: 4 kB
Rss: 420 kB
Pss: 420 kB
There is another piece that I don’t know what it is (see below). It feels related to the size of the set "LV_MEM_SIZE". After setting the "LV_MEM_SIZE" to 4kb, this piece is not displayed
b5cc9000-b659c000 rwxp 00000000 00:00 0
Size: 9036 kB
KernelPageSize: 4 kB
MMUPageSize: 4 kB
Rss: 852 kB
Pss: 852 kB

paulie-g, 2020-12-22:
The correct way to estimate memory use of a process on a modern linux is to parse its /proc/$pid/smaps to establish what its unique set size (USS) is. PSS is relatively pointless, since it depends on how many other processes have, say, the same shared libs mapped into their memory space. Thus it varies with the state of your system. How you apportion RSS - USS to a given process depends on what you care about. On a graphical desktop, say, you can assume most Qt, GL etc libs to be in memory already and therefore relatively 'free' besides heap. On a barebones Linux-on-SBC system, they might be full price as the process in question might be the only thing using them.

closed 2021-01-25 by stale[bot]