lvgl/lvgl #1906: Performance analysis of LVGL

zdp1994, 2020-11-16:
I run lvgl's own course on my device—— "lv_demo_benchmark.h"
and I used double buffer,——lv_disp_buf_init(&disp_buf1, buf1_1, buf1_2, LV_HOR_RES_MAX * LV_VER_RES_MAX);	
the display dev is fb,
ioctl(fbfd, FBIOGET_FSCREENINFO, &finfo) // Get fixed screen information
ioctl(fbfd, FBIOGET_VSCREENINFO, &vinfo)//Get variable screen information
ioctl(fbfd, FBIOPUT_VSCREENINFO, &vinfo) //Set the device's screen information，But the hardware actually only supports up to 60hz，
So I want to know what is the mechanism of lvgl's dual buffer display, and which file to look at?

 The fps displayed on the screen exceeds 100，but

kisvegabor, 2020-11-17:
LVGL just draws what it needs in in every `LV_DISP_DEF_REFR_PERIOD`.

For > 100 FPS see the explanation here: https://github.com/lvgl/lv_examples/tree/master/src/lv_demo_benchmark

zdp1994, 2020-11-17:
I am very excited to be able to ask a big cow like you (the Chinese call for a very skilled person).
To be honest, I was mentally prepared to receive a reply after a week at first. I hope that my frequent consultations will not disturb you.
If so, I can only apologize, but I will continue to consult frequently😊

zdp1994, 2020-11-17:
I have set the variable value of LV_DISP_DEF_REFR_PERIOD to 15, which means that the maximum refresh rate is 1000/15=66.67.
At the same time, I also uncommented lv_obj_invalidate(lv_scr_act()),
The frame rate displayed on the screen did not exceed 100, but it was basically around 80.
and disp_drv.flush_cb = fbdev_flush;
The interval between the two fbdev_flush functions is actually between 13-17ms, which is in line with expectations

zdp1994, 2020-11-17:
Now my question is why there is such a big error between the frame rate displayed by monitor_cb and the frame rate obtained by testing fbdev_flush interval?
Is the real frame rate obtained by obtaining fbdev_flush interval time?

kisvegabor, 2020-11-18:
> I am very excited to be able to ask a big cow like you (the Chinese call for a very skilled person).

Thanks, but certainly not I'm the "biggest cow" here. :laughing: 

> At the same time, I also uncommented lv_obj_invalidate(lv_scr_act()),

This way always the whole screen will be redrawn which is not always necessary. E.g. is an image is moved 1 px, only the image +1 pixel area should be redrawn. 

The benchmark measures the pure rendering time which can be e.g. 2 ms and converts it to FPS. 2 ms -> 500 FPS. It doesn't mean your display is refreshed 500 times / sec but LVGL's theoretical max speed would be 500 FPS. 
 
> The interval between the two fbdev_flush functions is actually between 13-17ms, which is in line with expectations

That's correct. `flush_cb` is called the rate of `LV_DISP_DEF_REFR_PERIOD`.

zdp1994, 2020-11-18:
Thank you very much, I feel that my doubts about frame rate have been resolved. Next, I want to go deep into the lvgl kernel to check the principle of drawing and refreshing the screen. Do you have any good suggestions?

kisvegabor, 2020-11-18:
The basics of drawing are described here: https://docs.lvgl.io/latest/en/html/overview/drawing.html

zdp1994, 2020-11-27:
This question has been understood, thanks for the explanation, it can be closed

closed 2020-11-27 by zdp1994